Ysurac/openmptcprouter #3496: Errors in the omr-tracker-server in the log.

vempire-ghost, 2024-08-15:
Many of these errors in the log, but the OMR seems to work fine; something to worry about?
Aug 15 02:56:28 OpenMPTCProuter daemon.err omr-tracker-server[15480]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 15 02:58:51 OpenMPTCProuter daemon.err omr-tracker-server[15480]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 15 03:07:33 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Aug 15 03:07:33 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Aug 15 03:09:42 OpenMPTCProuter daemon.err omr-tracker-server[15480]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 15 03:09:53 OpenMPTCProuter daemon.err omr-tracker-server[15480]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 15 03:15:35 OpenMPTCProuter daemon.err omr-tracker-server[15480]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-snapshot-5.4 r0+16862-1 snap from 14/08
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1031-test
  - OpenMPTCProuter VPS provider: GCP
  - OpenMPTCProuter platform: x86_64
  - Country:

Ysurac, 2024-08-16:
This will be fixed in next snapshot.

vempire-ghost, 2024-08-31:
In the today snaptshot the erros is still present.

Aug 31 22:21:20 OpenMPTCProuter daemon.err omr-tracker-server[15456]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 31 22:24:14 OpenMPTCProuter daemon.err omr-tracker-server[15456]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 31 22:41:50 OpenMPTCProuter daemon.err omr-tracker-server[15456]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 31 22:49:27 OpenMPTCProuter daemon.err omr-tracker-server[15456]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 31 22:52:28 OpenMPTCProuter daemon.err omr-tracker-server[15456]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 31 22:53:36 OpenMPTCProuter daemon.err omr-tracker-server[15456]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 31 22:56:15 OpenMPTCProuter daemon.err omr-tracker-server[15456]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe
Aug 31 22:59:04 OpenMPTCProuter daemon.err omr-tracker-server[15456]: /bin/omr-tracker-server: line 52: echo: write error: Broken pipe