Ysurac/openmptcprouter #3498: ZBT Z8102AX

Yessine1200, 2024-08-19:
## Expected Behavior
OMR works perfectly with the ZBT Z8102AX with two 5G modems(Quectel RM520N-GL).
## Current Behavior
I have two 5G modems in the ZBT Z8102AX-T, which I configured with ModemManager. On the status page, I only see the LTE logo for both modems, and it only uses 4G, not 5G, so the aggregation is for two 4G connections, not 5G. And also I see warnings on the status page (is that normal ?)
[image]

## Specifications

  - OpenMPTCProuter version: snapshot 6.1 6.6
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: ZBT Z8102AX

Yessine1200, 2024-08-19:
` First modem:
`-----------------------------------
  General  |                    path: /org/freedesktop/ModemManager1/Modem/0
           |               device id: 031616276bf9c4e3e4cf8e5d6c7763ff33864dca
  -----------------------------------
  Hardware |            manufacturer: Quectel
           |                   model: RM520N-GL
           |       firmware revision: RM520NGLAAR01A07M4G
           |          carrier config: ROW_Commercial
           | carrier config revision: 0A010809
           |            h/w revision: 20000
           |               supported: gsm-umts, lte, 5gnr
           |                 current: gsm-umts, lte, 5gnr
           |            equipment id: 868371052867849
  -----------------------------------
  System   |                  device: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
           |                 physdev: /sys/devices/platform/soc/11200000.usb/usb2/2-1/2-1.1
           |                 drivers: option1, qmi_wwan
           |                  plugin: quectel
           |            primary port: cdc-wdm0
           |                   ports: cdc-wdm0 (qmi), ttyUSB0 (ignored), ttyUSB1 (gps), 
           |                          ttyUSB2 (at), ttyUSB3 (at), wwan0 (net)
  -----------------------------------
  Status   |                    lock: sim-pin2
           |          unlock retries: sim-pin (3), sim-puk (10), sim-pin2 (3), sim-puk2 (10)
           |                   state: connected
           |             power state: on
           |             access tech: lte, 5gnr
           |          signal quality: 62% (recent)
  -----------------------------------
  Modes    |               supported: allowed: 3g; preferred: none
           |                          allowed: 4g; preferred: none
           |                          allowed: 3g, 4g; preferred: 4g
           |                          allowed: 3g, 4g; preferred: 3g
           |                          allowed: 5g; preferred: none
           |                          allowed: 4g, 5g; preferred: 5g
           |                          allowed: 4g, 5g; preferred: 4g
           |                          allowed: 3g, 5g; preferred: 5g
           |                          allowed: 3g, 5g; preferred: 3g
           |                          allowed: 3g, 4g, 5g; preferred: 5g
           |                          allowed: 3g, 4g, 5g; preferred: 4g
           |                          allowed: 3g, 4g, 5g; preferred: 3g
           |                 current: allowed: 3g, 4g, 5g; preferred: 5g
  -----------------------------------
  Bands    |               supported: utran-1, utran-4, utran-6, utran-5, utran-8, utran-2, 
           |                          eutran-1, eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, eutran-8, 
           |                          eutran-12, eutran-13, eutran-14, eutran-17, eutran-18, eutran-19, 
           |                          eutran-20, eutran-25, eutran-26, eutran-28, eutran-29, eutran-30, 
           |                          eutran-32, eutran-34, eutran-38, eutran-39, eutran-40, eutran-41, 
           |                          eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, eutran-71, 
           |                          utran-19, ngran-1, ngran-2, ngran-3, ngran-5, ngran-7, ngran-8, 
           |                          ngran-12, ngran-13, ngran-14, ngran-18, ngran-20, ngran-25, ngran-26, 
           |                          ngran-28, ngran-29, ngran-30, ngran-38, ngran-40, ngran-41, ngran-48, 
           |                          ngran-66, ngran-70, ngran-71, ngran-75, ngran-76, ngran-77, ngran-78, 
           |                          ngran-79
           |                 current: utran-1, utran-4, utran-5, utran-8, utran-2, eutran-1, 
           |                          eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, eutran-8, 
           |                          eutran-12, eutran-13, eutran-14, eutran-17, eutran-18, eutran-19, 
           |                          eutran-20, eutran-25, eutran-26, eutran-28, eutran-29, eutran-30, 
           |                          eutran-32, eutran-34, eutran-38, eutran-39, eutran-40, eutran-41, 
           |                          eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, eutran-71, 
           |                          utran-19, ngran-1, ngran-2, ngran-3, ngran-5, ngran-7, ngran-8, 
           |                          ngran-12, ngran-13, ngran-14, ngran-18, ngran-20, ngran-25, ngran-26, 
           |                          ngran-28, ngran-29, ngran-30, ngran-38, ngran-40, ngran-41, ngran-48, 
           |                          ngran-66, ngran-70, ngran-71, ngran-75, ngran-76, ngran-77, ngran-78, 
           |                          ngran-79
  -----------------------------------
  IP       |               supported: ipv4, ipv6, ipv4v6
  -----------------------------------
  3GPP     |                    imei: XXXXXXXXXXX
[... 75 lines cut ...]
           |                         eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, eutran-71, 
           |                         utran-19, ngran-1, ngran-2, ngran-3, ngran-5, ngran-7, ngran-8, 
           |                         ngran-12, ngran-13, ngran-14, ngran-18, ngran-20, ngran-25, ngran-26, 
           |                         ngran-28, ngran-29, ngran-30, ngran-38, ngran-40, ngran-41, ngran-48, 
[... cut ...]
  3GPP EPS |   ue mode of operation: csps-2
           |    initial bearer path: /org/freedesktop/ModemManager1/Bearer/4
           | initial bearer ip type: ipv4v6
  ----------------------------------
  SIM      |       primary sim path: /org/freedesktop/ModemManager1/SIM/0
           |         sim slot paths: slot 1: /org/freedesktop/ModemManager1/SIM/0 (active)
           |                         slot 2: none
  ----------------------------------
  Bearer   |                  paths: /org/freedesktop/ModemManager1/Bearer/5

Ysurac, 2024-08-19:
For the LTE, it's due to "access tech: lte, 5gnr". I think it's using NSA mode, so still using 4G for control plane and 5G NR for user plane.
I will check if I can add both used tech.
For the "Gateway down" message, not a problem. Gateway often doesn't answer to ping in mobile network.
For the "unknow status", I will check.

Yessine1200, 2024-08-19:
Yes, it's NSA mode, but if I put the same SIM card in another router and connect it to OMR with an Ethernet cable, it will use 5G technology, not LTE (the LTE logo doesn't appear).
Ok, keep me updated. Thank you!

Ysurac, 2024-08-19:
It's because the other modem report only 5G even if still using also 4G for control part.

Yessine1200, 2024-08-19:
Also, when running omr-test-speed wwan0 and omr-test-speed wwan1, I get the full speed of the 5G connection, so it seems to work. However, when I perform aggregation using omr-test-speed I got only:
[image]
For wwan0:
[image]
For wwan1:
[image]

Also, I noticed there's a bug with the real-time traffic on the MPTCP page; it's not working properly.

Ysurac, 2024-08-19:
What is not working in MPTCP page ?

Yessine1200, 2024-08-19:
The real-time traffic should display the speed of each connection and the aggregated speed. However, the results shown when running omr-test-speed do not match what is displayed on the real-time traffic page, it sometimes takes a while for the traffic to appear and sometimes I have to refresh the page to work.

Ysurac, 2024-08-19:
Don't forget that omr-test-speed is in B/s while MPTCP traffic page is in b/s. The page is more to check if aggregation is working or not. So it's working or aggregation is only using one connection ?
You seems to have a high load average on the box, this can really make aggregation slower.
For information, from my test, we can reach with Shadowsocks-Rust as Proxy with AES, about 400Mb/s in perfect conditions on Z8102AX.

Yessine1200, 2024-08-19:
Yes, I know it's working, but it takes longer to visualize the traffic compared to previous versions.
I'm using the same configuration as you, but I haven't been able to reach 400 Mbps. Regarding the load average, even without performing aggregation, the load average remains high.

Ysurac, 2024-08-19:
The load average should stay < 2 on this box.
What do you have in Status->System log (or via SSH using ```logread``` command) ?

Ysurac, 2024-08-20:
For the SIM hotplug, yes it's not working, doesn't seems to be a bug on OMR side as the modem doesn't recognize correctly the new SIM.
For th log, it's just after boot of somes minutes after ?

Yessine1200, 2024-08-20:
And there is no solution to fix that ?
Yes, 3-5 minutes after boot.

Yessine1200, 2024-08-20:
I still can't achieve 400 Mbps. Even after changing the congestion protocol multiple times, I got better speeds with the highspeed algorithm compared to the others, but I still haven't reached 400 Mbps.
The load is < 5 during aggregation and < 2 when not.

Ysurac, 2024-08-24:
Can you give me the result of ```cat /proc/cpuinfo``` ?

Yessine1200, 2024-08-26:
hello Ysurac,
processor	: 0
BogoMIPS	: 26.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 26.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Ysurac, 2024-08-26:
OK, it was to be sure AES was available. Can you do the same on the VPS ?

Yessine1200, 2024-08-26:
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 15
model		: 107
model name	: QEMU Virtual CPU version 2.5+
stepping	: 1
microcode	: 0x1
cpu MHz		: 3092.734
cache size	: 16384 KB
physical id	: 0
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx lm constant_tsc nopl xtopology cpuid tsc_known_freq pni ssse3 cx16 sse4_1 sse4_2 x2apic popcnt aes hypervisor lahf_lm cpuid_fault pti
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown bhi
bogomips	: 6185.46
clflush size	: 64
cache_alignment	: 128
address sizes	: 40 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 15
model		: 107
model name	: QEMU Virtual CPU version 2.5+
stepping	: 1
microcode	: 0x1
cpu MHz		: 3092.734
cache size	: 16384 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx lm constant_tsc nopl xtopology cpuid tsc_known_freq pni ssse3 cx16 sse4_1 sse4_2 x2apic popcnt aes hypervisor lahf_lm cpuid_fault pti
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown bhi
bogomips	: 6185.46
clflush size	: 64
cache_alignment	: 128
address sizes	: 40 bits physical, 48 bits virtual
power management:

processor	: 2
vendor_id	: GenuineIntel
cpu family	: 15
model		: 107
model name	: QEMU Virtual CPU version 2.5+
stepping	: 1
microcode	: 0x1
cpu MHz		: 3092.734
cache size	: 16384 KB
physical id	: 1
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 2
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx lm constant_tsc nopl xtopology cpuid tsc_known_freq pni ssse3 cx16 sse4_1 sse4_2 x2apic popcnt aes hypervisor lahf_lm cpuid_fault pti
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown bhi
bogomips	: 6185.46
clflush size	: 64
cache_alignment	: 128
address sizes	: 40 bits physical, 48 bits virtual
power management:

processor	: 3
vendor_id	: GenuineIntel
cpu family	: 15
model		: 107
model name	: QEMU Virtual CPU version 2.5+
stepping	: 1
microcode	: 0x1
cpu MHz		: 3092.734
cache size	: 16384 KB
physical id	: 1
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 3
initial apicid	: 3
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx lm constant_tsc nopl xtopology cpuid tsc_known_freq pni ssse3 cx16 sse4_1 sse4_2 x2apic popcnt aes hypervisor lahf_lm cpuid_fault pti
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit mmio_unknown bhi
bogomips	: 6185.46
clflush size	: 64
cache_alignment	: 128
address sizes	: 40 bits physical, 48 bits virtual
power management:

Yessine1200, 2024-08-26:
Yes, AES is used. Also, for the upload speed, it is supposed to reach nearly 40 Mbps with aggregation, but I only get half of that. What could be the reason for not achieving the expected speed?

Ysurac, 2024-08-26:
I think it's due to high load average, but I don't know what you get so high load average...

Yessine1200, 2024-08-26:
I disabled the two radios, which improved the load average and slightly improved the aggregation, but I still haven't achieved 400 Mbps.

Ysurac, 2024-08-26:
What is the speed now using ```omr-test-speed``` ?

Yessine1200, 2024-08-26:
For now, it can go over 30M but still not the full speed
[image]

Ysurac, 2024-08-26:
31MB/s so 248Mb/s
Still a high load average ? SQM is disabled ?

Yessine1200, 2024-08-26:
Under 5 for now. What should be the load average when doing aggregation ?
Yes, SQM is disabled.

Ysurac, 2024-08-26:
The load should be under 2.

Ysurac, 2024-08-26:
If you disable encryption in the wizard, what speed do you get ?

Yessine1200, 2024-08-27:
When I disabled encryption, the throughput dropped, and I also encountered a problem with the server.
[image]
the aggregated speed:
[image]

Honestly, I don't understand why the load average is high—it's driving me crazy.

Ysurac, 2024-08-27:
The load average is also high without traffic ?
There is a problem on modem 1 & modem 2 on the screenshot : they are using same interface.

Yessine1200, 2024-08-27:
Without traffic, the load average is always below 2. However, when I start a speed test, it increases.
I set the correct interface for modem1, but I'm still getting an error on the server side when encryption is disabled.
[image]

Ysurac, 2024-08-27:
Without traffic (and without wifi), the load average should be below 1 on this box.
For Shadowsocks-Rust with encryption disabled there is an issue, I will fix that.

Yessine1200, 2024-08-28:
for now without traffic, the load average is below 1.
[image]
But still when doing aggregation, it's above 2.

Will this be fixed in the next snapshot?

Ysurac, 2024-08-28:
The fix is on VPS API, so you need to run again VPS script to apply the fix.

Ysurac, 2024-08-28:
You may have to restart VPS after script install and to validate wizard again.

closed 2024-12-02 by github-actions[bot]