Ysurac/openmptcprouter #3499: EdgeRouter X for v0.60

driccee, 2024-08-20:
Hello,

First of all, thank you for the very novel solution, the truth is that it looks quite promising and I hope it continues to improve.

I will comment a little about my attempts to test the solution. I am using **EdgeRouter X** as a router, version: v0.58.4, it works well, the detail is when I try to change the version to the most current one, the system is constantly booting, I don't know if it is not compatible or if an update is required, I have tried to upload a firmware but unfortunately I have not been able to upload it, I understand that the most current versions require a kernel higher than 6.x. Just to confirm, I don't know if you can help me that the hardware is not compatible?

In addition to this, I have purchased this equipment from Amazon: 
https://www.amazon.com/SENSTUN-Firewall-Appliance-Puertos-Barebone/dp/B0CG6RJ6KT/ref=sr_1_2_sspa?crid=2XFSA5F0BAUPQ&dib=eyJ2IjoiMSJ9.2LHtlLH_xkFgiOVelojVnGHXlW_Scrdza3XP3Lquvm-rDqMRNntDzKxb1SVQleZytGBncP7ayBwYoe1ImPdXZaru9h8uEcgmZi6tLzKLhRmM9_Xy88GHM0e950ihASRsAUc7Nbd-JCL8rsgWSuXerrV2ua0ZYHP84F2r3IMYOYhZHxyxiOkkWYHbyeXXxKlHplyHu_iM8oWh4VLnHcpUE7Of9uuPYMa2uTix1gVVs3o.SSVhwpvltYD9ewIlXNYXRbhwc4FHhHMh64v1Vqt-msA&dib_tag=se&keywords=router%2Bbarebone&qid=1720539380&sprefix=%2Caps%2C271&sr=8-2-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1

And Unfortunately it restarts every time, I uploaded the image directly to the SSD with Etcher, I also tried with USB and microSD, I get the same problem, I don't know if it's because of the hardware or something I'm doing wrong. Maybe someone can guide me in something. I'm looking for where I can install and not get these problems.

Thanks.

driccee, 2024-08-20:
Hello @Ysurac , 

Sorry for the category selecction, 

I'm using this images. See if you can help me with this. 

**Ubiquiti Edgerouter X** (alpha support with issues)
squashfs image: [openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-ramips-mt7621-ubnt_edgerouter-x-squashfs-sysupgrade.bin](https://download.openmptcprouter.com/release/v0.60-6.1/ubnt-erx/targets/ramips/mt7621/openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-ramips-mt7621-ubnt_edgerouter-x-squashfs-sysupgrade.bin) (2024/04/09 - 41.75Mo - [sha256sum]

**For appliance** 
**x86-64 (64-bits)**
ext4 image (EFI): [openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined-efi.img.gz](https://download.openmptcprouter.com/release/v0.60-6.1/x86_64/targets/x86/64/openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined-efi.img.gz) (2024/04/09 - 71.83Mo - [sha256sum]

Thanks.

Ysurac, 2024-08-20:
For x86_64, use the snapshot image: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
For the Ubiquiti, I need to try but not sure if I will keep support on this, it's really too slow.

driccee, 2024-08-21:
Thank you @Ysurac 
My appliance is now installed. Thank you very much for your support.

[image]

closed 2024-11-25 by github-actions[bot]