Ysurac/openmptcprouter #3501: Download Speed Problem

IlPirata01, 2024-08-22:
## Expected Behavior
Stable download speed.

## Current Behavior
Hello, I hope you can help me with my issue. I'm currently aggregating 5 internet lines, using ShadowSocks-Rust 2022 as the proxy and Glorytun TCP as the default VPN. Let me explain the behavior more clearly. After configuring everything, the aggregated download speed is good. However, after a certain period of time (I apologize, but I haven't timed it), the download speed drops drastically, reaching a maximum of only 10 Mbit/sec.

At this point, I change the default VPN settings, switching to another one randomly, and the download speed recovers, returning to the levels seen during the initial configuration. I thought it might be one of the lines going down, so I disabled multipath TCP on the interfaces one by one, but the download speed remained low. I also tried changing the proxy, and doing so restored the download speed to the levels of the initial configuration.

Additionally, there are no errors in the logs. Let me know if you need more information, and I hope you can help me resolve this issue.
Thank you

## Specifications

  - OpenMPTCProuter version: 0.60-6.1 r0+24843-acf40c022
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-23-amd64
  - OpenMPTCProuter VPS provider: aruba vps
  - OpenMPTCProuter platform:  x86-64 on Proxmox VM 4 cpu core and 4GB RAM

Ysurac, 2024-08-22:
I would need a screenshot of System->OpenMPTCProuter, status page.

IlPirata01, 2024-08-22:
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_77f2b'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[9252]: hotplug: remove network interface SQM_IFB_77f2b: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice omr-tracker: Launching...
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_5d980'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[9499]: hotplug: add network interface SQM_IFB_5d980: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_5d980'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[9600]: hotplug: remove network interface SQM_IFB_5d980: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_f9c0d'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[9621]: hotplug: add network interface SQM_IFB_f9c0d: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_f9c0d'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[9692]: hotplug: remove network interface SQM_IFB_f9c0d: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='ifb4eth4'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[9721]: hotplug: add network interface ifb4eth4: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_c3f8f'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[9808]: hotplug: add network interface SQM_IFB_c3f8f: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_c3f8f'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[9874]: hotplug: remove network interface SQM_IFB_c3f8f: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_f194e'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10011]: hotplug: add network interface SQM_IFB_f194e: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_f194e'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10084]: hotplug: remove network interface SQM_IFB_f194e: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_3d395'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10105]: hotplug: add network interface SQM_IFB_3d395: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_3d395'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10177]: hotplug: remove network interface SQM_IFB_3d395: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='ifb4tun0'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10206]: hotplug: add network interface ifb4tun0: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='ifb4tun0'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10434]: hotplug: remove network interface ifb4tun0: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_1d1de'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10715]: hotplug: add network interface SQM_IFB_1d1de: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_1d1de'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10835]: hotplug: remove network interface SQM_IFB_1d1de: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_6c77c'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10869]: hotplug: add network interface SQM_IFB_6c77c: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_6c77c'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[10989]: hotplug: remove network interface SQM_IFB_6c77c: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='ifb4tun0'
Aug 22 10:53:16 OpenMPTCProuter daemon.info ModemManager[11018]: hotplug: add network interface ifb4tun0: event processed
Aug 22 10:53:16 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Aug 22 10:53:16 OpenMPTCProuter daemon.err glorytun[8329]: 80.211.121.179.65001: connected
Aug 22 10:53:16 OpenMPTCProuter daemon.err glorytun[8329]: couldn't set socket option `TCP_KEEPCNT'
Aug 22 10:53:16 OpenMPTCProuter daemon.err glorytun[8329]: couldn't set socket option `TCP_KEEPIDLE'
Aug 22 10:53:16 OpenMPTCProuter daemon.err glorytun[8329]: couldn't set socket option `TCP_KEEPINTVL'
Aug 22 10:53:16 OpenMPTCProuter daemon.err glorytun[8329]: couldn't set socket option `TCP_USER_TIMEOUT'
Aug 22 10:53:16 OpenMPTCProuter daemon.err glorytun[8329]: peer wants chacha20
Aug 22 10:53:16 OpenMPTCProuter daemon.info glorytun[8329]: STARTED tun0
Aug 22 10:53:17 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Aug 22 10:53:17 OpenMPTCProuter daemon.notice ttyd[6249]: [2024/08/22 10:53:17:1545] N: rops_handle_POLLIN_netlink: DELADDR
[... cut ...]
Aug 22 11:28:57 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Aug 22 11:28:57 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Aug 22 11:28:57 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Aug 22 13:28:58 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Aug 22 13:28:58 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Aug 22 13:28:58 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses

IlPirata01, 2024-08-22:
I have now switched the proxy to ShadowSocks, but I need to wait a bit to see if the problem reoccurs.

IlPirata01, 2024-08-22:
[image]

IlPirata01, 2024-08-22:
It has been 15 minutes, and the problem has reoccurred.

Ysurac, 2024-08-22:
How do you test speed ?

IlPirata01, 2024-08-22:
Speedtest By Ookla and Nperf

Ysurac, 2024-08-22:
Did you try with another VPS provider ?

IlPirata01, 2024-08-22:
No, I haven’t tried with another VPS provider, but I did try reinstalling the script on the VPS and doing a clean installation of the router, and nothing changed. I forgot to mention one thing: I installed Docker on the VPS, where I ran OpenSpeedTest for some tests, now 
the Docker is currently turned off. Could it be causing the problems?

Ysurac, 2024-08-22:
It can break everything if there is changes in Firewall but shouldn't do anything about perf if this doesn't increase load average.

IlPirata01, 2024-08-22:
I modified the VPS firewall and opened ports 3000 and 3001. Now I'll try restarting the VPS and reinstalling the script, then I'll run a test to see if the issue was caused by opening the ports, but it seems like I've already tried this test.

Ysurac, 2024-08-22:
Insteadn use ```omr-test-speed``` from the router to test speed.

IlPirata01, 2024-08-23:
OMR-test-speed isn't running. I'm reinstalling everything now to see if that solves the problem.

Ysurac, 2024-08-24:
Can you give me the error of omr-test-speed ?

IlPirata01, 2024-08-24:
I reset the VPS and reinstalled the script. I also reset the Proxmox VM, configured the router again, and the problem was resolved.