Ysurac/openmptcprouter #3507: "No Data" on OpenMPTCP Status

jhenkens, 2024-08-28:
## Expected Behavior

When I click Status, it should show the status!

## Current Behavior

It takes a few seconds, then shows "no data". Once, a few hours ago, it showed the interfaces. Nothing has changed since then.

Please see system log [here](https://gist.github.com/jhenkens/2685a4a48cd49c4cb14296335b1b8a59)

It used to reliably show. Not sure what changed. About a month ago I switched to snapshot due to OMR Bypass bugs.

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.61-snapshot-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 6.1
  - OpenMPTCProuter VPS provider: Racknerd
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-09-03:
There is something strange with eth0.8.
Can you give me the result of ```ip a``` ?

Ysurac, 2024-09-03:
What Proxy are you using ?
Can you try to validate again wizard ?

jhenkens, 2024-09-03:
I'm using a private VPS hosted by RackNerd, doing nothing but OpenMPTCPRouter. I have the output of the VPS connecting to a residential fiber network via WireGuard, so that my exit IP is a residential one, not a public one. I'm not using any built in VPN features of OpenMPTCPRouter.

I can confirm, if I disable the `eth0.8` interface from the OpenMPTCPRouter Wizard, that the status page comes back.

In fact, I changed the static IP on `eth0.8` from `192.168.9.2` to `192.168.9.254`, reenabled it via the wizard, and now the status page is working fine. No other changes made.

closed 2024-12-09 by github-actions[bot]