Ysurac/openmptcprouter #3509: I can't find hardware specifications

kurty100, 2024-08-29:
## Expected Behavior
Hi, thanks for your project !
I run some instances of openmptcprouter (0.59.1-5.4) and upgraded 2 in v0.60-6.1 (full reinstall of the Proxmox image and VPS image in debian 12).
The instances have : 1 * Intel(R) Xeon(R) CPU E5-2470 v2 @ 2.40GHz, RAM 512M, HDD 580M
This night I had a crash of an instance in v0.60-6.1, "Out of memory" (see attached), I can't find the requiered hardware specifications on your website or github, could you give me this information or the way to get it on your site ? (I would like to know if 512M is enough)
Thanks in advance.

## Specifications
  - OpenMPTCProuter version: openmptcprouter v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: v0.60-6.1
  - OpenMPTCProuter VPS provider: OVH with 100M limitation
  - OpenMPTCProuter platform: Proxmox

[2024-08-29 08_41_40-P10 - Proxmox Virtual Environment](https://github.com/user-attachments/assets/87eac5b1-afd7-451c-a349-f37eccf8482e)

Ysurac, 2024-08-29:
At least 1024M of RAM I would say.
But it's strange that it crash like that... You have a specific usage during the night ?

kurty100, 2024-08-29:
Thanks for your quick answer !
I have 5 VM on this router (Cumulus Flux Nodes), I don't see many network traffic on them but I'm going to analyse the logs on them.
I had this issue (out of memory) when I installed the 0.59.1-5.4 and I configured 128M RAM, so I put 256M and after 512M, and I never had others crashes. (I have 8 nodes on others routers)
Perhaps like you said in this version I need 1014M. I'm going to let like this and to enable some permanent logging on router if a crash occured again.
I think it could be good to add a section on website about hardware requierment, it could save you some time to answer to questions :-)
Thanks again for your impressive project ! If you have a flux wallet do not hesitate to give it ;-)

[image]

closed 2024-12-02 by github-actions[bot]