Ysurac/openmptcprouter #3514: N100 device

YB1200, 2024-09-02:
I want to achieve a throughput of up to 1 Gbps using OMR, and I see that many people can reach this throughput using an N100 device. Could you recommend the ideal device for me to purchase? There are many options available, and I really want to choose the ideal one.

Ysurac, 2024-09-03:
You can reach near 1Gbps with Raspberry PI 5 also (limited by the port speed).
For N100 device, the CPU should have AES-NI available and ethernet devices using supported chipset.

GermanAizek, 2024-09-04:
> an you recommend an N100 device?

@YB1200 he has an AES-NI, I myself have mini-PC with N100, but I did not use it as an OMRRouter, but LTE station works very well on it, several subscribers can use mobile traffic on highload.