Ysurac/openmptcprouter #3515: VPS script fails with "Cannot write to ‘-’ (Broken pipe)."

raidzx, 2024-09-03:
## Expected Behavior

Run VPS script with this command, everything should install:

wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.6" sh

## Current Behavior

Fails with the following error:
 Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Update repo.openmptcprouter.com key
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
W: The key(s) in the keyring /etc/apt/trusted.gpg.d/xanmod.gpg are ignored as the file has an unsupported filetype.
--2024-09-03 11:19:55--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 169.150.221.147
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|169.150.221.147|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: '/etc/apt/trusted.gpg.d/openmptcprouter.gpg'

/etc/apt/trusted.gpg.d/openmptcprouter.gpg                  100%[========================================================================================================================================>]   1.21K  --.-KB/s    in 0s

2024-09-03 11:19:55 (14.0 MB/s) - '/etc/apt/trusted.gpg.d/openmptcprouter.gpg' saved [1241/1241]

Remove lock and update packages list...
Hit:1 http://security.debian.org/debian-security bookworm-security InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://deb.debian.org/debian bullseye InRelease
Get:5 http://deb.xanmod.org releases InRelease [2091 B]
Hit:6 https://repo.openmptcprouter.com buster InRelease
Err:5 http://deb.xanmod.org releases InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 86F7D09EE734E623
Reading package lists... Done
W: http://security.debian.org/debian-security/dists/bookworm-security/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/xanmod.gpg are ignored as the file has an unsupported filetype.
W: http://deb.debian.org/debian/dists/bookworm/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/xanmod.gpg are ignored as the file has an unsupported filetype.
W: http://deb.debian.org/debian/dists/bookworm-updates/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/xanmod.gpg are ignored as the file has an unsupported filetype.
W: http://deb.debian.org/debian/dists/bullseye/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/xanmod.gpg are ignored as the file has an unsupported filetype.
W: http://deb.xanmod.org/dists/releases/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/xanmod.gpg are ignored as the file has an unsupported filetype.
W: GPG error: http://deb.xanmod.org releases InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 86F7D09EE734E623
E: The repository 'http://deb.xanmod.org releases InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: https://repo.openmptcprouter.com/dists/buster/InRelease: The key(s) in the keyring /etc/apt/trusted.gpg.d/xanmod.gpg are ignored as the file has an unsupported filetype.
-                                                            90%[==========================================================================================================================>              ]  80.00K  --.-KB/s    in 1.3s

Cannot write to ‘-’ (Broken pipe).

## Possible Solution

Not sure 

## Steps to Reproduce the Problem

  1. Take a fresh VPS with Debian 11 or 12
  2. Run VPS install script (I am attempting kernel 6.6 but tried both 6.1 and 6.6
  3. Script fails with "Cannot write to ‘-’ (Broken pipe)."

## Context (Environment)

I am trying to get kernel 6.6 on my VPS

## Specifications

  - OpenMPTCProuter version: 
  - n/a
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: whichever recent script is
  - OpenMPTCProuter platform: 
  - x86_64
  - Country: 
- US

Ysurac, 2024-09-03:
Kernel >6.1 is only supported on snapshot script for now.

raidzx, 2024-09-03:
Hrm,

no matter which I try from that link they all fail the same way: https://pastebin.com/6SAsq6EA

Ysurac, 2024-09-04:
It's due to normal script, you have to do a ```rm /etc/apt/sources.list.d/xanmod*``` and run snapshot script again

raidzx, 2024-12-03:
I never responded to this but wanted to let you know that your comment fixed my issue, thanks @Ysurac ! Closing now.

closed 2024-12-03 by raidzx

ovehbe, 2025-01-13:
I also have this issue and I'm already using snapshot command.

wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="6.6" sh