Ysurac/openmptcprouter #3516: OpenMPTCPRouter 6.7 building failed on 'master' branch

GermanAizek, 2024-09-04:
```
Patch failed!  Please fix /home/debian/GIT/openmptcprouter/x86_64/6.7/source/target/linux/generic/pending-6.7/727-net-phy-realtek-use-inline-functions-for-10GbE-adver.patch!
make[3]: *** [Makefile:123: /home/debian/GIT/openmptcprouter/x86_64/6.7/source/build_dir/toolchain-x86_64_gcc-13.2.0_musl/linux-github_com_multipath_tcp_mptcp_net_next_git-3d1856db1e4d551bffe5987f57b84b62e245689b/.prepared] Error 1
make[3]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#14.35#13.25#19.27
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:93: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source'
make[1]: *** [toolchain/Makefile:89: /home/debian/GIT/openmptcprouter/x86_64/6.7/source/staging_dir/toolchain-x86_64_gcc-13.2.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.7/source'
make: *** [/home/debian/GIT/openmptcprouter/x86_64/6.7/source/include/toplevel.mk:233: world] Error 2
```

Ysurac, 2024-09-04:
kernel 6.7 is not supported.

GermanAizek, 2024-09-04:
> kernel 6.7 is not supported.

ok building 6.10 from develop branch?

Ysurac, 2024-09-04:
No, it's still in dev. Not sure if I will keep 6.10 or switch to 6.11.

closed 2024-12-09 by github-actions[bot]