Ysurac/openmptcprouter #3517: Speed not aggregated

Yessine1200, 2024-09-04:
## Expected Behavior
Have a full aggregation.

## Current Behavior
I installed OMR on a miniPC (13th Gen Intel® Core™ i7-13700H: 20 cores + AES-NI support), booting from a USB key. The miniPC has two Ethernet interfaces—one for the configuration machine and the other for links using macVLAN on the settings tab. I have three connections: Fiber, 5G SFR, and 5G ORANGE. When I run om-test-speed, I should be seeing an aggregated speed of at least 80M (over 800Mbps) with these three connections, but I'm only getting 50M. However, when I aggregate the two 5G connections, I get full speed. When I aggregate Fiber and one 5G connection, I don't get full speed.
For your information, the speed test results for the 3 connections when running omr-test-speed wanx are:
Fiber: 65-75M
5G Orange: 25-35M
5G SFR: 20-25M

NB: When doing aggregation, the load avreage always under 1.
All green on the status page:
[image]

## Specifications

  - OpenMPTCProuter version: 6.1 6.6 snapshot
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-09-04:
What speed do you get on VPS side ? You can use ```omr-test-speed``` also on VPS.
You should also put the fiber connection as master.

Yessine1200, 2024-09-04:
On the VPS side, I got 270M, which is more than enough to aggregate the 3 connections. 
I set it as the master, but the issue remains the same.

Ysurac, 2024-09-04:
What do you get with only fiber ?

Yessine1200, 2024-09-04:
[image]

Yessine1200, 2024-09-04:
"What could be the problem? Could it be because I'm booting from a USB key or because I'm using macVLAN?

Yessine1200, 2024-09-04:
`omr-test-speed `with only fiber connecttion:
[image]
`omr-test-speed `with all connections:
[image]
Yes, aggregation works, but with reduced speed on the 5 connections.

Ysurac, 2024-09-04:
What is the proxy you use ? In System->OpenMPTCProuter, Wizard tab, advanced settings checkbox ?
Default is Shadowsocks-Rust. Check also that AES is correctly enabled.

Yessine1200, 2024-09-04:
Could it be a problem because I'm using macVLAN?

Yessine1200, 2024-09-05:
How can I know that ? With lspci ?

Ysurac, 2024-09-05:
yes it's a solution. ```opkg update && opkg install pciutils``` and then  ```lspci -v```. We will also get the driver used.

Yessine1200, 2024-09-05:
Here are the two Ethernet interfaces:
56:00.0 Ethernet controller: Intel Corporation Ethernet Controller I225-V (rev 03)
	Subsystem: Intel Corporation Device 0000
	Flags: bus master, fast devsel, latency 0, IRQ 19
	Memory at 6c400000 (32-bit, non-prefetchable) [size=1M]
	Memory at 6c500000 (32-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
	Capabilities: [70] MSI-X: Enable+ Count=5 Masked-
	Capabilities: [a0] Express Endpoint, IntMsgNum 0
	Kernel driver in use: igc

57:00.0 Ethernet controller: Intel Corporation Ethernet Controller I225-V (rev 03)
	Subsystem: Intel Corporation Device 0000
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at 6c200000 (32-bit, non-prefetchable) [size=1M]
	Memory at 6c300000 (32-bit, non-prefetchable) [size=16K]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/1 Maskable+ 64bit+
	Capabilities: [70] MSI-X: Enable+ Count=5 Masked-
	Capabilities: [a0] Express Endpoint, IntMsgNum 0
	Kernel driver in use: igc

Ysurac, 2024-09-05:
I225-V and I226-V have many hardware issues (there is already some topics here, maybe you can check and this may help).

closed 2024-12-09 by github-actions[bot]