Ysurac/openmptcprouter #3518: NAT Loopback

FPi-sh, 2024-09-04:
## Expected Behavior
Reaching exposed services on public ip from LAN should work.

## Current Behavior
Connection is reset

## Possible Solution
policy based routes?

## Steps to Reproduce the Problem

  1. Expose a service
  2. From internet it works, from lan does not work

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.60
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country:

Ysurac, 2024-09-04:
Can you check on your destination if you have traffic using tcpdump ?

FPi-sh, 2024-09-04:
Here's my network topology:

OMR LAN (192.168.101.1/24) <--> LinuxVM (192.168.101.200/24), PC (192.168.101.148/24)
Port forwarding: OMR.PUBL.IP:8081 to 192.168.101.200:80 (where there's httpd listening)

When I try curl from an host inside my lan, I don't see any request reaching server's interface (tcpdump is empty).

From OMR on eth0 I see this:

```
root@OpenMPTCProuter:~# tcpdump -ni eth0 port 8081
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
21:59:41.046713 IP 192.168.101.148.47800 > OMR.PUBL.IP.8081: Flags [S], seq 1595500041, win 65535, options [mss 1460,sackOK,TS val 4245108839 ecr 0,nop,wscale 9], length 0
21:59:41.046877 IP OMR.PUBL.IP.8081 > 192.168.101.148.47800: Flags [S.], seq 21635800, ack 1595500042, win 43440, options [mss 1460,sackOK,TS val 3999143332 ecr 4245108839,nop,wscale 9], length 0
21:59:41.053178 IP 192.168.101.148.47800 > OMR.PUBL.IP.8081: Flags [.], ack 1, win 128, options [nop,nop,TS val 4245108863 ecr 3999143332], length 0
21:59:41.053178 IP 192.168.101.148.47800 > OMR.PUBL.IP.8081: Flags [P.], seq 1:374, ack 1, win 128, options [nop,nop,TS val 4245108863 ecr 3999143332], length 373
21:59:41.053369 IP OMR.PUBL.IP.8081 > 192.168.101.148.47800: Flags [.], ack 374, win 85, options [nop,nop,TS val 3999143338 ecr 4245108863], length 0
21:59:41.123226 IP OMR.PUBL.IP.8081 > 192.168.101.148.47800: Flags [F.], seq 1, ack 374, win 85, options [nop,nop,TS val 3999143408 ecr 4245108863], length 0
21:59:41.129841 IP 192.168.101.148.47800 > OMR.PUBL.IP.8081: Flags [F.], seq 374, ack 2, win 128, options [nop,nop,TS val 4245108940 ecr 3999143408], length 0
21:59:41.129990 IP OMR.PUBL.IP.8081 > 192.168.101.148.47800: Flags [.], ack 375, win 85, options [nop,nop,TS val 3999143415 ecr 4245108940], length 0
21:59:41.131399 IP 192.168.101.148.47804 > OMR.PUBL.IP.8081: Flags [S], seq 2233263747, win 65535, options [mss 1460,sackOK,TS val 4245108941 ecr 0,nop,wscale 9], length 0
21:59:41.131576 IP OMR.PUBL.IP.8081 > 192.168.101.148.47804: Flags [S.], seq 2296785639, ack 2233263748, win 43440, options [mss 1460,sackOK,TS val 3999143416 ecr 4245108941,nop,wscale 9], length 0
21:59:41.134651 IP 192.168.101.148.47804 > OMR.PUBL.IP.8081: Flags [.], ack 1, win 128, options [nop,nop,TS val 4245108946 ecr 3999143416], length 0
21:59:41.134651 IP 192.168.101.148.47804 > OMR.PUBL.IP.8081: Flags [P.], seq 1:374, ack 1, win 128, options [nop,nop,TS val 4245108946 ecr 3999143416], length 373
21:59:41.134877 IP OMR.PUBL.IP.8081 > 192.168.101.148.47804: Flags [.], ack 374, win 85, options [nop,nop,TS val 3999143420 ecr 4245108946], length 0
21:59:41.197145 IP OMR.PUBL.IP.8081 > 192.168.101.148.47804: Flags [F.], seq 1, ack 374, win 85, options [nop,nop,TS val 3999143482 ecr 4245108946], length 0
21:59:41.203240 IP 192.168.101.148.47804 > OMR.PUBL.IP.8081: Flags [.], ack 2, win 128, options [nop,nop,TS val 4245109014 ecr 3999143482], length 0
21:59:41.203240 IP 192.168.101.148.47804 > OMR.PUBL.IP.8081: Flags [F.], seq 374, ack 2, win 128, options [nop,nop,TS val 4245109014 ecr 3999143482], length 0
21:59:41.203477 IP OMR.PUBL.IP.8081 > 192.168.101.148.47804: Flags [.], ack 375, win 85, options [nop,nop,TS val 3999143488 ecr 4245109014], length 0
21:59:41.205896 IP 192.168.101.148.47808 > OMR.PUBL.IP.8081: Flags [S], seq 4136874738, win 65535, options [mss 1460,sackOK,TS val 4245109015 ecr 0,nop,wscale 9], length 0
21:59:41.206110 IP OMR.PUBL.IP.8081 > 192.168.101.148.47808: Flags [S.], seq 1817016729, ack 4136874739, win 43440, options [mss 1460,sackOK,TS val 3999143491 ecr 4245109015,nop,wscale 9], length 0
21:59:41.209928 IP 192.168.101.148.47808 > OMR.PUBL.IP.8081: Flags [.], ack 1, win 128, options [nop,nop,TS val 4245109020 ecr 3999143491], length 0
21:59:41.209928 IP 192.168.101.148.47808 > OMR.PUBL.IP.8081: Flags [P.], seq 1:374, ack 1, win 128, options [nop,nop,TS val 4245109021 ecr 3999143491], length 373
21:59:41.210200 IP OMR.PUBL.IP.8081 > 192.168.101.148.47808: Flags [.], ack 374, win 85, options [nop,nop,TS val 3999143495 ecr 4245109021], length 0
21:59:41.276113 IP OMR.PUBL.IP.8081 > 192.168.101.148.47808: Flags [F.], seq 1, ack 374, win 85, options [nop,nop,TS val 3999143561 ecr 4245109021], length 0
21:59:41.283210 IP 192.168.101.148.47808 > OMR.PUBL.IP.8081: Flags [.], ack 2, win 128, options [nop,nop,TS val 4245109094 ecr 3999143561], length 0
21:59:41.283210 IP 192.168.101.148.47808 > OMR.PUBL.IP.8081: Flags [F.], seq 374, ack 2, win 128, options [nop,nop,TS val 4245109094 ecr 3999143561], length 0
21:59:41.283457 IP OMR.PUBL.IP.8081 > 192.168.101.148.47808: Flags [.], ack 375, win 85, options [nop,nop,TS val 3999143568 ecr 4245109094], length 0
21:59:41.286165 IP 192.168.101.148.47814 > OMR.PUBL.IP.8081: Flags [S], seq 3819683432, win 65535, options [mss 1460,sackOK,TS val 4245109096 ecr 0,nop,wscale 9], length 0
21:59:41.286449 IP OMR.PUBL.IP.8081 > 192.168.101.148.47814: Flags [S.], seq 203606993, ack 3819683433, win 43440, options [mss 1460,sackOK,TS val 3999143571 ecr 4245109096,nop,wscale 9], length 0
21:59:41.290273 IP 192.168.101.148.47814 > OMR.PUBL.IP.8081: Flags [.], ack 1, win 128, options [nop,nop,TS val 4245109101 ecr 3999143571], length 0
21:59:41.290273 IP 192.168.101.148.47814 > OMR.PUBL.IP.8081: Flags [P.], seq 1:374, ack 1, win 128, options [nop,nop,TS val 4245109101 ecr 3999143571], length 373
[... cut ...]
21:59:41.839360 IP OMR.PUBL.IP.8081 > 192.168.101.148.47862: Flags [F.], seq 1, ack 374, win 85, options [nop,nop,TS val 3999144124 ecr 4245109586], length 0
21:59:41.847197 IP 192.168.101.148.47862 > OMR.PUBL.IP.8081: Flags [.], ack 2, win 128, options [nop,nop,TS val 4245109658 ecr 3999144124], length 0
21:59:41.847197 IP 192.168.101.148.47862 > OMR.PUBL.IP.8081: Flags [F.], seq 374, ack 2, win 128, options [nop,nop,TS val 4245109658 ecr 3999144124], length 0
21:59:41.847388 IP OMR.PUBL.IP.8081 > 192.168.101.148.47862: Flags [.], ack 375, win 85, options [nop,nop,TS val 3999144132 ecr 4245109658], length 0
```

closed 2024-12-09 by github-actions[bot]