Ysurac/openmptcprouter #3519: Port forwarding from WAN zone

FPi-sh, 2024-09-04:
## Expected Behavior
I should reach services exposed in the wan zone.

## Current Behavior
The connection enters from wan1, forwards to the device exposed in lan, the device reply but the reply goes on the default route intead of going to the wan1.

## Possible Solution
Could be managed with packet/connection marking with mangle?

## Steps to Reproduce the Problem

  1. Expose a service selecting wan zone
  2. Connection times out

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.60
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64
  - Country:

Ysurac, 2024-09-04:
You can try to apply a firewall mark on the rule. You can find all available bypass mark via SSH with ```uci show firewall```

AndreaLeidi, 2024-09-16:
I won't call it "a bug", for what you ask the safe approach is to put the server/services in a (DMZ) subnet downstream the OMR router.

A quick-and-dirty solution (at the cost of a little more broadcast traffic around) is to set WANx router address as gateway on the server you are going to expose, as well as set the server IP in the same subnet.
In other words:
OMR: 192.168.100.1/24
OMR WAN1: 192.168.1.2/24 GW 192.168.1.1
SERVER: 192.168.1.3/24 GW 192.168.1.1

I did it with my provider's branded VoIP adapter, which must necessarily link to a specific DSL line to work.

This will bypass OMR (and aggregation) and link your device to the WAN1 router. 
The server will remain reachable OMR LAN side, no need to set any route.
If you still need OMR bandwidth aggregation and failover for your server, containerize your service apart and set its networking, or add a 2nd network interface / set a 2nd virtual network device and bind your service on it.

closed 2024-12-21 by github-actions[bot]