Ysurac/openmptcprouter #3521: Limited traffic on month for OMR-VPS - it bad idea?

GermanAizek, 2024-09-04:
@Ysurac 
How much traffic is transferred both in idle time between OMR-VPS and OMR-Router, and with net transfer 1 GB, how much traffic will actually be spent. During obfuscation, packets may be divided and size will be larger. Am I right or is it technically It's done differently? How much is 1, 10, 20, 30 TB per month enough on average?

Ysurac, 2024-09-04:
No idea and not a lot of packets. Only ping for VPN side, http request for Proxy side and some HTTPS data to reach the API when needed. Far less than 1GB/month.

GermanAizek, 2024-09-04:
> No idea and not a lot of packets. Only ping for VPN side, http request for Proxy side and some HTTPS data to reach the API when needed. Far less than 1GB/month.

Minimum RAM requirements is 1 GB, is this relevant with new versions of linux kernels?

Ysurac, 2024-09-04:
Yes, still 1GB.