Ysurac/openmptcprouter #3527: Build 6.6 in develop is failed

GermanAizek, 2024-09-10:
```
...
Making install in projects
make[7]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools/projects'
make[8]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools/projects'
make[8]: Nothing to be done for 'install-exec-am'.
/usr/bin/mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects
/usr/bin/mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/TP
/usr/bin/mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/KDE
/usr/bin/mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/GNOME
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c ./team-address /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/team-address
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c ./TP/trigger /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/TP/trigger
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c ./TP/team-address /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/TP/team-address
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c ./KDE/trigger /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/KDE/trigger
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c ./KDE/team-address /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/KDE/team-address
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c ./GNOME/trigger /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/GNOME/trigger
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c ./GNOME/team-address /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/GNOME/team-address
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c -m 644 ./index /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/index
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c -m 644 ./TP/teams.url /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/TP/teams.url
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c -m 644 ./TP/teams.html /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/TP/teams.html
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c -m 644 ./KDE/teams.url /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/KDE/teams.url
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c -m 644 ./KDE/teams.html /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/KDE/teams.html
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c -m 644 ./GNOME/teams.url /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/GNOME/teams.url
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c -m 644 ./GNOME/teams.html /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/share/gettext/projects/GNOME/teams.html
make[8]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools/projects'
make[7]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools/projects'
Making install in src
make[7]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools/src'
make  install-am
make[8]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools/src'
make[9]: Entering directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools/src'
 /usr/bin/mkdir -p '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin'
  /usr/bin/env bash ../libtool   --mode=install /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgcmp msgfmt msgmerge msgunfmt xgettext msgattrib msgcat msgcomm msgconv msgen msgexec msgfilter msggrep msginit msguniq recode-sr-latin '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin'
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgcmp /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgcmp
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgfmt /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgfmt
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgmerge /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgmerge
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgunfmt /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgunfmt
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c xgettext /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/xgettext
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgattrib /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgattrib
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgcat /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgcat
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgcomm /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgcomm
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgconv /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgconv
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgen /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgen
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgexec /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgexec
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msgfilter /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msgfilter
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msggrep /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msggrep
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msginit /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msginit
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c msguniq /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/msguniq
OpenWrt-libtool: install: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c recode-sr-latin /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/bin/recode-sr-latin
/usr/bin/mkdir -p /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/lib/gettext
/home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/host/bin/install -c -m 644 msgfmt.net.exe /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/hostpkg/lib/gettext/msgfmt.net.exe
[... cut ...]
make[3]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#985.48#326.10#1758.50
make[2]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source'
make[1]: *** [package/Makefile:171: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source'
make: *** [/home/debian/GIT/openmptcprouter/x86_64/6.6/source/include/toplevel.mk:248: world] Error 2
```

Ysurac, 2024-09-10:
Please use template.
I see no errors here. Try again.

GermanAizek, 2024-09-10:
> I see no errors here. Try again.
```
/usr/bin//opt -O2 -mtriple=bpfel < /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/mptcp-bpf-burst/mptcp_bpf_burst.bc > /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/mptcp-bpf-burst/mptcp_bpf_burst.opt
bash: line 1: /usr/bin//opt: No such file or directory
make[3]: *** [Makefile:53: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/build_dir/target-x86_64_musl/linux-x86_64/mptcp-bpf-burst/.built] Error 127
make[3]: Leaving directory '/home/debian/GIT/openmptcprouter/feeds/openmptcprouter/mptcp-bpf-burst'
time: package/feeds/openmptcprouter/mptcp-bpf-burst/compile#0.22#0.25#0.53
    ERROR: package/feeds/openmptcprouter/mptcp-bpf-burst failed to build.
make[2]: *** [package/Makefile:177: package/feeds/openmptcprouter/mptcp-bpf-burst/compile] Error 1
make[2]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source'
make[1]: *** [package/Makefile:171: /home/debian/GIT/openmptcprouter/x86_64/6.6/source/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian/GIT/openmptcprouter/x86_64/6.6/source'
make: *** [/home/debian/GIT/openmptcprouter/x86_64/6.6/source/include/toplevel.mk:248: world] Error 2

```

same thing

I think its error line:

I have Debian 12.

```
bash: line 1: /usr/bin//opt: No such file or directory
```

Ysurac, 2024-09-10:
It's not the same. Here you are missing llvm (a typo in the doc is now fixed)

closed 2024-09-15 by GermanAizek