Ysurac/openmptcprouter #3531: DNS issue: can't resolve hostname after reboot

max93776, 2024-09-12:
## Expected Behavior
Addresses can be resolved by both network interfaces. As soon as one connection breaks, the other is used

## Current Behavior
Two problems occur: suddenly no addresses can be resolved after reboot and my modem is marked as red even though it has an internet connection. 
The FM modem is set up again using a script every time omr is restarted, and a static IP is assigned to the interface. DNS didn't work over this connection without rebooting the vm another time (the modem only gets a new ip if the vps is rebooted). The server 8.8.8.8 could be pinged, but addresses were not resolved. After reboot the address resolution worked. Now there was another reboot and suddenly the address resolution didn't work, only directly via dig @8.8.8.8 ... the resolution worked without a reboot. The modem is still marked as red even though there is internet connectivity.
But the strange thing is that I can connect to the OMR via SSH via the open tunnel address...

## Specifications

  - OpenMPTCProuter version: 0.60_6.1
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-20-amd64
  - OpenMPTCProuter VPS provider: seeed
  - OpenMPTCProuter platform: amd64

[image]
Log:
Sep 12 15:41:41 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.36.1
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.89 cachesize 1000
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5053
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5054
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sep 12 15:41:41 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 0 names
Sep 12 13:41:41 OpenMPTCProuter authpriv.info dropbear[4124]: Not backgrounding
Sep 12 13:41:41 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sep 12 13:41:41 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Sep 12 13:41:41 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Sep 12 13:41:41 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Sep 12 13:41:41 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Sep 12 13:41:41 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Sep 12 13:41:41 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S19firewall: Failed to parse json data: unexpected character
Sep 12 13:41:41 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S25packet_steering: uci: Entry not found
Sep 12 13:41:41 OpenMPTCProuter user.notice : Added device handler type: bonding
Sep 12 13:41:41 OpenMPTCProuter user.notice : Added device handler type: 8021ad
Sep 12 13:41:41 OpenMPTCProuter user.notice : Added device handler type: 8021q
Sep 12 13:41:41 OpenMPTCProuter user.notice : Added device handler type: macvlan
Sep 12 13:41:41 OpenMPTCProuter user.notice : Added device handler type: veth
Sep 12 13:41:41 OpenMPTCProuter user.notice : Added device handler type: bridge
Sep 12 13:41:41 OpenMPTCProuter user.notice : Added device handler type: Network device
Sep 12 13:41:41 OpenMPTCProuter user.notice : Added device handler type: tunnel
Sep 12 13:41:41 OpenMPTCProuter daemon.notice wpa_supplicant[4534]: Successfully initialized wpa_supplicant
Sep 12 13:41:41 OpenMPTCProuter daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Sep 12 13:41:41 OpenMPTCProuter daemon.notice netifd: Interface 'FM350GL' is enabled
Sep 12 13:41:41 OpenMPTCProuter daemon.notice netifd: Interface 'FM350GL' is setting up now
Sep 12 13:41:41 OpenMPTCProuter daemon.notice netifd: Interface 'FM350GL' is now up
Sep 12 13:41:41 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_d66fa'
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is enabled
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is setting up now
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'loopback' is now up
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is enabled
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is setting up now
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'wan2' is now up
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Network device 'eth2' link is up
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Interface 'FM350GL' has link connectivity 
Sep 12 13:41:42 OpenMPTCProuter daemon.notice netifd: Network device 'lo' link is up
[... 394 lines cut ...]
Sep 12 14:33:02 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:33:15 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:33:21 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:33:33 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:33:39 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:33:52 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:34:10 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:34:28 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:34:34 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:34:47 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:35:05 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:35:23 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:35:29 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:35:41 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:35:47 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:36:00 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:36:18 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
[... cut ...]
Sep 12 14:37:01 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:37:14 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:37:32 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:37:50 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Sep 12 14:37:56 OpenMPTCProuter daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)

Ysurac, 2024-09-12:
You can try to increase "Max. concurrent queries" to 1500 in Network->DHCP & DNS, "Advanced settings" tab.

max93776, 2024-09-12:
> You can try to increase "Max. concurrent queries" to 1500 in Network->DHCP & DNS, "Advanced settings" tab.

I rebooted the vps and now the DNS error messages in the logs disappeared + the web ui says that omr is running correctly but the UI gives me the error message "can't get public IP address from shadowsock" in the vps. FM modem is still marked with a red cross

Ysurac, 2024-09-12:
There is a problem on gateway of FM modem. You are using ModemManager or something else for it ?

Ysurac, 2024-09-12:
hum yes I remember this modem... Seems here you are not getting gateway correctly.
To fix shadowsocks issue, you can try to update to v0.61rc1, I fixed some issues and this is maybe one of them...

max93776, 2024-09-12:
Thank you I will give it a try next week. Do you have any idea how I could solve the gateway issue? I remember one time when I used only the modem, I changed the gateway address to the modem IP x.x.x.1, then the web ui broke down because of I guess the wrong gateway, but when I turned it back to 0.0.0.0 the red cross was replaced by the green hook and it seemed to work fine for a while (until the next reboot)

Ysurac, 2024-09-12:
0.0.0.0 seems strange (but as you get an IP seems to be the case).
Can you try a ```ping -B -I eth2 1.1.1.1``` to check if some traffic is possible ?

max93776, 2024-09-17:
The ping to 1.1.1.1 was successful and when I changed the gateway address to 1.1.1.1 it worked for 3 minutes. Now the gateway is down again:
[image]

Ysurac, 2024-09-17:
The gateway can't be 1.1.1.1. You need to find the correct gateway, at commands can't give you that ?

max93776, 2024-09-17:
No, the command specified for that is AT+CGCONTRDP which gives me only 2 DNS addresses. There is also a big thread opened [here](https://forum.openwrt.org/t/fibocom-fm350-gl-support/142682/25). Some say that you have to set the gateway to ip x.x.x.1, others say that no gateway is necessary. Both variants only work half the time, as it is still displayed as incorrect in the OMR.

Ysurac, 2024-09-17:
You can try the .1, but it's strange for me to use any random IP...
For the problem on server side, you need to fix time on both VPS and router (I would say it's on router, check the timezone if you set time manually)

max93776, 2024-09-24:
> In the GUI (Luci) you can set your default gateway as 0.0.0.0/0, there is no need to use any other address. For the interface itself you can use any netmask, /32 works fine.
> From the console you can achieve the same with ip r add default dev eth3 (or whatever device name you have).

> OMR seems to conflict with gw address 0.0.0.0. Can you explain in more detail why exactly 0.0.0.0?

> This is the way to say "I want to send traffic to the interface (or device) and not to some IP address since my connection is just a point-to-point link".
OMR should probably mimic what is used in OpenVPN for ages:
--redirect-gateway flags...
def1 -- Use this flag to override the default gateway by using 0.0.0.0/1 and 128.0.0.0/1 rather than 0.0.0.0/0

That's what I got from a conversation in the forum. 

I was able to solve the shadowsock problem by updating the omr, but unfortunately I apparently broke the vps with an update/upgrade (omr vm is also gone)... I will tackle the problem again when I have the vps+omr reinstalled...

max93776, 2024-09-25:
It works now with GW 0.0.0.0 and netmask 255.255.255.255 . I guess flashing the new image was a good idea.
[image]
[image]
Thank you for your time and effort!

closed 2024-09-25 by max93776