Ysurac/openmptcprouter #3536: upgrade for 0.61rc1 router not working

bobradamus, 2024-09-15:
## Expected Behavior
I downloaded the upgrade image from the link provided to upgrade to the 0.61rc1 version, here: https://snapshots.openmptcprouter.com/6.6/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.61rc1-6.6-r0+27346-c7ba5574f5-bcm27xx-bcm2711-rpi-4-ext4-sysupgrade.img.gz
I installed it on the web interface to upgrade the router after backing up my settings and ran the script to update the VPS

## Current Behavior
The VPS appears to have been upgraded sucessfully.
The router still reports being on the previous stable version and does not show the 0.61rc1 version anywhere.

## Possible Solution
n/a

## Steps to Reproduce the Problem
as above

## Context (Environment)
Raspberry Pi4 and IONOS VPS.

## Specifications
n/a

Ysurac, 2024-09-16:
Did you try to update again ?

bobradamus, 2024-09-16:
Yes, I’ve tried three times and rebooted between.

kevinh-csalabs, 2024-09-16:
I got a key warning on debian 12 and had to resolve that before it would properly upgrade
`sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 86F7D09EE734E623`
That said there are definitely some other problems with this build. Caveat emptor!

closed 2024-12-21 by github-actions[bot]