Ysurac/openmptcprouter #3538: multipath seems to be blocked after new install

talitha77, 2024-09-16:
## Expected Behavior
green flags all round on older OMR and VPS realease

## Current Behavior
multipath seems to be blocked message on all three starlinks after new install both on local rpi5 and vps

## Specifications

  - OpenMPTCProuter version: v0.60-6.1 r0+24843-acf40c022e
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-22-amd64

  - OpenMPTCProuter VPS provider: VPSSERVER.com
  - OpenMPTCProuter platform:RPI5

[image]
[image]
using 5.9 with my older vps and all ok.. three starlinks green and happy.
I have a new rpi5 with latest OMR and new vps with 6.1.0-22-amd64 # Debian 6.1.94-1 (2024-06-21) x86_64 GNU/Linux
and now all three starlinks say that mptcp appears to be blocked. If i swap to the older vps that message goes away on one or two of the wans (of course i get a kernel mismatch)..
note that mptctp testing is not there on latest release unless its moved. on the older OMR all starlinks report OK mptcp

Ysurac, 2024-09-16:
Aggregation is working or not ?

talitha77, 2024-09-16:
I'm not sure..  speed of starlink fluctuates .. is there a way to test?
I'm getting quite a few  OpenMPTCProuter daemon.err /usr/bin/ss-redir[18582]: remote recv: Socket not connected

Ysurac, 2024-09-16:
Check in Network->MPTCP, bandwitdh tab, and check if more than one WAN is used during a speedtest.

talitha77, 2024-09-16:
[image]

talitha77, 2024-09-16:
the problem with being on a ship is that each WAN is subject to outages. So i dont know which to assign as mptcp master. if WAN1 is master and then goes down, i lose connection for several seconds. I have assigned WAN2 as master as its less flaky.

Ysurac, 2024-09-16:
Can you test latest v0.61 rc2 ? https://github.com/Ysurac/openmptcprouter/issues/3528

talitha77, 2024-09-16:
[image]
nothing works at all now.. no eth1 - eth4, so guess wan1 wan2.. however cannot save and apply changes at all

talitha77, 2024-09-16:
system seems to be unwriteable

talitha77, 2024-09-16:
i will have to put the old one back. i will try to work on the rpi5, perhaps using vlans instead of ethernet adapters..  does omr have the ability to configure vlans?

talitha77, 2024-09-16:
[image]
old one back in and running.. however it chokes when i put the network online.. i assumed a pi4 was too slow

Ysurac, 2024-09-16:
The filesystem readonly is due to corrupted FS or bad SD card.

talitha77, 2024-09-16:
ah ok.. i guess i will try again with the old image. at the moment it is working so i'll take a rest

talitha77, 2024-09-18:
tried again with different card, shut down properly, new install.. still says filesystem read only

Ysurac, 2024-09-18:
What is the filesystem you choose ? ext4 or squashfs ?

talitha77, 2024-09-18:
ext4..  am trying again with 6.0.

talitha77, 2024-09-21:
I have tried everything i can think of.. now i'm not even getting aggregation. I have a suspicion three starlinks is not a good mix.. initially they all had 192.168.1.x subnets, and things were sort of ok. I changed each starlink routers subnet to 192.168.1.x , 192.168.2,x and 192.168.3.x just to keep them organised but things have got worse.. If its not something particular to starlink then i''m doing something wrong. But i'm just setting up with defaults

talitha77, 2024-09-22:
i am wondering if its connected with which WAN is set as multipath master. of the three starlinks all have long periods of outage, so i never know which is going to be the reliable connection. thats why we need the OMR. But i cant keep changing which one is master.

Ysurac, 2024-09-22:
Master is only used for initial connection. It's the default gateway and OMR-Tracker will change this when it detect a connection failure (your can change settings if you want in the UI).
But you must not have a read only file system. SquashFS write less data on flash so should have less issues on flash/SD card.

talitha77, 2024-09-23:
read only system was only a problem on 0.61rc..  maybe due to multiple USB ethernet adaptors. I back on the 0.59 which used to work fine when i had all three starlinks on 192.168.1.x subnets, but i was advised to change them. I guess that was a mistake. do you think that could be right?
I want to get it working smoothly before i try 0.61 again with vlans (if i can!)
There is no ipv4 gateway on the LAN connection (eth0).. but i had to change the OMR ip to 192.168.101.1 as starlink address is 192.168.100.1, but that shouldn't present a problem..

yannick2801, 2024-09-23:
you can try this to repair your FS : 

_X = 2 for me._ 

```
umount /dev/mmcblk2pX
e2fsck -f /dev/mmcblk2pX
```

darkman1983, 2024-09-25:
Good to see i'm not the only one with this problem.

I add another one here:
If i assume the problem is really there and set the 5G connection to MPTCP over VPN (Wireguard), then it keeps glorytun getting disconnected every 1-3 mins, restarting it like 30 times with keyexchange error.

### Secifications
OpenMPTCProuter version: v0.61rc2-6.6 r0+27346-c7ba5574f5

OpenMPTCProuter VPS version: 0.1031-test 6.10.2-x64v3-xanmod1

OpenMPTCProuter VPS provider: 1blu.de

OpenMPTCProuter platform: x86_64 docker

[image]

talitha77, 2024-09-26:
> you can try this to repair your FS :
> 
> _X = 2 for me._
> 
> ```
> umount /dev/mmcblk2pX
> e2fsck -f /dev/mmcblk2pX
> reboot
> ```

unfortunately it didnt work..
i have tried three different rpis and all seem to have this issue on the latest version.  I'm happy to stay on the older release if i can just get it working again

talitha77, 2024-09-26:
OK so i finally got 6.1.0-22 with squash-fs..  2 out of the three starlinks are giving me the blocked message.. which i still don't get.. but it looks like i'm aggregating ..  i need to test by killing the main starlink and seeing if it carries on..
[image]
[image]

talitha77, 2024-09-26:
OK it didn't drop a ping this time..  can it really be as simple as that? 
[image]
[image]

Ysurac, 2024-09-26:
The "Multipath seems to be blocked ont he connection" is only a warning. I need to find a better way to check if multipath is really blocked or not, and why it's sometimes working or not...
For now it seems in most case related to CGNAT.

talitha77, 2024-09-26:
aggregation is working but service is very patchy..Sep 26 10:59:53 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Sep 26 10:59:53 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Sep 26 10:59:53 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Sep 26 10:59:53 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Sep 26 10:59:53 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='tun0'
Sep 26 10:59:53 OpenMPTCProuter daemon.info ModemManager[17389]: hotplug: add network interface tun0: event processed
Sep 26 10:59:54 OpenMPTCProuter daemon.err openvpn(omr)[17349]: Connection reset, restarting [0]
Sep 26 10:59:54 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.3 255.255.255.0 init
Sep 26 10:59:54 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: net_addr_v4_del: 10.255.252.3 dev tun0
Sep 26 10:59:54 OpenMPTCProuter daemon.notice ttyd[13708]: [2024/09/26 10:59:54:4271] N: rops_handle_POLLIN_netlink: DELADDR
Sep 26 10:59:54 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down
Sep 26 10:59:54 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sep 26 10:59:54 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Sep 26 10:59:54 OpenMPTCProuter daemon.notice ttyd[13708]: [2024/09/26 10:59:54:4281] N: rops_handle_POLLIN_netlink: DELADDR
Sep 26 10:59:54 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='tun0'
Sep 26 10:59:54 OpenMPTCProuter daemon.info ModemManager[17519]: hotplug: remove network interface tun0: event processed
Sep 26 10:59:54 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Sep 26 10:59:54 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.3 255.255.255.0 init
Sep 26 10:59:54 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: SIGUSR1[soft,connection-reset] received, process restarting
Sep 26 10:59:55 OpenMPTCProuter daemon.warn openvpn(omr)[17349]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Sep 26 10:59:55 OpenMPTCProuter daemon.warn openvpn(omr)[17349]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sep 26 10:59:55 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: TCP/UDP: Preserving recently used remote address: [AF_INET]185.127.19.147:65301
Sep 26 10:59:55 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: Attempting to establish TCP connection with [AF_INET]185.127.19.147:65301
Sep 26 10:59:55 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: TCP connection established with [AF_INET]185.127.19.147:65301
Sep 26 10:59:55 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: TCPv4_CLIENT link local: (not bound)
Sep 26 10:59:55 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: TCPv4_CLIENT link remote: [AF_INET]185.127.19.147:65301
Sep 26 10:59:55 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: [server] Peer Connection Initiated with [AF_INET]185.127.19.147:65301
Sep 26 10:59:56 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: TUN/TAP device tun0 opened
Sep 26 10:59:56 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: net_iface_mtu_set: mtu 1420 for tun0
Sep 26 10:59:56 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: net_iface_up: set tun0 up
Sep 26 10:59:56 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: net_addr_v4_add: 10.255.252.3/24 dev tun0
Sep 26 10:59:56 OpenMPTCProuter daemon.notice openvpn(omr)[17349]: /usr/libexec/openvpn-hotplug up omr tun0 1420 0 10.255.252.3 255.255.255.0 init
Sep 26 10:59:56 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Sep 26 10:59:56 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Sep 26 10:59:56 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Sep 26 10:59:56 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Sep 26 10:59:56 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Sep 26 10:59:56 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='tun0'
Sep 26 10:59:56 OpenMPTCProuter daemon.info ModemManager[18635]: hotplug: add network interface tun0: event processed
Sep 26 10:59:56 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='ifb4tun0'
Sep 26 10:59:56 OpenMPTCProuter daemon.info ModemManager[19777]: hotplug: remove network interface ifb4tun0: event processed
Sep 26 10:59:56 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_fcfad'
Sep 26 10:59:56 OpenMPTCProuter daemon.info ModemManager[19919]: hotplug: add network interface SQM_IFB_fcfad: event processed
Sep 26 10:59:56 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_fcfad'
Sep 26 10:59:56 OpenMPTCProuter daemon.info ModemManager[19982]: hotplug: remove network interface SQM_IFB_fcfad: event processed
Sep 26 10:59:56 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_3ac81'
Sep 26 10:59:56 OpenMPTCProuter daemon.info ModemManager[20008]: hotplug: add network interface SQM_IFB_3ac81: event processed
Sep 26 10:59:56 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_3ac81'
[... cut ...]
Sep 26 11:00:26 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='tun0'
Sep 26 11:00:26 OpenMPTCProuter daemon.info ModemManager[2843]: hotplug: remove network interface tun0: event processed
Sep 26 11:00:26 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Sep 26 11:00:27 OpenMPTCProuter daemon.notice openvpn(omr)[26526]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.3 255.255.255.0 init
Sep 26 11:00:27 OpenMPTCProuter daemon.notice openvpn(omr)[26526]: SIGTERM[hard,] received, process exiting

talitha77, 2024-10-03:
i seem to have sorted it out..  looks like i had a multitude of other network problems anyway, but having 2 OMRS with the same IP connected to my router didnt help.

closed 2024-10-03 by talitha77