Ysurac/openmptcprouter #3540: mptcp over vpn not working 0.61rc2

DivineRuins, 2024-09-17:
## Expected Behavior

wireguard or openvpn should work normally with just ticking the box like it did in v0.60 (at least openvpn did)

## Current Behavior

in v0.61rc2-6.6 and 5.4 when wireguard is chosen for mptcp over vpn the handshake is done over the master interface and the data sent through the same wan interface which doesn't need mptcp over vpn to start with.

in my current setup on v0.60 stable-6.1 wireguard doesn't even handshake or work, here it connects but still isn't functional because am guessing its a routing issue.

Regarding OpenVPN
In my current version v0.60-6.1 Openvpn udp doesn't connect at all no data will be received from the VPS until i switch protocol to tcp-client and all works perfectly with the caviat that openvpn tcp is not ideal for mptcp over vpn

in the new 0.61rc2 which is the main focus of this report openvpn connects normally over udp and tcp but very little to no data will be send through the interface tried all possible combinations it simply doesnt work. still shows all green through

so to put this in short in 0.61rc2 both openvpn and wireguard mptcp over vpn are not working

## Possible Solution

Wireguard should be routed through the correct interface which actually needs mptcp.
am not sure what could be wrong with Openvpn  as it sends very little data like 20 100kbit/s

## Steps to Reproduce the Problem

  1. using wizard, assign wan IPs , choose mptcp on blocked interface and set it to mp enabled, assign master to the second       interface (reversing interface order still leads to the issues)
  2. assign any proxy (currently suing xray VLESS with aes)
  3.  save config with with wireguard and then change to openvpn both won't work

i tried several different combinations and setting which lead to same results

## Context (Environment)

Well knowing that people who need such solutions are people who are suffering from low internet speeds worldwide and usually these kind of ISPs have a monopoly in certain region and enforce anyway possible to prevent people from doing such things as combining bandwidth so they block it.. 
MPTCP over vpn is the main reason this product work for hundreds of people in my country and am pretty sure same goes to other ones

this feature should be given the love it deserves since without it the product simply wont work for a lot it is a major issue.

## Specifications

  - OpenMPTCProuter version:  0.61rc2
  - OpenMPTCProuter VPS version: fresh OS install and script pulled from server-rc kernel 6.6
  - OpenMPTCProuter VPS provider: privately owned
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-09-17:
MPTCP over VPN is still only a dirty workaround. For now only Shadowsocks-libev is supported (maybe Shadowsocks-Rust too but not tested).

DivineRuins, 2024-09-17:
I see, but in v0.60-6.1 stable , its working fine on all protocols shadowsocks-rust, xray ,v2ray but only using openvpn tcp-client. i will keep using v0.60 for now hopefully more love is put towards this feature..

i really  thank you a lot for your effort you put in making something like this available.

closed 2024-12-21 by github-actions[bot]