Ysurac/openmptcprouter #3541: Shorewall not working?

DivineRuins, 2024-09-17:
Can you please confirm shorewall is working ? i made couple of tests and they failed to block anything or enforce any rules
a simple PING(DROP)    net     $fw   doesn't do nothing i am thinking of moving to something else altogether
is it just me?

Ysurac, 2024-09-17:
Please use the template.
I confirm Shorewall is working. And ping must not be blocked.

DivineRuins, 2024-09-17:
ah excuse me thought it was a quick question, i know icmp needs to be open but i was testing something simple and it didn't block it nothing i change in the /etc/shorewall/rules is reflected after a shorewall reload for me.
can you suggest any other test

Ysurac, 2024-09-17:
I've done same test to check and it worked. I commented the "Ping(ACCEPT)" line and uncommented the "Ping(DROP)", ```systemctl restart shorewall``` and ping was dropped.
You can check if shorewall correctly started using ```journalctl -u shorewall``` and via ```iptables-save```

DivineRuins, 2024-09-17:
having all ports open got me worried for a sec. maybe there is another firewall taking precedence in the debian img i used, i'll check and report back, thank you

closed 2024-12-22 by github-actions[bot]