Ysurac/openmptcprouter #3542: Issues found with 0.61rc2-6.6 on rpi4 (some with fixes some unresolved)

datapharmer, 2024-09-17:
## Expected Behavior

run upgrade script and upgrade flash and everything goes well!

## Current Behavior

Issues to which I have a resolution/workaround:
-ran into a key issue when running the upgrade script on the vps. Ran: `sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 86F7D09EE734E623` and reran script to fix
-IPV6 DNS was not resolving (seems to be a general ipv6 failure). I had to disable local IPV6 DNS on LAN which isn't a good fix for ipv6 only clients but works for now. It still was advertising the IPV6 DNS IP after save and apply but rebooting seems to have resolved the advertisement
-VPS Hard Freeze: I started experiencing hard freeze on VPS which could be reproduced by running a test at speedtest.net and determined 1GB is no longer sufficient for VPS. I added a 4GB swap which seems to have resolved this freeze.
-fail2ban stopped working. set backend=systemd in /etc/fail2ban/jail.local and installed python3-systemd and is now working

## Possible Solution

I'm out of ideas - please help!

## Steps to Reproduce the Problem
  1. service systemd-networkd-wait-online fails to start and results in very long boot times as a result. Gives message "Wait for Network to be configured" I can disable the service but not sure if it is needed....
  2. service wg-quick@client-wg0 doesn't start with "WireGuard via wg-quick(8) for client/wg0 which is filling up /var/log/messages
  3. services using websockets are constantly resetting (for example buyvm web console resets every minute or two and must be reconnected
  4. IPv6 in general doesn't seem to be working. Clients are issued dhcpv6 addresses but even pinging router fails
  5. When VPS is unreachable (such as before reconnecting after a configuration change) the DNS Forwarders are not working and DNS on the router returns no results causing an effective internet outage until the vps connection is resumed instead of using a single wan route
  6. Getting constant reports on VPS of `workqueue: drm_fb_helper_damage_work hogged CPU for >1000us x times, consider switching to WQUNBOUND`
  7. OMR-Bypass seems to be ignored, at least for UDP and traceroute (regression?)
  
   

## Context (Environment)

IPV6 only parts of the internet can't be reached and changes to settings result in extended outages. Boot times on the VPS are poor

## Specifications

  - OpenMPTCProuter version:  0.61rc2-6.6
  - OpenMPTCProuter VPS version:  0.1031-test 6.10.10-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: BuyVM
  - OpenMPTCProuter platform:  RPI4
  - Country:  USA

[image]

[image]

[image]

Ysurac, 2024-09-17:
What is the error for VPS when you need the key ? I'm not able to reproduce this.

2)What is the result of ```journalctl -u wg-quick@client-wg0``` ?
4)In System->OpenMPTCProuter, "advanced settings" tab, do you have "enable IPv6" and "enable 6in4" checked ? Possible that 6in4 is not enabled by default and should be...
5) I will check for DNS when VPS is not here. I would need log I think, possible that OMRVPN interface is still up even if not working. What VPN are you using ?
6) For the message in kernel log, seems to be related to kernel, not sure we have to worry about that
7) What proxy are you using ?

datapharmer, 2024-09-17:
1. As I recall xanmod was failing to install with the message "NO_PUBKEY 86F7D09EE734E623" 
2. `-- no entries --`
4. I confirmed this 6in4 was not selected. I will try that and report back if it isn't resolved - maybe a good thing for a script to check or warn?
5. Let me know what logs I can provide... thinking about this more I wish pihole could run on the router too instead of the vps as that would solve a lot of headaches of using secondary unfiltered dns when vps is unavailable
7. I have tried several but saw this issue with XRAY VLESS Reality, Shadowsocks, and Shadowsocks 2022 Rust

Ysurac, 2024-09-17:
1. xanmod repo is no used anymore, what was the previous version ?
2.  And ```journalctl -u systemd-networkd-wait-online``` ?
3. This should be enabled by default or maybe added in wizard page.
4. Status->System log. You are using pi-hole ?
5. I will check this. It's by bypassing domains, IP, service, ?

datapharmer, 2024-09-17:
1. Previous version running was .59 I don't recall the exact build but can track down if needed... I think this was maybe something related to the deb files in the elif of 440 of the vps script
2. `-- Boot 1ea982c20ef849ceb825ed903d9a5065 --
Sep 17 07:05:42 localhost systemd[1]: Starting systemd-networkd-wait-online.service - Wait for Network to be Configured...
Sep 17 07:07:42 localhost systemd-networkd-wait-online[460]: Timeout occurred while waiting for network connectivity.
Sep 17 07:07:42 localhost systemd[1]: systemd-networkd-wait-online.service: Main process exited, code=exited, status=1/FAILURE
Sep 17 07:07:42 localhost systemd[1]: systemd-networkd-wait-online.service: Failed with result 'exit-code'.
Sep 17 07:07:42 localhost systemd[1]: Failed to start systemd-networkd-wait-online.service - Wait for Network to be Configured.`
4. Yes using the provided pihole script on vps currently, but also set some general internet forwarders with strict order enabled in case vps is down since pihole is on vps logs:
 `Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[204]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[204]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Invalid argument
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[205]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[205]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[205]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[205]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[205]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[205]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[205]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[205]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Resource busy
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Resource busy
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[208]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[207]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[207]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[207]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:36 OpenMPTCProuter daemon.err dnsmasq[207]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:37 OpenMPTCProuter daemon.err dnsmasq[211]: nftset inet fw4 omr_dscp_cs5_4 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:37 OpenMPTCProuter daemon.err dnsmasq[210]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:37 OpenMPTCProuter daemon.err dnsmasq[213]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Protocol error
Sep 17 17:26:37 OpenMPTCProuter daemon.err dnsmasq[215]: nftset inet fw4 omr_dscp_cs5_6 netlink: Error: cache initialization failed: Protocol error
[... cut ...]
  5    88 ms    83 ms    88 ms  te0-1-0-7.rcr21.b002802-2.mia01.atlas.cogentco.com [154.24.1.93]
  6    90 ms   105 ms   123 ms  be3410.ccr21.mia01.atlas.cogentco.com [154.54.6.85]
  7    82 ms    76 ms    72 ms  be3081.ccr41.mia03.atlas.cogentco.com [154.54.88.226]
  8    83 ms    75 ms    67 ms  ae30.cr6-mia1.ip4.gtt.net [173.241.128.133]
  9   102 ms    66 ms    88 ms  ae6.cr9-mia1.ip4.gtt.net [213.200.113.205]
 10    51 ms   114 ms    88 ms  ip4.gtt.net [98.124.172.178]
 11    70 ms    68 ms    51 ms  204.74.99.103

Trace complete.
```

datapharmer, 2024-09-18:
In regards to IPV6 after enabling IPV6 6In4 under advanced and rebooting the router the IPV6 initially worked and I was excited that it was such a simple fix, however after a couple hours the IPV6 dns has stopped responding and ping to the advertised IPV6 DNS server address also doesn't respond so it seems the router IPV6 isn't stable for some reason ping to a google IPv6 address is still working however so I believe it may just be a dns issue but I'm not certain.

Websocket connections are the other big problem still - they are still very unstable.

Ysurac, 2024-09-18:
What is the advertised IPv6 DNS server ?

datapharmer, 2024-09-18:
fd27:ca97:230:200::1

Ysurac, 2024-09-18:
What is the result, via ssh, of  ```ip -6 a``` on the router ?
What IPs do you have on your client computer ?

datapharmer, 2024-09-18:
`# ip -6 a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 state UNKNOWN qlen 1000
    inet6 ::1/128 scope host proto kernel_lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fd27:ca97:230:200::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::e65f:1ff:fea3:d95/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
132: eth0.2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fd27:ca97:230:100::1/56 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::e65f:1ff:fea3:d95/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
138: eth0.3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 state UP qlen 1000
    inet6 fe80::e65f:1ff:fea3:d95/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
139: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1280 state UNKNOWN qlen 1000
    inet6 fd00::a00:2/126 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::aff:ff02/64 scope link
       valid_lft forever preferred_lft forever
148: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 state UNKNOWN qlen 100
    inet6 fe80::98a6:eb33:4be4:a08b/64 scope link stable-privacy proto kernel_ll
       valid_lft forever preferred_lft forever
`

client:
`IPv6 Address. . . . . . . . . . . : fd27:ca97:230:200::694(Preferred)
   IPv6 Address. . . . . . . . . . . : fd27:ca97:230:200:f563:82ed:f84c:83f5(Preferred)
   Temporary IPv6 Address. . . . . . : fd27:ca97:230:200:4d5d:9895:2d97:9400(Preferred)
   Link-local IPv6 Address . . . . . : fe80::9065:9226:1cd7:4bc%22(Preferred)`

In regards to omr bypass issues I tried restarting the service and here's what the logs show:

`Sep 18 12:40:01 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Sep 18 12:40:02 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 18 12:40:02 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 18 12:40:02 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 18 12:40:02 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 18 12:40:28 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:29 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:29 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:29 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:29 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:29 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:30 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:30 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:30 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:30 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:31 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:31 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:31 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:31 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:31 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_6 Error: No such file or directory
Sep 18 12:40:31 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_6 Error: No such file or directory
Sep 18 12:40:31 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:32 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:32 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:32 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:32 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:32 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 18 12:40:32 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
[... cut ...]
Sep 18 12:40:50 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth0_3_4 Error: No such file or directory
Sep 18 12:40:50 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth0_3_4 Error: No such file or directory
Sep 18 12:40:50 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth0_3_4 Error: No such file or directory
Sep 18 12:40:51 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth0_3_6 Error: No such file or directory`

Ysurac, 2024-09-18:
Hum for the bypass this seems related to the use of VLAN.

Ysurac, 2024-09-18:
dns should not be set on any interface. I will force remove it.
What is the result, via SSH on the router, of ```ifstatus ATT``` and ```ifstatus Starlink``` ?

datapharmer, 2024-09-18:
Hrm I can't remember why dns was added to the interface but I think maybe starlink requires it to report stats to the app?

Here is the ifstatus:

`# ifstatus ATT
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 4340,
        "l3_device": "eth0.4",
        "proto": "dhcp",
        "device": "eth0.4",
        "updated": [
                "addresses",
                "routes",
                "data"
        ],
        "metric": 5,
        "dns_metric": 0,
        "delegation": false,
        "ipv4-address": [
                {
                        "address": "192.168.2.73",
                        "mask": 24
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [
                {
                        "address": "fd27:ca97:230::",
                        "mask": 56,
                        "local-address": {

                        }
                }
        ],
        "route": [

        ],
        "dns-server": [
                "10.255.254.1",
                "1.0.0.2",
                "1.1.1.2"
        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [
                        {
                                "target": "0.0.0.0",
                                "mask": 0,
                                "nexthop": "192.168.2.1",
[... 174 lines cut ...]
                "route": [
                        {
                                "target": "0.0.0.0",
                                "mask": 0,
                                "nexthop": "100.64.0.1",
                                "source": "100.124.24.171/32"
                        }
                ],
                "dns-server": [
                        "8.8.8.8",
                        "1.1.1.1"
                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {
                "dhcpserver": "100.64.0.1",
                "leasetime": 300
        }
}
`

datapharmer, 2024-09-19:
Hi I'm also seeing some VPS disconnects that seem to be tied to the firewall reloading on the router but not sure why it is reloading as nothing is being changed in the ui:

`Sep 19 14:50:32 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sep 19 14:50:33 OpenMPTCProuter user.notice MPTCP: Add route 192.168.100.1/32 via 192.168.100.1 dev eth0.2
Sep 19 14:50:33 OpenMPTCProuter user.notice MPTCP: Add route 100.64.0.1/32 via 100.64.0.1 dev eth0.2
Sep 19 14:50:34 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 14:50:34 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 14:50:34 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 14:50:34 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 14:50:34 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Sep 19 14:50:34 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Sep 19 14:50:34 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Sep 19 14:50:34 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sep 19 14:50:35 OpenMPTCProuter user.notice omr-tracker: Launched
Sep 19 14:50:35 OpenMPTCProuter user.notice omr-tracker: Launching...
Sep 19 14:50:35 OpenMPTCProuter daemon.info procd: - shutdown -
Sep 19 14:50:35 OpenMPTCProuter daemon.info procd: - shutdown -
Sep 19 14:50:35 OpenMPTCProuter daemon.info procd: - shutdown -
Sep 19 14:50:35 OpenMPTCProuter daemon.info procd: - shutdown -
Sep 19 14:50:35 OpenMPTCProuter daemon.info procd: - shutdown -
Sep 19 14:50:35 OpenMPTCProuter user.notice post-tracking-002-error: Restart omrvpn
Sep 19 14:50:36 OpenMPTCProuter user.notice MPTCP: Add route 192.168.100.1/32 via 192.168.100.1 dev eth0.2
Sep 19 14:50:36 OpenMPTCProuter user.notice MPTCP: Add route 100.64.0.1/32 via 100.64.0.1 dev eth0.2
Sep 19 14:50:36 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sep 19 14:50:37 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 14:50:37 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 14:50:37 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 14:50:37 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 14:50:38 OpenMPTCProuter user.notice MPTCP: Add route 192.168.100.1/32 via 192.168.100.1 dev eth0.2
Sep 19 14:50:38 OpenMPTCProuter user.notice MPTCP: Add route 100.64.0.1/32 via 100.64.0.1 dev eth0.2
Sep 19 14:50:40 OpenMPTCProuter daemon.info procd: Instance omr-tracker::ATT pid 22822 not stopped on SIGTERM, sending SIGKILL instead
Sep 19 14:50:40 OpenMPTCProuter daemon.info procd: Instance omr-tracker::Starlink pid 22820 not stopped on SIGTERM, sending SIGKILL instead
Sep 19 14:50:40 OpenMPTCProuter daemon.info procd: Instance omr-tracker::TMobile pid 22821 not stopped on SIGTERM, sending SIGKILL instead
Sep 19 14:50:40 OpenMPTCProuter daemon.info procd: Instance omr-tracker::instance5 pid 22823 not stopped on SIGTERM, sending SIGKILL instead
Sep 19 14:50:40 OpenMPTCProuter daemon.info procd: Instance omr-tracker::omrvpn pid 22819 not stopped on SIGTERM, sending SIGKILL instead
Sep 19 14:50:40 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs0_6 Error: Could not resolve hostname: Name has no usable address`

when this happens the internet is barely usable - things load really slowly or timeout... much slower than just disabling the vps server.

Ysurac, 2024-09-19:
Do you know why there is this:
```
Sep 19 14:50:38 OpenMPTCProuter user.notice MPTCP: Add route 192.168.100.1/32 via 192.168.100.1 dev eth0.2
Sep 19 14:50:38 OpenMPTCProuter user.notice MPTCP: Add route 100.64.0.1/32 via 100.64.0.1 dev eth0.2
```
like if 2 networks use same interface ?

datapharmer, 2024-09-19:
This is for starlink gen2 (rectangular dish) in bypass mode. I believe I couldn't get the app to recognize it was local at one point without these static routes set but not sure if they are still needed. the 192.168.100.1 just shows a "starlink" logo now (used to show stats) the issued ip range currently in bypass mode is 100.124.24.171/10 there is some discussion of a similar setup from a year or so ago here: https://github.com/Ysurac/openmptcprouter/issues/2947 but seems maybe the allocated ranges have since changed.

kevinh-csalabs, 2024-09-19:
I will try disabling these routes and removed the dns entries from interfaces to see if there is any improvement. I also just noticed that under Smart Queue Management the wan interfaces are duplicated - once with enable SQM on this instance enabled and once with it disabled? Is this expected?

kevinh-csalabs, 2024-09-19:
# uci show sqm
sqm.Starlink=queue
sqm.Starlink.interface='eth0.2'
sqm.Starlink.debug_logging='0'
sqm.Starlink.download='0'
sqm.Starlink.upload='0'
sqm.Starlink.autorate='0'
sqm.Starlink.min_download='3090'
sqm.Starlink.min_upload='2282'
sqm.Starlink.max_upload='22829'
sqm.Starlink.reflector_ping_interval_s='0.2'
sqm.Starlink.no_pingers='4'
sqm.Starlink.delay_thr_ms='25'
sqm.Starlink.connection_active_thr_kbps='500'
sqm.Starlink.substained_idle_sleep_thr_s='60'
sqm.Starlink.startup_wait_s='60'
sqm.Starlink.linklayer='atm'
sqm.Starlink.overhead='40'
sqm.Starlink.max_download='30906'
sqm.Starlink.qdisc_advanced='0'
sqm.Starlink.verbosity='5'
sqm.Starlink.iqdisc_opts='autorate-ingress'
sqm.Starlink.qdisc='cake'
sqm.Starlink.script='piece_of_cake.qos'
sqm.Starlink.enabled='1'
sqm.lan=queue
sqm.lan.interface='eth0'
sqm.lan.qdisc='fq_codel'
sqm.lan.script='simple.qos'
sqm.lan.linklayer='atm'
sqm.lan.overhead='40'
sqm.lan.enabled='0'
sqm.lan.debug_logging='0'
sqm.lan.download='0'
sqm.lan.upload='0'
sqm.lan.autorate='0'
sqm.lan.qdisc_advanced='0'
sqm.lan.verbosity='5'
sqm.lan.iqdisc_opts='autorate-ingress'
sqm.TMobile=queue
sqm.TMobile.interface='eth0.3'
sqm.TMobile.linklayer='atm'
sqm.TMobile.overhead='40'
sqm.TMobile.debug_logging='0'
sqm.TMobile.download='0'
sqm.TMobile.upload='0'
sqm.TMobile.autorate='0'
sqm.TMobile.min_download='1653'
sqm.TMobile.max_download='16533'
sqm.TMobile.min_upload='288'
sqm.TMobile.max_upload='2883'
sqm.TMobile.reflector_ping_interval_s='0.2'
sqm.TMobile.no_pingers='4'
sqm.TMobile.delay_thr_ms='25'
sqm.TMobile.connection_active_thr_kbps='500'
sqm.TMobile.substained_idle_sleep_thr_s='60'
sqm.TMobile.startup_wait_s='60'
sqm.TMobile.qdisc_advanced='0'
sqm.TMobile.verbosity='5'
sqm.TMobile.iqdisc_opts='autorate-ingress'
sqm.TMobile.qdisc='cake'
sqm.TMobile.script='piece_of_cake.qos'
sqm.TMobile.enabled='1'
sqm.ATT=queue
sqm.ATT.interface='eth0.4'
sqm.ATT.linklayer='atm'
sqm.ATT.overhead='40'
sqm.ATT.debug_logging='0'
sqm.ATT.download='0'
sqm.ATT.upload='0'
sqm.ATT.autorate='0'
sqm.ATT.min_download='892'
sqm.ATT.max_download='8925'
sqm.ATT.min_upload='1165'
sqm.ATT.max_upload='11656'
sqm.ATT.reflector_ping_interval_s='0.2'
sqm.ATT.no_pingers='4'
sqm.ATT.delay_thr_ms='25'
sqm.ATT.connection_active_thr_kbps='500'
sqm.ATT.substained_idle_sleep_thr_s='60'
sqm.ATT.startup_wait_s='60'
sqm.ATT.qdisc_advanced='0'
sqm.ATT.verbosity='5'
sqm.ATT.iqdisc_opts='autorate-ingress'
sqm.ATT.qdisc='cake'
sqm.ATT.script='piece_of_cake.qos'
sqm.ATT.enabled='1'
sqm.TMobile6=queue
sqm.TMobile6.interface='eth0.3'
sqm.TMobile6.qdisc='fq_codel'
sqm.TMobile6.script='simple.qos'
sqm.TMobile6.qdisc_advanced='0'
sqm.TMobile6.linklayer='atm'
sqm.TMobile6.overhead='40'
sqm.TMobile6.enabled='0'
sqm.TMobile6.debug_logging='0'
sqm.TMobile6.verbosity='5'
sqm.TMobile6.download='0'
sqm.TMobile6.upload='0'
sqm.TMobile6.iqdisc_opts='autorate-ingress'
sqm.TMobile6.autorate='0'
sqm.Starlink6=queue
sqm.Starlink6.interface='eth0.2'
sqm.Starlink6.qdisc='fq_codel'
sqm.Starlink6.script='simple.qos'
sqm.Starlink6.qdisc_advanced='0'
sqm.Starlink6.linklayer='atm'
sqm.Starlink6.overhead='40'
sqm.Starlink6.enabled='0'
sqm.Starlink6.debug_logging='0'
sqm.Starlink6.verbosity='5'
sqm.Starlink6.download='0'
sqm.Starlink6.upload='0'
sqm.Starlink6.iqdisc_opts='autorate-ingress'
sqm.Starlink6.autorate='0'
sqm.ATT6=queue
sqm.ATT6.interface='eth0.4'
sqm.ATT6.qdisc='fq_codel'
sqm.ATT6.script='simple.qos'
sqm.ATT6.qdisc_advanced='0'
sqm.ATT6.linklayer='atm'
sqm.ATT6.overhead='40'
sqm.ATT6.enabled='0'
sqm.ATT6.debug_logging='0'
sqm.ATT6.verbosity='5'
sqm.ATT6.download='0'
sqm.ATT6.upload='0'
sqm.ATT6.iqdisc_opts='autorate-ingress'
sqm.ATT6.autorate='0'
sqm.omrvpn=queue
sqm.omrvpn.qdisc='cake'
sqm.omrvpn.script='piece_of_cake.qos'
sqm.omrvpn.qdisc_advanced='0'
sqm.omrvpn.linklayer='none'
sqm.omrvpn.enabled='0'
sqm.omrvpn.interface='tun0'
sqm.omrvpn.download='0'
sqm.omrvpn.upload='0'
sqm.omrvpn.debug_logging='0'
sqm.omrvpn.verbosity='5'
sqm.omrvpn.overhead='44'
sqm.omrvpn.iqdisc_opts='autorate-ingress'
sqm.omrvpn.delay_thr_ms='300.0'
sqm.omrvpn.reflector_ping_interval_s='0.5'
sqm.omrvpn.enable_sleep_function='0'
sqm.omrvpn.autorate='0'

Ysurac, 2024-09-19:
What are the ATT6/Starlink6/Tmobile6 interfaces ?

kevinh-csalabs, 2024-09-19:
hrm maybe artifacts not properly removed via gui or present because IPV6 is enabled? A long long time ago I tested separate interfaces for IPV6 from IPV4 when having problems with IPV6 connectivity to VPN but that was well before upgrading from .59 and they don't show up under interfaces or devices.

kevinh-csalabs, 2024-09-19:
I've removed the ones corresponding with 6 interfaces to see if it helps at all.

kevinh-csalabs, 2024-09-19:
ok speeds (at least initially) look acceptable after that change but still having some weird networking issues and anything requiring an uninterrupted connection like websockets is unusable. It seems like maybe the wan connections are getting reset repeatedly (specifically starlink and tmobile) and then there is a rebind attack warning tied to the pihole so not sure if that false detection could be causing dns issues?

`Sep 19 16:25:49 OpenMPTCProuter daemon.notice netifd: Starlink (1151): udhcpc: sending renew to server 100.64.0.1
Sep 19 16:25:49 OpenMPTCProuter daemon.notice netifd: Starlink (1151): udhcpc: lease of 100.124.24.171 obtained from 100.64.0.1, lease time 300
Sep 19 16:26:09 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:09 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:09 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:09 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:11 OpenMPTCProuter daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: dns.msftncsi.com
Sep 19 16:26:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:27 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:29 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:30 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:30 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:33 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:33 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:33 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:35 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:35 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:35 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:35 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:35 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:35 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:35 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:35 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:36 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:43 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs1_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:50 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:52 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs0_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:26:52 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs0_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:27:36 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:27:57 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:28:17 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth0) 192.168.42.131 80:2a:a8:02:af:32 
Sep 19 16:28:17 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth0) 192.168.42.131 80:2a:a8:02:af:32 AFi-R-HD-02AF32
Sep 19 16:28:19 OpenMPTCProuter daemon.notice netifd: Starlink (1151): udhcpc: sending renew to server 100.64.0.1
[... cut ...]
Sep 19 16:46:18 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:46:22 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:46:29 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs4_6 Error: Could not resolve hostname: Name has no usable address
Sep 19 16:46:29 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address`

Ysurac, 2024-09-20:
Can you do same tests from the VPS ?

datapharmer, 2024-09-20:
curl -iv test on VPS seems fine.
Test on client with VPS disabled is fine.
Test on client with VPS enabled still problematic
ssh to wan when VPS disabled is solid
ssh to wan when VPS is connected disconnects almost immediately (trouble even logging in before it drops)
ssh to internal tunnel ip when VPS is connected still drops frequently (1-10 minutes)
with VPS enabled and shadowsocks rust as proxy journalctl -u shadowsocks-go gives:
`Sep 19 02:15:51 localhost shadowsocks-go[1079]: 2024-09-19T02:15:51.671-0700 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:51784", "username": "openmptcprouter", "targetAddress": "151.101.129.164:80", "client": "direct", "initialPayloadLength": 79}
Sep 19 02:15:51 localhost shadowsocks-go[1079]: 2024-09-19T02:15:51.755-0700 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:51784", "username": "openmptcprouter", "targetAddress": "151.101.129.164:80", "client": "direct", "nl2r": 79, "nr2l": 287}
Sep 19 02:15:53 localhost shadowsocks-go[1079]: 2024-09-19T02:15:53.885-0700 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:42896", "username": "openmptcprouter", "targetAddress": "1.1.1.1:80", "client": "direct", "initialPayloadLength": 71}
Sep 19 02:15:54 localhost shadowsocks-go[1079]: 2024-09-19T02:15:54.952-0700 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:42896", "username": "openmptcprouter", "targetAddress": "1.1.1.1:80", "client": "direct", "nl2r": 71, "nr2l": 219}
Sep 19 02:15:55 localhost shadowsocks-go[1079]: 2024-09-19T02:15:55.222-0700 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:54345", "username": "openmptcprouter", "targetAddress": "[2a03:2880:f22c:c6:face:b00c:0:7260]:80", "client": "direct", "initialPayloadLength": 504}
Sep 19 02:15:57 localhost shadowsocks-go[1079]: 2024-09-19T02:15:57.164-0700 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:36201", "username": "openmptcprouter", "targetAddress": "74.82.42.42:80", "client": "direct", "initialPayloadLength": 75}
Sep 19 02:15:57 localhost shadowsocks-go[1079]: 2024-09-19T02:15:57.244-0700 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:36201", "username": "openmptcprouter", "targetAddress": "74.82.42.42:80", "client": "direct", "nl2r": 75, "nr2l": 134}
Sep 19 02:15:59 localhost shadowsocks-go[1079]: 2024-09-19T02:15:59.350-0700 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:59516", "username": "openmptcprouter", "targetAddress": "198.41.212.162:80", "client": "direct", "initialPayloadLength": 78}
Sep 19 02:16:00 localhost shadowsocks-go[1079]: 2024-09-19T02:16:00.422-0700 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:59516", "username": "openmptcprouter", "targetAddress": "198.41.212.162:80", "client": "direct", "nl2r": 78, "nr2l": 391}
Sep 19 02:16:02 localhost shadowsocks-go[1079]: 2024-09-19T02:16:02.684-0700 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:9902", "username": "openmptcprouter", "targetAddress": "212.27.48.10:80", "client": "direct", "initialPayloadLength": 76}
Sep 19 02:16:02 localhost shadowsocks-go[1079]: 2024-09-19T02:16:02.951-0700 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:9902", "username": "openmptcprouter", "targetAddress": "212.27.48.10:80", "client": "direct", "nl2r": 76, "nr2l": 195}
Sep 19 02:16:05 localhost shadowsocks-go[1079]: 2024-09-19T02:16:05.041-0700 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:46006", "username": "openmptcprouter", "targetAddress": "198.27.92.1:80", "client": "direct", "initialPayloadLength": 75}
Sep 19 02:16:05 localhost shadowsocks-go[1079]: 2024-09-19T02:16:05.174-0700 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:46006", "username": "openmptcprouter", "targetAddress": "198.27.92.1:80", "client": "direct", "nl2r": 75, "nr2l": 348}
Sep 19 02:16:05 localhost shadowsocks-go[1079]: 2024-09-19T02:16:05.855-0700 INFO service/tcp.go:263 Two-way relay completed {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:54345", "username": "openmptcprouter", "targetAddress": "[2a03:2880:f22c:c6:face:b00c:0:7260]:80", "client": "direct", "nl2r": 1267, "nr2l": 973}
[... cut ...] nmptcprouter", "targetAddress": "198.41.212.162:80", "client": "direct", "nl2r": 78, "nr2l": 391}
Sep 19 02:18:46 localhost shadowsocks-go[1079]: 2024-09-19T02:18:46.177-0700 INFO service/tcp.go:246 Two-way relay started {"server": "ss-2022", "listener": 0, "listenAddress": "[::]:65280", "clientAddress": "[::ffff:129.222.253.84]:27831", "username": "openmptcprouter", "targetAddress": "212.27.48.10:80", "client": "direct", "initialPayloadLength": 76}
[logs truncated due to comment length limit]
`

datapharmer, 2024-09-20:
btw these logs are from attempting to visit via browser.... curl request including curl -a didn't record anything i that log.

datapharmer, 2024-09-20:
Is it possible this is a firewall issue? starting firewall on the router I see:
`
Section @defaults[0] option 'disable_ipv6' is not supported by fw4
Section @rule[8] (Support-UDP-Traceroute) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstip_4 (omr_dst_bypass_eth0_rule) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstip_4_accept (omr_dst_bypass_eth0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcip_4 (omr_dst_bypass_eth0_srcip) is disabled, ignoring section
Section omr_dst_bypass_eth0_mac_4 (omr_dst_bypass_eth0_mac) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcport_tcp_4 (omr_dst_bypass_eth0_srcport) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcport_udp_4 (omr_dst_bypass_eth0_srcport) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstport_tcp_4 (omr_dst_bypass_eth0_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstport_udp_4 (omr_dst_bypass_eth0_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstip_6 (omr_dst_bypass_eth0_rule) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstip_6_accept (omr_dst_bypass_eth0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcip_6 (omr_dst_bypass_eth0_srcip) is disabled, ignoring section
Section omr_dst_bypass_eth0_mac_6 (omr_dst_bypass_eth0_mac) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcport_tcp_6 (omr_dst_bypass_eth0_srcport) is disabled, ignoring section
Section omr_dst_bypass_eth0_srcport_udp_6 (omr_dst_bypass_eth0_srcport) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstport_tcp_6 (omr_dst_bypass_eth0_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth0_dstport_udp_6 (omr_dst_bypass_eth0_dstport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstip_4 (omr_dst_bypass_tun0_rule) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstip_4_accept (omr_dst_bypass_tun0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcip_4 (omr_dst_bypass_tun0_srcip) is disabled, ignoring section
Section omr_dst_bypass_tun0_mac_4 (omr_dst_bypass_tun0_mac) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcport_tcp_4 (omr_dst_bypass_tun0_srcport) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcport_udp_4 (omr_dst_bypass_tun0_srcport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstport_tcp_4 (omr_dst_bypass_tun0_dstport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstport_udp_4 (omr_dst_bypass_tun0_dstport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstip_6 (omr_dst_bypass_tun0_rule) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstip_6_accept (omr_dst_bypass_tun0_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcip_6 (omr_dst_bypass_tun0_srcip) is disabled, ignoring section
Section omr_dst_bypass_tun0_mac_6 (omr_dst_bypass_tun0_mac) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcport_tcp_6 (omr_dst_bypass_tun0_srcport) is disabled, ignoring section
Section omr_dst_bypass_tun0_srcport_udp_6 (omr_dst_bypass_tun0_srcport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstport_tcp_6 (omr_dst_bypass_tun0_dstport) is disabled, ignoring section
Section omr_dst_bypass_tun0_dstport_udp_6 (omr_dst_bypass_tun0_dstport) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_dstip_4 (omr_dst_bypass_6in4_omr6in4_rule) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_dstip_4_accept (omr_dst_bypass_6in4_omr6in4_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_srcip_4 (omr_dst_bypass_6in4_omr6in4_srcip) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_mac_4 (omr_dst_bypass_6in4_omr6in4_mac) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_srcport_tcp_4 (omr_dst_bypass_6in4_omr6in4_srcport) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_srcport_udp_4 (omr_dst_bypass_6in4_omr6in4_srcport) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_dstport_tcp_4 (omr_dst_bypass_6in4_omr6in4_dstport) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_dstport_udp_4 (omr_dst_bypass_6in4_omr6in4_dstport) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_dstip_6 (omr_dst_bypass_6in4_omr6in4_rule) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_dstip_6_accept (omr_dst_bypass_6in4_omr6in4_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_srcip_6 (omr_dst_bypass_6in4_omr6in4_srcip) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_mac_6 (omr_dst_bypass_6in4_omr6in4_mac) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_srcport_tcp_6 (omr_dst_bypass_6in4_omr6in4_srcport) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_srcport_udp_6 (omr_dst_bypass_6in4_omr6in4_srcport) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_dstport_tcp_6 (omr_dst_bypass_6in4_omr6in4_dstport) is disabled, ignoring section
Section omr_dst_bypass_6in4_omr6in4_dstport_udp_6 (omr_dst_bypass_6in4_omr6in4_dstport) is disabled, ignoring section
Section omr_dst_bypass_eth0_2_dstip_4_accept (omr_dst_bypass_eth0_2_rule_accept) is disabled, ignoring section
Section omr_dst_bypass_eth0_2_srcip_4 (omr_dst_bypass_eth0_2_srcip) is disabled, ignoring section
[... cut ...]
Section miniupnpd option 'family' is not supported by fw4
Section miniupnpd option 'reload' is not supported by fw4
Section miniupnpd specifies unreachable path '/usr/share/miniupnpd/firewall.include', ignoring section
Automatically including '/usr/share/nftables.d/table-post/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/dstnat/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/forward/20-miniupnpd.nft'
Automatically including '/usr/share/nftables.d/chain-post/srcnat/20-miniupnpd.nft'
`

Ysurac, 2024-09-20:
No, it's only about omr-bypass: all needed rules are created but when not used they are disabled.

datapharmer, 2024-09-25:
@Ysurac  I was able to resolve the hang on boot issue on debian 12 kernel 6.10 with systemd-networkd-wait-online.service by adding an override to only wait on eth0:

/etc/systemd/system/systemd-networkd-wait-online.service.d/override.conf
`[Service]
ExecStart=
ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --interface=eth0`

I haven't found a solution for the main issue however, so I've still been bypassing the VPS other than testing until I can figure out what's wrong.

Ysurac, 2024-09-25:
At the boot you should have only the default interfaces, so the script should work without issues... strange...

datapharmer, 2024-09-25:
I agree it is odd.... it still hung with `--any` specified

datapharmer, 2024-09-26:
The connection resets are the most problematic issue for me at this point, and the best I can tell they correlate with the firewall and OMR-Tracker getting restarted and then tracker reporting issues with the proxy not getting test replies which seems to happen around the time starlink requests its dhcp renewal every 5 minutes. I'm not seeing any renewal failures and I'm not having any problems with connectivity via starlink when the VPS is disabled, but maybe these dhcp renewal are causing problems for the omr-tracker or proxy for some reason. I'm not sure if these are causes or just symptoms... open to other ideas on what to be looking for.

For the `dnsmasq[1]: nftset inet fw4 omr_dscp_cs5_6 Error: Could not resolve hostname: Name has no usable address` and similar errors it seems the domains listed under IP Sets in DHCP and DNS are not getting resolved to IPs but I'm not sure why. This is a problem even with the VPS disabled. Nslookup from the router cli seems to work fine for all of these domains. There was a known issue in versions of dnsmasq prior to 2.87 that could cause this but it should be fixed since I am running 2.90.

Ysurac, 2024-09-26:
DHCP renewal should not be an issue if it's fast (should be), but you can change timeout values in Services->OMR-Tracker manager
For the dnsmasq it's in this case a domain that doesn't resolve using IPv6.

kevinh-csalabs, 2024-10-02:
Changing timeout values hasn't resolved this unfortunately. For dnsmasq is that safe to ignore or do those domains need to be removed? I was concerned that could be breaking omr-bypass but not sure the two issues are related.

Ysurac, 2024-10-02:
No, the dnsmasq warnings are not related.

datapharmer, 2024-10-08:
Ok, I have upgraded VPS to latest test script listed in v0.61RC3 post with kernel 6.11 and RPI4 to v0.61RC3 and had the same tls/handshake issues. Long story short, I made a number of changes and the issues haven't shown up in the last 20 minutes or so - I'm keeping my fingers crossed, and will do some more testing. If the issue doesn't return I'll try to narrow down exactly which setting or combination was causing the headaches and report back if I can make it reproducible. My initial suspicion is ShadowSocks Obfuscating is problematic. The plugin seems to be deprecated, so it may be best to replace or remove this in any case, but right not that's just a hunch.

As far as issues go, the only active problem I'm seeing so far is OMR-Bypass is still completely broken. Running a traceroute to a website assigned to a specific wan interface by domain still goes through the VPS.

kevinh-csalabs, 2024-10-10:
@Ysurac  so interestingly if the router can't connect to the vps the omr-bypass entries with a wan interface specified try to use the specified interface, which is also true if that interface is failing, so it seems maybe there is a route prioritization conflict between the tunnel and omr-bypass? Any suggestions on troubleshooting are appreciated!

Ysurac, 2024-10-10:
For OMR-ByPass, try latest snapshot.

datapharmer, 2024-10-15:
At least for domains, ipv6 does not seem to be getting bypassed as expected (both for icmp and tcp). I added www.google.com and confirmed ipv4 traceroute was showing it going there directly, but whenever it is loaded via na ipv6 address google shows a captcha that matches the VPS network IPv6 and traceroute via IPv6 also confirms this is routing through the VPS and not being bypassed. IPv4 seems ok other than any reloads taking several minutes.

I checked "Disable AAAA IPv6 DNS" on the omr-bypass entry and and save/applied and confirmed omr-bypass had loaded, but the router is still returning IPv6 records, so that seems to be broken too.

IPv4 addresses under "IPs and Networks" still seem to be ignored on the latest snapshot for RPI4 (October 14 found here https://snapshots.openmptcprouter.com/6.6/rpi4/targets/bcm27xx/bcm2711/ ) when domain entries are working which seems odd but I've tested it several times and it seems consistently broken.

I think the other failures outside IPs and Networks and I{v6 domains probably corresponds to long reload times as you mention, as it definitely takes an extended period to become active.

datapharmer, 2024-10-16:
I did some additional testing on omr-bypass, and marking all ASNs as not enabled and save and apply and the omr bypass restart still look several minutes to complete. I think there must be something else/additional holding things up. I'm currently using domain, ip/network, mac, port destination, and protocols/services options.

datapharmer, 2024-10-30:
@Ysurac Just letting you know that after installing .61-6.6 stable release on rpi4  I am still seeing issues with omr-bypass. Rules entered under "IPs and Networks" such as a single ip address are not bypassing the VPS as expected, and domain entries do work but OMR-Bypass still takes a very very long time to start or restart after changes  (5-10 minutes)

Also after restart when apply the upgrade ntp-sync still did not happen in a timely manner. I had to sync via button in the ui (waited about 5 minutes after the router came back up and it still complained of a big time difference). It connected fine after the manual sync though.

Ysurac, 2024-10-30:
@datapharmer I'm not able to reproduce issues on IPs and Networks (it's in fact using the same part for domains).
OMR-ByPass is slow on start when ASN is used or when you bypass some services (like Amazon).
For NTP, should be fixed in snapshot.

closed 2025-02-03 by github-actions[bot]