Ysurac/openmptcprouter #3544: Finally upgraded to 0.6 release, omr-bypass is not working.

ioogithub, 2024-09-19:
## Expected Behavior

0.6 is a major release all features should be expected to work.

## Current Behavior

Omr bypass is not working.

## Possible Solution

It looks like iplist functionality was added to the web interface, omr-bypass uses these lists, is there a known issue or bug with omr-bypass in this stable release version? I can't see any documentation of this issue.

## Steps to Reproduce the Problem

  1. Using the web interface, enter a website to bypass (services-omr-bypass): ipconfig.co
  2. Ensure clients use OMR for DNS.
  3.  Check the iplist to see omr_dst_bypass_eth1_4 and omr_dst_bypass_eth1_6
  4. curl ipconfig.co on a client, it will return the VPS IP and not the IP of the bypassed entry (eth1)

## Context (Environment)

It affects me because I will not be able to use OMR 0.6 if major systems are broken with this release. I waited a long time to upgrade to 0.6. If this is a known issue can you please document how to fix it so I can continue testing the 0.6 release?

## Specifications

  - OpenMPTCProuter version:
 PACKAGE:     openmptcprouter
 VERSION:     v0.60
 TARGET:      x86_64
 ARCH:        x86_64

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Tue Apr  9 19:57:53 UTC 2024

Ysurac, 2024-09-19:
It's a know issue, should be fixed in 0.61rc2

ioogithub, 2024-09-19:
ah okay, I have been searching for a solution. Do you remember what the fix was? I would like to try to fix it on this stable release version because so far everything else is working well for this deployment and the stability is also increased over the past stable version. You added some really great improvements to this version, well done!

Could you please link to the code or the file change in github where you fixed omr-bypass so I can look at the change and try to fix this deployment myself? I was searching for some time looking at code changes but could not find it.

ioogithub, 2024-09-19:
Hi [Ysurac](https://github.com/Ysurac),

I spend a few hours looking at every commit from today back to last august looking for where you fixed the omr bypass but could not find it.

I installed the version: [Powered by openmptcprouter v0.61rc2-6.6 r0+27346-c7ba5574f5](https://github.com/ysurac/openmptcprouter)

Questions:
1. With t6.1rc2  I seem to be losing bandwidth. 6.1r2 seems slower than 6.0. I can easily boot between the two systems with the 6.x version so it is easy to test both version.

2. When I check Network-MPTCP_Congestion Control I lost bbr1 and bbr3, I was using bbr3 on 6.0. What happened to bbr1 and bbr3?

3. The page says default is cubic however reno was selected from a fresh install, is this a mistake?

4. My initial test with omr-bypass seens to be working- curl ifconfig.co. What is the easiest way to import my bypass list from 6.0 to 6.1rc2?

5. Do I need the same kernel on OMR and VPS?

omr:
Linux OpenMPTCProuter 6.6.50 #0 SMP PREEMPT_DYNAMIC Tue Sep 10 22:37:34 2024 x86_64 GNU/Linux

vps:
Linux vps 6.1.0-25-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.106-3 (2024-08-26) x86_64 GNU/Linux

Do you have a list of changes or new features available when using new kernels? Which kernel have you tested the most and which would be suitable for production?

6. Are there any known major features or systems that are not working in 6.0rc2?

Ysurac, 2024-09-19:
1. Can you test to replace ```/etc/sysctl.d/openmptcprouter.conf``` by https://github.com/Ysurac/openmptcprouter-feeds/blob/ec22b7e248880e7a60770ba34360889f73092fca/openmptcprouter/files/etc/sysctl.d/openmptcprouter.conf on 0.61rc2 and reboot to check if the changes impact speed or not ?
2. BBR3 is now BBR
3. Should be cubic, I will check.
4. Not sure if it's an easy way, but you can try to copy /etc/config/omr-bypass to 0.61rc2
5. VPS should always have same kernel or higher than router. But try 1 before changing (this will help debug for now) kernel 6.6 is the most tested kernel and current production kernel. For changes, it's kernel changelog and lot of patches...
6. All that was working in 0.60 is working in 0.61rc2 and some part that was not working is now working (or at least should). Release candidate 3 will fix OMR-ByPass with some XRay/V2Ray protocols.

Ysurac, 2024-09-20:
1. It was to test if sysctl change reduce speed or not.
2. BBR3 now replace BBRv1 and BBRv2
3. What do you mean in flow control ?
4. This will be hide in next release
5. It's possible to upgrade by running the script again with the good KERNEL= argument and reboot
6. "No server ping after 1 second" is only a warning and it's the result of a ```ping -B -I eth1 <vpsip>``` command. I will look why you can get "unknown status"...
It's not always a real problem, Ping/ICMP is not high priority packets. This can also be due to real packets that can happen on any wireless connections.

ioogithub, 2024-09-20:
1. I will test this further, I can't say yes or no right now for sure.
2. okay thank you. 
3. Sorry I mean congestion control, what do you recommend for a starlink and an lte- starlink speed varies throughout the day, lte speed is very constant and reliable but I do see those missing pings, not sure if it is a problem. Anyway which congestion control is recommended, right now both connections have 38 and 42ms latency this is typical.
4. Why hide it, i like seeing this in the new version because exposing it helps the user to understand how the omr-bypass system is working.
5. would you recommend upgrading? are there known benefits to upgrade from kernel 6.1 to 6.6 on the vps? any new multipath features introduced?
6. I will monitor maybe it isn't an issue.

If there is anything else you are working on that you would like me to test please let me know.

Ysurac, 2024-09-21:
3. BBRv3 should be the best in all cases now. Else Cubic will always work but slower.
4. I hide them now because they are not editable via DHCP&DNS interface
5. All my tests are currently done on 6.6 kernel on both VPS & router side. On router side this add more Multipath Scheduler support. And on all side this fixe some Multipath issues.

ioogithub, 2024-09-21:
When I reboot OMR I see some omr-bypass errors, are these a problem:

```
Sep 21 16:40:18 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S98omr-bypass: sed: to: No such file or directory

Sep 21 16:40:44 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S98omr-bypass: /etc/rc.common: line 33: /sbin/wifi: not found
Sep 21 16:40:44 OpenMPTCProuter daemon.notice procd: /etc/rc.d/S98omr-bypass: /etc/rc.common: line 33: /sbin/wifi: not found

Sep 21 16:41:11 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:11 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:12 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:13 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:13 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth1_4 Error: No such file or directory
Sep 21 16:41:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:15 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:15 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:15 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:15 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:16 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:17 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
[... cut ...]
Sep 21 16:41:48 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:48 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_eth2_4 Error: No such file or directory
Sep 21 16:41:49 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Sep 21 16:41:50 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
```

Ysurac, 2024-09-22:
Dnsmasq display error when it start (or restart) before OMR-ByPass. But yes here seems there is a  problem with the first errors...

Nicras, 2024-10-02:
Hey @Ysurac,

thank you very much for this amazing software. I'm using it for one week now with Telekom 5G and Starlink on a NanoPi R5S and I get a stable internet connection with appropriate speeds. Made my life a bit better!

But my wife is complaining hard because we can't stream RTL+ and Disney+ in Germany anymore. I'm using a VPS from Strato which is fine, but the IP gets recognized as VPN or Proxy. I tried 4 different mostly unknown VPS Hostings but each had the same result. RTL+ uses a service called maxmind and I asked them to whitelist my IP but I don't think, they will do. So, I need to bypass the streaming traffic.

Where can I get the correct 0.61rc2 image for my ARM-based R5S? Or can you tell me another way of upgrading or do you have an ETA for an official release?

Thanks in advance

closed 2025-01-06 by github-actions[bot]