Ysurac/openmptcprouter #3546: Random TLS handshake errors

x4rd0o1Vtx, 2024-09-20:
## Expected Behavior

Reach successfully HTTPS requests on browser / cli.

## Current Behavior

Random TLS handshake error and timeout while fetching ressources, a simple retry can result of success but this may happens several times per day...
 
Browser error : PR_END_OF_FILE_ERROR

Git fetch errors : (seems to occurs on random stage)

  - error: RPC failed; curl 56 GnuTLS recv error (-110): The TLS connection was non-properly terminated.
fatal: expected flush after ref listing
- fatal: unable to access 'https://github.com/Ysurac/openmptcprouter.git/': GnuTLS recv error (-110): The TLS connection was non-properly terminated.
- error: RPC failed; curl 16 Error in the HTTP2 framing layer
fatal: expected flush after ref listing
- gnutls_handshake() failed: The TLS connection was non-properly terminated.
- Process was killed because it reached the timeout

## Steps to Reproduce the Problem

Because this happens randomly it's difficult to reproduce but the following method works very well:

```bash
cd /tmp && git clone https://github.com/Ysurac/openmptcprouter.git && cd openmptcprouter/
while true; do error=$(git fetch --prune 1> /dev/null); [[ -z "$error" ]] && echo -n "." || echo -e "$error"; done
```

## Context (Environment)

  - Router

One active wan interface (fiber ISP)

  - VPS

Debian GNU/Linux 12 (bookworm)
Kernel 6.1.106-3 (2024-08-26) x86_64 GNU/Linux

## Specifications

  - OpenMPTCProuter version: v0.59.1-5.4 r0+16594-ce92de8c8c
  - OpenMPTCProuter VPS version: 0.1030
  - OpenMPTCProuter platform: RPI4

## Comments

I've a fresh install on both router & VPS that is only use for that and I didn't notice any RAM/CPU saturation. Everything was working well for several months but this happens since few weeks for no reason because I didn't change anything. Maybe I've just updated the VPS packages (apt update && apt upgrade) but I'm not sure it's related.

The only thing I'm sure about it's related to the VPS proxy part because when I check the "Disable server" option on the router I no longer have any errors.

I've try the following but none of these solve the problem:

  - Set V2Ray instead of Shadowsocks default proxy
  - Even set the default proxy to None and using Glorytun UDP
  - Update couple times the VPS server with the script
  - Reboot router & VPS
  - Check router system log & VPS journalctl but nothing when the error occured
  - Use the router Debug option but no more info

I've also found this [issue](https://github.com/Ysurac/openmptcprouter/issues/813) that describe the same behavior but since it's "resolved" just some people comment that they still have the issue but nothing more...

I believed it's was first related to shadowsocks as the issue mention it but I've the right config with LimitNOFILE=99999 & LimitNPROC=99999 and still have the error when I disable shadowsocks proxy on the router so I really don't know.

So frustrating because your work is amazing and it was so stable before I got these errors from nowhere. If someone have some advises to investigate and help me to understand what is happening that would be really great :)

Ysurac, 2024-09-20:
I don't understand the versions you have...
If it's 0.59.1 with 5.4 kernel and VPS 0.1030 with 6.1 kernel then it's not supported (mix of kernels and old release on router).

x4rd0o1Vtx, 2024-09-20:
You shocked me with your quick response :o and you've just pointed out something I hadn't seen at all...
I'm not sure if it was like that since the initial setup or if I've upgraded the VPS kernel without noticed.
So I imagine it's better to upgrade the old release on router instead of downgrading the kernel on the VPS server.
Thanks again I will try to have the same kernel on both side :)

datapharmer, 2024-09-20:
I was just about to post that I'm seeing this same issue on the latest .61rc2 as well also when I saw this post. Current settings are proxy: shadowsocks rust 2022 and glorytun tcp but I don't think the issue is specific to the proxy/vpn selection.

I suspect this may be the underlying cause of the websocket issues I mentioned in the other issue I raised about .61rc2 

With the server disabled I'm not seeing the same websocket disconnects or tls issues currently but not sure if it is happenstance or not.

I tried lowering MTU to 1412 thinking possibly it was being caused by packet fragmentation but that didn't help. If there are any particular logs I should be checking when this happens let me know!

I noticed that vps running debian12 wasn't using ntp by default and the time seems to have drifted a smidge so I'm curious if that could be the culprit. I just installed ntp and confirmed the time sync matches my client and the router... it may be a good idea to make sure ntp is in the vps script in general though unless time is being maintained another way as that can definitely reek havoc on encrypted connections.

I'll report back if the ntp install fixes this or not.

datapharmer, 2024-09-20:
@Ysurac here is the error I'm getting via curl on the client - it looks like connections are getting reset frequently for some reason - you can see after running it repeatedly eventually it load.... there were probably another half dozen before the successful one that were the same as these resets:

`>curl -iv https://www.bankofamerica.com:443
* Host www.bankofamerica.com:443 was resolved.
* IPv6: (none)
* IPv4: 171.161.116.100
*   Trying 171.161.116.100:443...
* Connected to www.bankofamerica.com (171.161.116.100) port 443
* schannel: disabled automatic use of client certificate
* ALPN: curl offers http/1.1
* Recv failure: Connection was reset
* schannel: failed to receive handshake, SSL/TLS connection failed
* Closing connection
* schannel: shutting down SSL/TLS connection with www.bankofamerica.com port 443
* Send failure: Connection was reset
* schannel: failed to send close msg: Failed sending data to the peer (bytes written: -1)
curl: (35) Recv failure: Connection was reset

>curl -iv https://www.bankofamerica.com:443
* Host www.bankofamerica.com:443 was resolved.
* IPv6: (none)
* IPv4: 171.161.116.100
*   Trying 171.161.116.100:443...
* Connected to www.bankofamerica.com (171.161.116.100) port 443
* schannel: disabled automatic use of client certificate
* ALPN: curl offers http/1.1
* schannel: next InitializeSecurityContext failed: CRYPT_E_REVOCATION_OFFLINE (0x80092013) - The revocation function was unable to check revocation because the revocation server was offline.
* Closing connection
* schannel: shutting down SSL/TLS connection with www.bankofamerica.com port 443
curl: (35) schannel: next InitializeSecurityContext failed: CRYPT_E_REVOCATION_OFFLINE (0x80092013) - The revocation function was unable to check revocation because the revocation server was offline.

>curl -iv https://www.bankofamerica.com:443
* Host www.bankofamerica.com:443 was resolved.
* IPv6: (none)
* IPv4: 171.161.100.100
*   Trying 171.161.100.100:443...
* Connected to www.bankofamerica.com (171.161.100.100) port 443
* schannel: disabled automatic use of client certificate
* ALPN: curl offers http/1.1
* Recv failure: Connection was reset
* schannel: failed to receive handshake, SSL/TLS connection failed
* Closing connection
* schannel: shutting down SSL/TLS connection with www.bankofamerica.com port 443
* Send failure: Connection was reset
* schannel: failed to send close msg: Failed sending data to the peer (bytes written: -1)
curl: (35) Recv failure: Connection was reset

>curl -iv https://www.bankofamerica.com:443
* Host www.bankofamerica.com:443 was resolved.
* IPv6: (none)
* IPv4: 171.161.100.100
*   Trying 171.161.100.100:443...
* Connected to www.bankofamerica.com (171.161.100.100) port 443
* schannel: disabled automatic use of client certificate
* ALPN: curl offers http/1.1
* Recv failure: Connection was reset
* schannel: failed to receive handshake, SSL/TLS connection failed
* Closing connection
* schannel: shutting down SSL/TLS connection with www.bankofamerica.com port 443
* Send failure: Connection was reset
* schannel: failed to send close msg: Failed sending data to the peer (bytes written: -1)
curl: (35) Recv failure: Connection was reset

>curl -iv https://www.bankofamerica.com:443
* Host www.bankofamerica.com:443 was resolved.
* IPv6: (none)
* IPv4: 171.161.100.100
*   Trying 171.161.100.100:443...
* Connected to www.bankofamerica.com (171.161.100.100) port 443
* schannel: disabled automatic use of client certificate
[... 32 lines cut ...]
X-BOA-RequestID: [sanitized]
< Content-Type: text/html
Content-Type: text/html
< Set-Cookie: WPID=F1S1; Path=/; Domain=.bankofamerica.com
Set-Cookie: WPID=F1S1; Path=/; Domain=.bankofamerica.com
< Set-Cookie: SID=0006751D0B0066ED702F; Path=/; Domain=.bankofamerica.com
Set-Cookie: SID=0006751D0B0066ED702F; Path=/; Domain=.bankofamerica.com
< Set-Cookie: BOFA_LOCALE_COOKIE=en-US; Path=/; Domain=.bankofamerica.com
Set-Cookie: BOFA_LOCALE_COOKIE=en-US; Path=/; Domain=.bankofamerica.com
< X-Serviced-By: /sparta/homepage/--jckIumsBTxxkY0Z0BCdHbg==--vKCIl4+eB7OPiYNHMYqlUw==
X-Serviced-By: /sparta/homepage/--jckIumsBTxxkY0Z0BCdHbg==--vKCIl4+eB7OPiYNHMYqlUw==
< Set-Cookie: [sanitized]; Path=/; Domain=.www.bankofamerica.com; Secure; HTTPOnly
Set-Cookie: [sanitized]; Path=/; Domain=.www.bankofamerica.com; Secure; HTTPOnly
< Strict-Transport-Security: max-age=31536000; includeSubDomains
Strict-Transport-Security: max-age=31536000; includeSubDomains
< Content-Security-Policy: default-src 'self' *.bankofamerica.com *.bac-assets.com *.ml.com https: wss: data: blob:; script-src 'self' *.bankofamerica.com *.bac-assets.com *.ml.com fsa.merrilledge.com merrilledge.com s3.amazonaws.com boa-api.arkoselabs.com cdn.cookielaw.org resources.digital-cloud.medallia.com players.brightcove.net metrics.brightcove.com cdnapisec.kaltura.com tags.tiqcdn.com akamai.tiqcdn.com glance.net beta.glancecdn.net storage.glancecdn.net cct.google cdn.mplxtms.com cdn.tt.omtrdc.net data.cmcore.com data.coremetrics.com iocdn.coremetrics.com libs.coremetrics.com mc.coremetrics.com mcdata.coremetrics.com mktgcdn.coremetrics.com recs.coremetrics.com secure-cdn.mplxtms.com convertro.com stage.convertro.com idsync.rlcdn.com test.coremetrics.com testdata.coremetrics.com tmscdn.coremetrics.com glancecdn.net www.google-analytics.com maps.googleapis.com www.googletagmanager.com mboxedge34.tt.omtrdc.net anrdoezrs.net cj.dotomi.com cj.com cj.mplxtms.com emjcd.com mczbf.com sjwoe.com secure-cdn.mplxtms.com cdn.mplxtms.com  [... cut ...] com idsync.rlcdn.com glancecdn.net cdnstorage.myglance.net www.google-analytics.com cdn-bofa.myglance.net resources.digital-cloud.medallia.com 'unsafe-inline'; worker-src 'self' blob:; frame-ancestors 'self' *.bankofamerica.com *.ml.com *.merrilledge.com;
< Server: Oops
Server: Oops
<

* schannel: failed to decrypt data, need more data
* schannel: failed to decrypt data, need more data
<!doctype html><html [content truncated.....]</html>* Connection #0 to host www.bankofamerica.com left intact
`

datapharmer, 2024-09-20:
sorry -will do

x4rd0o1Vtx, 2024-09-20:
I've upgraded the router to the latest version (v0.60-6.1 r0+24843-acf40c022e) but unfortunately the behavior has not changed...
I've also tested the new Shadowsocks-Rust 2022 proxy but same as well.
I'm sure nothing has changed on the router part so it makes me feel that may related to a VPS server update.

x4rd0o1Vtx, 2024-09-20:
@datapharmer I'm not sure you're facing the same issue but it's seems to be very close to the error message I got. Did you try to use my method to see if you've some error on your side too ?

That's what I get on my side:

> ..........fatal: unable to access 'https://github.com/Ysurac/openmptcprouter.git/': Recv failure: Connection reset by peer
......................................................................................................................................fatal: unable to access 'https://github.com/Ysurac/openmptcprouter.git/': Recv failure: Connection reset by peer
..............................................................................................................................................................................................................................................................................fatal: unable to access 'https://github.com/Ysurac/openmptcprouter.git/': GnuTLS recv error (-110): The TLS connection was non-properly terminated.

Ysurac, 2024-09-20:
Did you check time on both router and VPS ?

x4rd0o1Vtx, 2024-09-20:
Yes exact same time on both side and I've also installed the ntp package on the VPS.
All is green with the classic Shadowsocks, but with Shadowsocks Rust I got a warning on the VPS (Can't get public IP address from ShadowSocks Rust) but it still works.

Ysurac, 2024-09-20:
Did you reboot VPS after script update ? Do you have all keys in System->OpenMPTCProuter, wizard, advanced settings checkbox ?

x4rd0o1Vtx, 2024-09-20:
I've just thinking that I didn't run my method to reproduce directly on the VPS server and I got the same error :o It's very weird and I would tend to say that it's not related to the link between the router and the server.

Yes I've all the keys in the advanced settings.

Ysurac, 2024-09-20:
So same problem on the VPS ? What is the VPS host ?
For shadowsocks-rust issue, can I have Status->System log ?

x4rd0o1Vtx, 2024-09-20:
I've tested to switch twice to try extracting the interesting logs for the shadowsocks-rust issue, I think these lines may interest you:

```
Sep 20 17:51:10 router-mptcp daemon.err glorytun[11960]: couldn't set socket option `TCP_KEEPCNT'
Sep 20 17:51:10 router-mptcp daemon.err glorytun[11960]: couldn't set socket option `TCP_KEEPIDLE'
Sep 20 17:51:10 router-mptcp daemon.err glorytun[11960]: couldn't set socket option `TCP_KEEPINTVL'
Sep 20 17:51:10 router-mptcp daemon.err glorytun[11960]: couldn't set socket option `TCP_USER_TIMEOUT'
...
Sep 20 17:51:28 router-mptcp user.notice OMR-VPS: Setting server config error, try again
...
Sep 20 17:51:34 router-mptcp daemon.err procd: Got unexpected signal 1
...
Sep 20 17:52:10 router-mptcp user.notice omr-schedule-010-services: Set firewall on server vps
Sep 20 17:52:19 router-mptcp daemon.info omr-tracker-ss: Shadowsocks sss0 is down (can't contact via http 212.27.48.10, 198.27.92.1, 151.101.129.164)
Sep 20 17:52:20 router-mptcp user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
```

But for the initial issue the only conclusion I have now is that it seems more an hosting company problem (hardware or something else) but nothing related to the project. It was very nice to take your time for me and I'll not abuse on it.

x4rd0o1Vtx, 2024-09-20:
I'm not even sure it's related to TLS I've just tried targeting http site from the VPS server and I also got some Recv failure: Connection reset by peer...

I've disabled the ipv6 on the system but no change too.
Maybe the firewall but I've no more idea :/

Ysurac, 2024-09-21:
There is no reason this can be the VPS firewall, but to test you can disable it temporarily: ```systemctl stop shorewall```.
I think it's more a problem on hosting company.

x4rd0o1Vtx, 2024-09-21:
Yes I think the same thing and disable the firewall did not change the behavior. I should have started with that from the beginning...

Thank you again for your help and devotion, your work is truly incredible !

closed 2024-09-21 by x4rd0o1Vtx