Ysurac/openmptcprouter #3547: BPI-R4 Support

Yon2068, 2024-09-21:
**Is your feature request related to a problem? Please describe.**
I am trying to create an image for the BPI-R4, but so far I have encountered errors. Is it possible to create this image or not?

Ysurac, 2024-09-22:
I only support hardware that I own. Else it's difficult to test.
What error did you get when you compiled it ?

Yon2068, 2024-09-22:
> 我只支持我拥有的硬件。否则很难测试。编译时遇到了什么错误？

curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:110: /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/dl/rustc-1.78.0-src.tar.gz] Error 2
make[3]: Leaving directory '/media/yon/p0/omr/openmptcprouter/feeds/openmptcprouter/rust'
time: package/feeds/openmptcprouter/rust/host-compile#7.71#10.49#5050.80
    ERROR: package/feeds/openmptcprouter/rust [host] failed to build.
make[2]: *** [package/Makefile:175: package/feeds/openmptcprouter/rust/host/compile] Error 1
make[2]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source'
make[1]: *** [package/Makefile:171: /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2

Yon2068, 2024-09-22:
> I only support hardware that I own. Else it's difficult to test. What error did you get when you compiled it ?

curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.
make[3]: *** [Makefile:110: /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/dl/rustc-1.78.0-src.tar.gz] Error 2
make[3]: Leaving directory '/media/yon/p0/omr/openmptcprouter/feeds/openmptcprouter/rust'
time: package/feeds/openmptcprouter/rust/host-compile#7.71#10.49#5050.80
    ERROR: package/feeds/openmptcprouter/rust [host] failed to build.
make[2]: *** [package/Makefile:175: package/feeds/openmptcprouter/rust/host/compile] Error 1
make[2]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source'
make[1]: *** [package/Makefile:171: /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2

Yon2068, 2024-09-22:
> I only support hardware that I own. Else it's difficult to test. What error did you get when you compiled it ?

make[8]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools'
make[7]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools'
make[6]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5/gettext-tools'
make[6]: Entering directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5'
make[7]: Entering directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5'
make[7]: Nothing to be done for 'install-exec-am'.
make[7]: Nothing to be done for 'install-data-am'.
make[7]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5'
make[6]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5'
make[5]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5'
make[4]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5'
ln -sf msgfmt /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/staging_dir/hostpkg/bin/gmsgfmt
mkdir -p /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/staging_dir/hostpkg/stamp
touch /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5/.built
touch /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/staging_dir/hostpkg/stamp/.gettext-full_installed 
find /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/build_dir/hostpkg/gettext-0.22.5 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
make[3]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#1092.76#563.53#1578.64
make[2]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source'
make[1]: *** [package/Makefile:171: /media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source'
make: *** [/media/yon/p0/omr/openmptcprouter/myplatform/6.6/source/include/toplevel.mk:248：world]

-------------------

I encountered a new error during a recompilation

BlueWaterCrystal, 2024-09-24:
Have you tried compiling for kernel 6.1?

I had trouble compiling 6.6 for BPI-R3 but got 6.1 to compile. 

I don't know about the R4 but for the R3 I had to disable ramdisk, otherwise, it wouldn't create the SD card image.
Additionally, I had to create a non-root user and near the end, I used `make -j1` for the rest of the build.

This is the command I used to generate the platform:
`OMR_TARGET="BPI-R3" OMR_FEED_SRC="master" OMR_KERNEL="6.1" ./build.sh`

Yon2068, 2024-09-24:
> Have you tried compiling for kernel 6.1?
> 
> I had trouble compiling 6.6 for BPI-R3 but got 6.1 to compile.
> 
> I don't know about the R4 but for the R3 I had to disable ramdisk, otherwise, it wouldn't create the SD card image. Additionally, I had to create a non-root user and near the end, I used for the rest of the build.`make -j1`
> 
> This is the command I used to generate the platform: `OMR_TARGET="BPI-R3" OMR_FEED_SRC="master" OMR_KERNEL="6.1" ./build.sh`
---------------
Thanks for your suggestion; I will try again.

Yon2068, 2024-09-24:
> Have you tried compiling for kernel 6.1?
>
> I had trouble compiling 6.6 for BPI-R3 but got 6.1 to compile.
>
> I don't know about the R4 but for the R3 I had to disable ramdisk, otherwise, it wouldn't [...]

--------
When nearing the end, is the command make -j1 used because there are errors at that stage, and should I use make -j1 -k to let it continue compiling?

BlueWaterCrystal, 2024-09-24:
> When nearing the end, is the command make -j1 used because there are errors at that stage, and should I use make -j1 -k to let it continue compiling?

I think some components just don't like being built in parallel and end up failing the build.
If the build machine has all the required dependencies installed, `make -j$(nproc)` should work during the whole build.

I believe when I switched to 6.1, if there were errors it at least gave an explanation, which was mostly missing dependencies
vs 6.6 which didn't seem to give many answers and was more of some sort of compatibility issue.

Yon2068, 2024-09-24:
> Have you tried compiling for kernel 6.1?
>
> I had trouble compiling 6.6 for BPI-R3 but got 6.1 to compile.
>
> I don't know about the R4 but for the R3 I had to disable ramdisk, otherwise, it wouldn't [...]

---------------
Thank you, I successfully compiled it this time. I'm going to test it now.

BlueWaterCrystal, 2024-09-24:
Was there a target for BPI-R4?

I would Imagine the suffix of the file would be 
`mediatek-filogic-bananapi_bpi-r4-sdcard` instead of `mediatek-filogic-mediatek_mt7988a-rfb-sdcard`

I'm guessing there is some other sbc that uses the same CPU but has a different device tree and so on. 

bananapi_bpi-r4:
https://github.com/openwrt/openwrt/blob/23ac1ad9515588321970985859b2ee0cc8243f91/target/linux/mediatek/image/filogic.mk#L411

mediatek_mt7988a-rfb:
https://github.com/openwrt/openwrt/blob/23ac1ad9515588321970985859b2ee0cc8243f91/target/linux/mediatek/image/filogic.mk#L890C15-L890C35

Yon2068, 2024-09-24:
> BPI-R4 有靶点吗？
> 
> 我想文件的后缀会是 而不是`mediatek-filogic-bananapi_bpi-r4-sdcard``mediatek-filogic-mediatek_mt7988a-rfb-sdcard`
> 
> 我猜还有一些其他的 sbc 使用相同的 CPU，但具有不同的设备树，等等。
> 
> bananapi_bpi-r4： https://github.com/openwrt/openwrt/blob/23ac1ad9515588321970985859b2ee0cc8243f91/target/linux/mediatek/image/filogic.mk#L411
> 
> mediatek_mt7988a-RFB：https://github.com/openwrt/openwrt/blob/23ac1ad9515588321970985859b2ee0cc8243f91/target/linux/mediatek/image/filogic.mk#L890C15-L890C35

----------
I can't find the option for Bananapi BPi-R4 in the Target Profile, but it is indeed available in the OpenWRT compilation.

Yon2068, 2024-09-24:
> Was there a target for BPI-R4?
>
> I would Imagine the suffix of the file would be `mediatek-filogic-bananapi_bpi-r4-sdcard` instead of `mediatek-filogic-mediatek_mt7988a-rfb-sdcard`
>
> I'm guessing [...]

---------
I can't find the option for Bananapi BPi-R4 in the Target Profile, but it is indeed available in the OpenWRT compilation.

Ysurac, 2024-09-25:
I use a custom target/linux/mediatek/image/filogic.mk, I think BPI-R4 is missing, you can replace it with the one from openwrt in the link from @BlueWaterCrystal

Ysurac, 2024-09-25:
I've updated filogik.mk on my repo, so you should be able to compile for BPI-R4 now.

Yon2068, 2024-09-25:
> I've updated filogik.mk on my repo, so you should be able to compile for BPI-R4 now.

------
Thank you! I have replaced the target/linux/mediatek/image/filogic.mk file and am currently compiling. However, I will pull the code again and recompile once more.

Yon2068, 2024-09-25:
> I use a custom target/linux/mediatek/image/filogic.mk, I think BPI-R4 is missing, you can replace it with the one from openwrt in the link from @BlueWaterCrystal

------------
../dts/mt7986a-smartrg-bonanza-peak.dtsi:267.22-269.5: Warning (avoid_default_addr_size): /soc/pcie@11280000/pcie@0,0/mt7915@0,0: Relying on default #size-cells value
rm -f /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-smartrg-SDG-8612.dtb.tmp
rm -f /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-smartrg-SDG-8622.dtb.tmp
rm -f /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-asus-rt-ax59u.dtb.tmp
rm -f /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-asus-tuf-ax6000.dtb.tmp
rm -f /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-smartrg-SDG-8632.dtb.tmp
rm -f /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-smartrg-SDG-8614.dtb.tmp
rm -f /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-bananapi-bpi-r3.dtb.tmp
rm -f /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7986a-asus-tuf-ax4200.dtb.tmp
make[5]: Leaving directory '/media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/target/linux/mediatek/image'
make[4]: *** [Makefile:18: install] Error 2
make[4]: Leaving directory '/media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/target/linux/mediatek'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/target/linux'
time: target/linux/install#37.68#23.34#5.89
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source'
make[1]: *** [target/Makefile:24: /media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source'
make: *** [/media/yon/p0/omr1/openmptcprouter/BPI-R4/6.1/source/include/toplevel.mk:232：world] 

I replaced the target/linux/mediatek/image/filogic.mk, but the compilation failed.

Yon2068, 2024-09-25:
> I've updated filogik.mk on my repo, so you should be able to compile for BPI-R4 now.

--------

install: creating uninstall script at /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rust-src'

    rust src installed.

install: creating uninstall script at /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/host/lib/rustlib/uninstall.sh
install: installing component 'rust-std-aarch64-unknown-linux-musl'

    rust std installed.

install: creating uninstall script at /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/host/lib/rustlib/uninstall.sh
install: installing component 'build-manifest'

    build manifest installed.

install: creating uninstall script at /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/host/lib/rustlib/uninstall.sh
install: installing component 'cargo'

    cargo installed.

mkdir -p /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/staging_dir/hostpkg/stamp
touch /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/host/rustc-1.78.0-src/.built
touch /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/staging_dir/hostpkg/stamp/.rust_installed 
find /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/host/rustc-1.78.0-src -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -print0 | xargs -r -0 rm -rf
make[3]: Leaving directory '/media/yon/p0/omr2/openmptcprouter/feeds/openmptcprouter/rust'
time: package/feeds/openmptcprouter/rust/host-compile#16938.33#1717.20#1876.33
make[2]: Leaving directory '/media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source'
make[1]: *** [package/Makefile:171: /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source'
make: *** [/media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/include/toplevel.mk:248：world]

 pulled the latest code and used the command OMR_TARGET="BPI-R4" OMR_FEED_SRC="develop" OMR_KERNEL="6.6" ./build.sh to compile, but it returned an error.

Yon2068, 2024-09-26:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-thermal for kmod-hwmon-pwmfan
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-hwmon-pwmfan found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-hwmon-pwmfan.
 * resolve_conffiles: Existing conffile /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/config/glorytun is different from the conffile in the new package. The new conffile will be placed at /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/etc/config/glorytun-opkg.
make[2]: *** [package/Makefile:99: package/install] Error 255
make[2]: Leaving directory '/media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source'
make[1]: *** [package/Makefile:172: /media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source'
make: *** [/media/yon/p0/omr2/openmptcprouter/BPI-R4/6.6/source/include/toplevel.mk:248：world
---------

I'm stuck here. 
This is a similar case I found: https://github.com/openwrt/openwrt/pull/16400. 
What should I do now?

Ysurac, 2024-09-26:
Check that ```CONFIG_THERMAL``` is set to yes via ```make kernel_menuconfig``` in BPI-R4/6.6/source directory (use the key "/" to search).

Yon2068, 2024-09-26:
> Check that is set to yes via in BPI-R4/6.6/source directory (use the key "/" to search).`CONFIG_THERMAL``make kernel_menuconfig`

-------
I checked the .config file and only found CONFIG_PACKAGE_kmod-thermal=m. Shouldn't it be changed to y?

Yon2068, 2024-09-26:
┌─                                                                           Search Results ─                                                                          ┐
  │ Symbol: THERMAL [=y]                                                                                                                                                 │  
  │ Type  : bool                                                                                                                                                         │  
  │ Defined at drivers/thermal/Kconfig:6                                                                                                                                 │  
  │   Prompt: Thermal drivers                                                                                                                                            │  
  │   Location:                                                                                                                                                          │  
  │     -> Device Drivers                                                                                                                                                │  
  │ (1)   -> Thermal drivers (THERMAL [=y])                                                                                                                              │  
  │ Selected by [n]:                                                                                                                                                     │  
  │   - ACPI_VIDEO [=n] && ACPI [=n] && BACKLIGHT_CLASS_DEVICE [=n] && INPUT [=n] && (ACPI_WMI [=n] || !X86)                                                             │  
  │   - ACPI_PROCESSOR [=n] && ACPI [=n] && (X86 || IA64 || ARM64 [=y] || LOONGARCH)                                                                                     │  
  │   - ACPI_THERMAL [=n] && ACPI [=n] && ACPI_PROCESSOR [=n]                                                                                                            │  
  │   - AB8500_BM [=n] && POWER_SUPPLY [=y] && AB8500_CORE [=n] && AB8500_GPADC [=n] && IIO [=n]=y [=y] && OF [=y]                                                       │  
  │   - DRM_NOUVEAU [=n] && HAS_IOMEM [=y] && DRM [=n] && PCI [=y] && MMU [=y] && ACPI [=n] && X86                                                                       │  
  │   - DRM_ETNAVIV [=n] && HAS_IOMEM [=y] && DRM [=n] && MMU [=y] && DRM_ETNAVIV_THERMAL [=n]                                                                           │  
  │   - MMC_SDHCI_OMAP [=n] && MMC [=y] && (ARCH_OMAP2PLUS || ARCH_KEYSTONE || COMPILE_TEST [=n]) && MMC_SDHCI_PLTFM [=n] && OF [=y]                                     │  
  │ Implied by [n]:                                                                                                                                                      │  
  │   - SENSORS_DELL_SMM [=n] && HWMON [=y] && X86                                                                                                                       │  
  │   - SENSORS_MLXREG_FAN [=n] && HWMON [=y] && MELLANOX_PLATFORM [=n]                                                                                                  │  
  │   - SENSORS_NPCM7XX [=n] && HWMON [=y]                                                                                                                               │  
  │   - SENSORS_EMC2305 [=n] && HWMON [=y] && I2C [=y] 

-----------

 THERMAL [=y]     
 Isn't CONFIG_THERMAL already set to Y here?

Yon2068, 2024-09-26:
> Check that is set to yes via in BPI-R4/6.6/source directory (use the key "/" to search).`CONFIG_THERMAL``make kernel_menuconfig`

  │ Symbol: MTK_THERMAL [=y]                                                                                                                                             │  
  │ Type  : tristate                                                                                                                                                     │  
  │ Defined at drivers/thermal/mediatek/Kconfig:1                                                                                                                        │  
  │   Prompt: MediaTek thermal drivers                                                                                                                                   │  
  │   Depends on: THERMAL [=y] && (ARCH_MEDIATEK [=y] || COMPILE_TEST [=n]) && THERMAL_OF [=y]                                                                           │  
  │   Location:                                                                                                                                                          │  
  │     -> Device Drivers                                                                                                                                                │  
  │       -> Thermal drivers (THERMAL [=y])                                                                                                                              │  
  │         -> Mediatek thermal drivers                                                                                                                                  │  
  │ (4)       -> MediaTek thermal drivers (MTK_THERMAL [=y])

Yon2068, 2024-09-26:
#
# Automatically generated file; DO NOT EDIT.
# OpenWrt Configuration
#
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
CONFIG_HOST_OS_LINUX=y
# CONFIG_HOST_OS_MACOS is not set
# CONFIG_TARGET_airoha is not set
# CONFIG_TARGET_d1 is not set
# CONFIG_TARGET_sunxi is not set
# CONFIG_TARGET_apm821xx is not set
# CONFIG_TARGET_armsr is not set
# CONFIG_TARGET_ath79 is not set
# CONFIG_TARGET_bcm27xx is not set
# CONFIG_TARGET_bcm53xx is not set
# CONFIG_TARGET_bcm47xx is not set
# CONFIG_TARGET_bcm4908 is not set
# CONFIG_TARGET_bmips is not set
# CONFIG_TARGET_octeon is not set
# CONFIG_TARGET_gemini is not set
# CONFIG_TARGET_mpc85xx is not set
# CONFIG_TARGET_mxs is not set
# CONFIG_TARGET_ixp4xx is not set
# CONFIG_TARGET_lantiq is not set
# CONFIG_TARGET_loongarch64 is not set
# CONFIG_TARGET_malta is not set
# CONFIG_TARGET_pistachio is not set
# CONFIG_TARGET_mvebu is not set
# CONFIG_TARGET_kirkwood is not set
CONFIG_TARGET_mediatek=y
# CONFIG_TARGET_ramips is not set
# CONFIG_TARGET_at91 is not set
# CONFIG_TARGET_tegra is not set
# CONFIG_TARGET_layerscape is not set
# CONFIG_TARGET_qoriq is not set
# CONFIG_TARGET_imx is not set
# CONFIG_TARGET_qualcommax is not set
# CONFIG_TARGET_ipq40xx is not set
# CONFIG_TARGET_ipq806x is not set
# CONFIG_TARGET_realtek is not set
# CONFIG_TARGET_rockchip is not set
# CONFIG_TARGET_sifiveu is not set
# CONFIG_TARGET_starfive is not set
# CONFIG_TARGET_archs38 is not set
# CONFIG_TARGET_omap is not set
# CONFIG_TARGET_uml is not set
# CONFIG_TARGET_zynq is not set
# CONFIG_TARGET_x86 is not set
# CONFIG_TARGET_mediatek_mt7622 is not set
# CONFIG_TARGET_mediatek_mt7623 is not set
# CONFIG_TARGET_mediatek_mt7629 is not set
CONFIG_TARGET_mediatek_filogic=y
# CONFIG_TARGET_MULTI_PROFILE is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_abt_asr3000 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_acelink_ew-7886cax is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_acer_predator-w6 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8612 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8614 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8622 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8632 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8733 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8733a is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8734 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_rt-ax59u is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_tuf-ax4200 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_tuf-ax6000 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3-mini is not set
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r4=y
[... 5225 lines cut ...]
# CONFIG_PACKAGE_luci-i18n-ttyd-nl is not set
CONFIG_PACKAGE_luci-i18n-ttyd-no=y
CONFIG_PACKAGE_luci-i18n-ttyd-pl=y
CONFIG_PACKAGE_luci-i18n-ttyd-pt=y
CONFIG_PACKAGE_luci-i18n-ttyd-pt-br=y
CONFIG_PACKAGE_luci-i18n-ttyd-ro=y
CONFIG_PACKAGE_luci-i18n-ttyd-ru=y
CONFIG_PACKAGE_luci-i18n-ttyd-sk=y
CONFIG_PACKAGE_luci-i18n-ttyd-sv=y
CONFIG_PACKAGE_luci-i18n-ttyd-tr=y
CONFIG_PACKAGE_luci-i18n-ttyd-uk=y
CONFIG_PACKAGE_luci-i18n-ttyd-vi=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-tw=y
CONFIG_PACKAGE_luci-i18n-uhttpd-ar=y
CONFIG_PACKAGE_luci-i18n-uhttpd-bg=y
CONFIG_PACKAGE_luci-i18n-uhttpd-bn=y
CONFIG_PACKAGE_luci-i18n-uhttpd-ca=y
CONFIG_PACKAGE_luci-i18n-uhttpd-cs=y
# CONFIG_PACKAGE_luci-i18n-uhttpd-da is not set
CONFIG_PACKAGE_luci-i18n-uhttpd-de=y
CONFIG_PACKAGE_luci-i18n-uhttpd-el=y
CONFIG_PACKAGE_luci-i18n-uhttpd-es=y
CONFIG_PACKAGE_luci-i18n-uhttpd-fi=y
CONFIG_PACKAGE_luci-i18n-uhttpd-

Yon2068, 2024-09-26:
`yon@debian:/media/yon/p0/omr3/openmptcprouter/BPI-R4/6.6/source$ cat .config
#
# Automatically generated file; DO NOT EDIT.
# OpenWrt Configuration
#
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
CONFIG_HOST_OS_LINUX=y
# CONFIG_HOST_OS_MACOS is not set
# CONFIG_TARGET_airoha is not set
# CONFIG_TARGET_d1 is not set
# CONFIG_TARGET_sunxi is not set
# CONFIG_TARGET_apm821xx is not set
# CONFIG_TARGET_armsr is not set
# CONFIG_TARGET_ath79 is not set
# CONFIG_TARGET_bcm27xx is not set
# CONFIG_TARGET_bcm53xx is not set
# CONFIG_TARGET_bcm47xx is not set
# CONFIG_TARGET_bcm4908 is not set
# CONFIG_TARGET_bmips is not set
# CONFIG_TARGET_octeon is not set
# CONFIG_TARGET_gemini is not set
# CONFIG_TARGET_mpc85xx is not set
# CONFIG_TARGET_mxs is not set
# CONFIG_TARGET_ixp4xx is not set
# CONFIG_TARGET_lantiq is not set
# CONFIG_TARGET_loongarch64 is not set
# CONFIG_TARGET_malta is not set
# CONFIG_TARGET_pistachio is not set
# CONFIG_TARGET_mvebu is not set
# CONFIG_TARGET_kirkwood is not set
CONFIG_TARGET_mediatek=y
# CONFIG_TARGET_ramips is not set
# CONFIG_TARGET_at91 is not set
# CONFIG_TARGET_tegra is not set
# CONFIG_TARGET_layerscape is not set
# CONFIG_TARGET_qoriq is not set
# CONFIG_TARGET_imx is not set
# CONFIG_TARGET_qualcommax is not set
# CONFIG_TARGET_ipq40xx is not set
# CONFIG_TARGET_ipq806x is not set
# CONFIG_TARGET_realtek is not set
# CONFIG_TARGET_rockchip is not set
# CONFIG_TARGET_sifiveu is not set
# CONFIG_TARGET_starfive is not set
# CONFIG_TARGET_archs38 is not set
# CONFIG_TARGET_omap is not set
# CONFIG_TARGET_uml is not set
# CONFIG_TARGET_zynq is not set
# CONFIG_TARGET_x86 is not set
# CONFIG_TARGET_mediatek_mt7622 is not set
# CONFIG_TARGET_mediatek_mt7623 is not set
# CONFIG_TARGET_mediatek_mt7629 is not set
CONFIG_TARGET_mediatek_filogic=y
# CONFIG_TARGET_MULTI_PROFILE is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_abt_asr3000 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_acelink_ew-7886cax is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_acer_predator-w6 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8612 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8614 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8622 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8632 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8733 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8733a is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8734 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_rt-ax59u is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_tuf-ax4200 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_tuf-ax6000 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3-mini is not set
[... 5224 lines cut ...]
CONFIG_PACKAGE_luci-i18n-ttyd-mr=y
CONFIG_PACKAGE_luci-i18n-ttyd-ms=y
# CONFIG_PACKAGE_luci-i18n-ttyd-nl is not set
CONFIG_PACKAGE_luci-i18n-ttyd-no=y
CONFIG_PACKAGE_luci-i18n-ttyd-pl=y
CONFIG_PACKAGE_luci-i18n-ttyd-pt=y
CONFIG_PACKAGE_luci-i18n-ttyd-pt-br=y
CONFIG_PACKAGE_luci-i18n-ttyd-ro=y
CONFIG_PACKAGE_luci-i18n-ttyd-ru=y
CONFIG_PACKAGE_luci-i18n-ttyd-sk=y
CONFIG_PACKAGE_luci-i18n-ttyd-sv=y
CONFIG_PACKAGE_luci-i18n-ttyd-tr=y
CONFIG_PACKAGE_luci-i18n-ttyd-uk=y
CONFIG_PACKAGE_luci-i18n-ttyd-vi=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-tw=y
CONFIG_PACKAGE_luci-i18n-uhttpd-ar=y
CONFIG_PACKAGE_luci-i18n-uhttpd-bg=y
CONFIG_PACKAGE_luci-i18n-uhttpd-bn=y
CONFIG_PACKAGE_luci-i18n-uhttpd-ca=y
CONFIG_PACKAGE_luci-i18n-uhttpd-cs=y
# CONFIG_PACKAGE_luci-i18n-uhttpd-da is not set
CONFIG_PACKAGE_luci-i18n-uhttpd-de=y
CONFIG_PACKAGE_luci-i18n-uhttpd-el=y
CONFIG_PACKAGE_luci-i18n-uhttpd

Yon2068, 2024-09-26:
```
#
# Automatically generated file; DO NOT EDIT.
# OpenWrt Configuration
#
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
CONFIG_HOST_OS_LINUX=y
# CONFIG_HOST_OS_MACOS is not set
# CONFIG_TARGET_airoha is not set
# CONFIG_TARGET_d1 is not set
# CONFIG_TARGET_sunxi is not set
# CONFIG_TARGET_apm821xx is not set
# CONFIG_TARGET_armsr is not set
# CONFIG_TARGET_ath79 is not set
# CONFIG_TARGET_bcm27xx is not set
# CONFIG_TARGET_bcm53xx is not set
# CONFIG_TARGET_bcm47xx is not set
# CONFIG_TARGET_bcm4908 is not set
# CONFIG_TARGET_bmips is not set
# CONFIG_TARGET_octeon is not set
# CONFIG_TARGET_gemini is not set
# CONFIG_TARGET_mpc85xx is not set
# CONFIG_TARGET_mxs is not set
# CONFIG_TARGET_ixp4xx is not set
# CONFIG_TARGET_lantiq is not set
# CONFIG_TARGET_loongarch64 is not set
# CONFIG_TARGET_malta is not set
# CONFIG_TARGET_pistachio is not set
# CONFIG_TARGET_mvebu is not set
# CONFIG_TARGET_kirkwood is not set
CONFIG_TARGET_mediatek=y
# CONFIG_TARGET_ramips is not set
# CONFIG_TARGET_at91 is not set
# CONFIG_TARGET_tegra is not set
# CONFIG_TARGET_layerscape is not set
# CONFIG_TARGET_qoriq is not set
# CONFIG_TARGET_imx is not set
# CONFIG_TARGET_qualcommax is not set
# CONFIG_TARGET_ipq40xx is not set
# CONFIG_TARGET_ipq806x is not set
# CONFIG_TARGET_realtek is not set
# CONFIG_TARGET_rockchip is not set
# CONFIG_TARGET_sifiveu is not set
# CONFIG_TARGET_starfive is not set
# CONFIG_TARGET_archs38 is not set
# CONFIG_TARGET_omap is not set
# CONFIG_TARGET_uml is not set
# CONFIG_TARGET_zynq is not set
# CONFIG_TARGET_x86 is not set
# CONFIG_TARGET_mediatek_mt7622 is not set
# CONFIG_TARGET_mediatek_mt7623 is not set
# CONFIG_TARGET_mediatek_mt7629 is not set
CONFIG_TARGET_mediatek_filogic=y
# CONFIG_TARGET_MULTI_PROFILE is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_abt_asr3000 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_acelink_ew-7886cax is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_acer_predator-w6 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8612 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8614 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8622 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8632 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8733 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8733a is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_smartrg_sdg-8734 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_rt-ax59u is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_tuf-ax4200 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_asus_tuf-ax6000 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3 is not set
# CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3-mini is not set
[... 5226 lines cut ...]
# CONFIG_PACKAGE_luci-i18n-ttyd-nl is not set
CONFIG_PACKAGE_luci-i18n-ttyd-no=y
CONFIG_PACKAGE_luci-i18n-ttyd-pl=y
CONFIG_PACKAGE_luci-i18n-ttyd-pt=y
CONFIG_PACKAGE_luci-i18n-ttyd-pt-br=y
CONFIG_PACKAGE_luci-i18n-ttyd-ro=y
CONFIG_PACKAGE_luci-i18n-ttyd-ru=y
CONFIG_PACKAGE_luci-i18n-ttyd-sk=y
CONFIG_PACKAGE_luci-i18n-ttyd-sv=y
CONFIG_PACKAGE_luci-i18n-ttyd-tr=y
CONFIG_PACKAGE_luci-i18n-ttyd-uk=y
CONFIG_PACKAGE_luci-i18n-ttyd-vi=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-tw=y
CONFIG_PACKAGE_luci-i18n-uhttpd-ar=y
CONFIG_PACKAGE_luci-i18n-uhttpd-bg=y
CONFIG_PACKAGE_luci-i18n-uhttpd-bn=y
CONFIG_PACKAGE_luci-i18n-uhttpd-ca=y
CONFIG_PACKAGE_luci-i18n-uhttpd-cs=y
# CONFIG_PACKAGE_luci-i18n-uhttpd-da is not set
CONFIG_PACKAGE_luci-i18n-uhttpd-de=y
CONFIG_PACKAGE_luci-i18n-uhttpd-el=y
CONFIG_PACKAGE_luci-i18n-uhttpd-es=y
CONFIG_PACKAGE_luci-i18n-uhttpd-fi=y
CONFIG_PACKAGE_luci-i18n-uhtt

Yon2068, 2024-09-26:
```
Error: ../dts/mt7988d-smartrg-SDG-8733A.dts:114.1-15 Label or path aqr_green_led not found
Error: ../dts/mt7988d-smartrg-SDG-8733A.dts:118.1-16 Label or path aqr_orange_led not found
Error: ../dts/mt7988d-smartrg-SDG-8733A.dts:122.1-15 Label or path aqr_white_led not found
FATAL ERROR: Syntax error parsing input tree
make[5]: *** [Makefile:43: /media/yon/p0/omr3/openmptcprouter/BPI-R4/6.6/source/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/image-mt7988d-smartrg-SDG-8733A.dtb] Error 1
```
I found the missing file in the dts folder of OpenWrt, and I copied it to the source/target/linux/mediatek/dts directory. However, I noticed that it gets automatically deleted during compilation. What should I do?

Ysurac, 2024-09-27:
```git pull```. It was fixed yesterday.

Yon2068, 2024-09-27:
> `git pull`. It was fixed yesterday.

Thank you, I successfully compiled it, but after starting, the network card light is off, and I can't access 192.168.100.1. The onboard green indicator light came on, and the RST button is responsive; the green light goes off and then turns back on (possibly rebooting).

Ysurac, 2024-09-27:
As it's not a supported platform, difficult to help as I don't have the device.
You can try to connect a serial USB cable to check the boot screen ( https://wiki.banana-pi.org/Getting_Started_with_BPI-R4 )

Yon2068, 2024-09-27:
```
[    2.379335] mtk-pcie-gen3 11280000.pcie: Parsing ranges property...
[    2.384649]  mmcblk0: p1 p2 p3 p4 p5 p6 p7
[    2.390893] mtk-pcie-gen3 11280000.pcie:       IO 0x0020000000..0x00201fffff -> 0x0020000000
[    2.403412] mtk-pcie-gen3 11280000.pcie:      MEM 0x0020200000..0x0027ffffff -> 0x0020200000
[    2.751567] mtk-pcie-gen3 11280000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    2.760378] mtk-pcie-gen3: probe of 11280000.pcie failed with error -110
[    2.767460] mtk-pcie-gen3 11290000.pcie: host bridge /soc/pcie@11290000 ranges:
[    2.774777] mtk-pcie-gen3 11290000.pcie: Parsing ranges property...
[    2.781042] mtk-pcie-gen3 11290000.pcie:       IO 0x0028000000..0x00281fffff -> 0x0028000000
[    2.789478] mtk-pcie-gen3 11290000.pcie:      MEM 0x0028200000..0x002fffffff -> 0x0028200000
[    3.131563] mtk-pcie-gen3 11290000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.140344] mtk-pcie-gen3: probe of 11290000.pcie failed with error -110
[    3.147211] mtk-pcie-gen3 11300000.pcie: host bridge /soc/pcie@11300000 ranges:
[    3.154522] mtk-pcie-gen3 11300000.pcie: Parsing ranges property...
[    3.160781] mtk-pcie-gen3 11300000.pcie:       IO 0x0030000000..0x00301fffff -> 0x0030000000
[    3.169214] mtk-pcie-gen3 11300000.pcie:      MEM 0x0030200000..0x0037ffffff -> 0x0030200000
[    3.511570] mtk-pcie-gen3 11300000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.520349] mtk-pcie-gen3: probe of 11300000.pcie failed with error -110
[    3.527211] mtk-pcie-gen3 11310000.pcie: host bridge /soc/pcie@11310000 ranges:
[    3.534519] mtk-pcie-gen3 11310000.pcie: Parsing ranges property...
[    3.540778] mtk-pcie-gen3 11310000.pcie:       IO 0x0038000000..0x00381fffff -> 0x0038000000
[    3.549210] mtk-pcie-gen3 11310000.pcie:      MEM 0x0038200000..0x003fffffff -> 0x0038200000
[    3.891565] mtk-pcie-gen3 11310000.pcie: PCIe link down, current LTSSM state: detect.quiet (0x1)
[    3.900344] mtk-pcie-gen3: probe of 11310000.pcie failed with error -110
[    4.293051] clk: Disabling unused clocks
[    4.297613] Waiting for root device /dev/fit0...
[   14.682967] platform 15020000.switch: deferred probe pending
[   14.688624] platform 15100000.ethernet: deferred probe pending

```
Here is TTL information. What should I do?

Ysurac, 2024-09-27:
Seems to be similar to https://github.com/openwrt/openwrt/issues/14767
Current OpenWRT snapshot is working ?

Yon2068, 2024-09-27:
> Seems to be similar to [openwrt/openwrt#14767](https://github.com/openwrt/openwrt/issues/14767) Current OpenWRT snapshot is working ?
---
Yes, the current OpenWrt snapshot is working fine.

Ysurac, 2024-09-27:
Without same TTL messages ?
Can you do also a ```uci show network``` and ```ip a``` on OpenWRT image ?

Yon2068, 2024-09-27:
> Without same TTL messages ? Can you do also a and on OpenWRT image ?`uci show network``ip a`

-----
I am using the official OpenWRT image, which can boot normally: https://downloads.openwrt.org/snapshots//targets/mediatek/filogic/openwrt-mediatek-filogic-bananapi_bpi-r4-sdcard.img.gz

Yon2068, 2024-09-27:
> Without same TTL messages ? Can you do also a and on OpenWRT image ?`uci show network``ip a`

----
```
root@OpenWrt:~# uci show network
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fd35:1493:650a::/48'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='lan1' 'lan2' 'lan3' 'eth1'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.1.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.@device[1]=device
network.@device[1].name='br-wan'
network.@device[1].type='bridge'
network.@device[1].ports='wan' 'eth2'
network.@device[2]=device
network.@device[2].name='wan'
network.@device[2].macaddr='d2:c3:9a:92:64:a6'
network.@device[3]=device
network.@device[3].name='eth2'
network.@device[3].macaddr='d2:c3:9a:92:64:a6'
network.wan=interface
network.wan.device='br-wan'
network.wan.proto='dhcp'
network.wan6=interface
network.wan6.device='br-wan'
network.wan6.proto='dhcpv6'
root@OpenWrt:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP qlen 1000
    link/ether d2:c3:9a:92:64:a5 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::d0c3:9aff:fe92:64a5/64 scope link 
       valid_lft forever preferred_lft forever
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN qlen 1000
    link/ether 96:f8:5e:d1:8a:f8 brd ff:ff:ff:ff:ff:ff
4: eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-wan state DOWN qlen 1000
    link/ether d2:c3:9a:92:64:a6 brd ff:ff:ff:ff:ff:ff
5: wan@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-wan state LOWERLAYERDOWN qlen 1000
    link/ether d2:c3:9a:92:64:a6 brd ff:ff:ff:ff:ff:ff
6: lan1@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether d2:c3:9a:92:64:a5 brd ff:ff:ff:ff:ff:ff
7: lan2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether d2:c3:9a:92:64:a5 brd ff:ff:ff:ff:ff:ff
8: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN qlen 1000
    link/ether d2:c3:9a:92:64:a5 brd ff:ff:ff:ff:ff:ff
9: br-lan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN qlen 1000
    link/ether d2:c3:9a:92:64:a5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fd35:1493:650a::1/60 scope global tentative noprefixroute 
       valid_lft forever preferred_lft forever
10: br-wan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN qlen 1000
    link/ether d2:c3:9a:92:64:a6 brd ff:ff:ff:ff:ff:ff
root@OpenWrt:~# 
```

Yon2068, 2024-09-27:
```
root@OpenWrt:~# uname -a
Linux OpenWrt 6.6.52 #0 SMP Thu Sep 19 21:11:58 2024 aarch64 GNU/Linux
```

Yon2068, 2024-09-28:
The official kernel 6.6.52 is working properly. How can I use the 6.6.52 kernel?
@Ysurac

Ysurac, 2024-10-01:
For my address, it's in legal notice: https://www.openmptcprouter.com/about
OpenMPTCProuter can't be a simple package for OpenWRT, there is some kernel patch and many fork.

loriszigliani, 2024-10-07:
I need help when i compile for platform BPI-R4 with kernel 6.6, show this error:

from this command "make kernel_menuconfig":
ERROR: toolchain/binutils failed to build.
make: *** [/home/loriszigliani/openmptcprouter/BPI-R4/6.6/source/include/toplevel.mk:248: toolchain/install] Error 1 from this command "make kernel_menuconfig"

on command "make -j1"
[screenlog.txt](https://github.com/user-attachments/files/17276952/screenlog.txt)

ERROR: package/kernel/bpf-headers failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/loriszigliani/openmptcprouter/BPI-R4/6.6/source/include/toplevel.mk:248: world] Error 1

esquilo444, 2024-11-03:
> Have you tried compiling for kernel 6.1?
>
> I had trouble compiling 6.6 for BPI-R3 but got 6.1 to compile.
>
> I don't know about the R4 but for the R3 I had to disable ramdisk, otherwise, it wouldn't [...]

Hello BlueWaterCrystal,

I have a bpi r3 and unfortunately I don't know how to compile. If you accept it, I would be happy to test your omr image for BPI-R3. I prefer to do things myself, but I don't have the skills. thank you in advance for your response. Jerome

BlueWaterCrystal, 2024-11-05:
> Hello BlueWaterCrystal,
>
> I have a bpi r3 and unfortunately I don't know how to compile. If you accept it, I would be happy to test your omr image for BPI-R3. I prefer to do things myself, but I [...]

Unfortunately the file is too big to upload, however here is a list of dependencies that are installed on a Debian 12 machine that may help with some errors:
[installed_packages.txt](https://github.com/user-attachments/files/17636011/installed_packages.txt)

```
sudo dpkg --set-selections < installed_packages.txt
sudo apt-get update
sudo apt-get dselect-upgrade
```

Additionally, weirdly `make -j6` fails but `make -j6 V=s` completes... 

v0.61 on kernel 6.1 works but It seems VPN bonding and IPv6 are broken.
I've managed to compile v0.62 on kernel 6.6; however, for some reason, the GPT partitions are broken.

esquilo444, 2024-11-05:
Good morning, first of all thank you very much for taking time for me. I tried again this morning with a Debian 12, but without success, a lot of errors... unfortunately my knowledge of Linux is very limited, I know how to analyze the problems and follow the tutorials... but that's it for Linux. I have never compiled anything on Linux. I am a video technician specialized in streaming, so video, networks ok but much less Linux... FYI, I don't use VPNs... just vray less as a proxy for TCP transport and mainly UDP which is the protocol used for the SRT video stream.. If I send you my email, can you consider sending SMASH, or wetransfer...? I just need the img image for the SD card, to test already. ***@***.*** many thanks in advance. Jerome PS one day I will get there on my own 😅

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

saltyautomation, 2024-11-05:
> @Ysurac roughly where are you located - I would be more than willing to supply you with an BPI-R4 unit. I just bought one thinking I could use it for network bonding, initially thinking Speedify would do the trick, but I am hating the concept of the Per Month price associated with it, so OpenMPT seems a far better option. Plus I may eventually have a need for more than one of these...so would be more than happy to ship this one to you, or buy another and drop-ship it - since right now I can't do much with the current unit other than fiddle around with OpenWRT...Tried booting with R64 image and it seems a no-go...
> 
> In the meantime I will build some test solutions using the X86 images for some 4 Port Qotom Mini-PCs I have which will also eventually be used in production, and link them to my VPS to do some testing.
> 
> Unless there is simply a means to use OpenWRT itself on the BPI-R4 - and install OpenMPT as a Package just like how Speedify installs ?!

Did you ship him a unit? If not I have BPI R3 that I can ship not sure how much that will be forward compatible with R4 board though. 
Let me know happy to send him mine as it has been sitting for a while.

Ysurac, 2024-11-05:
I didn't get any device.
BPI-R3 is not same CPU as BPI-R4.
I can try to build images for BPI-R4 but difficult then to fix any issues without an unit. But I can try and we will see.

esquilo444, 2024-11-05:
For my part, it is the BPI-R3 not the R4. I do not know where you live Ysurac, I live in Nantes and if you want I can lend you the same, without a functional openmptcp router image, I will not not do much with it. As T Morton managed to compile an image in kernel 6.1, I'm ready to get it to test it on my side..

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

BlueWaterCrystal, 2024-11-05:
> Nantes

> Good morning, first of all thank you very much for taking time for me. I tried again this morning with a Debian 12, but without success, a lot of errors... unfortunately my knowledge of Linux is [...]

Here are the BPI-R3 test files:
[BPI-R3-openmptcprouter](https://github.com/BlueWaterCrystal/BPI-R3-openmptcprouter.git)

LMK, if it works.

When it comes to SRT, I think It acts similarly to TCP over TCP, as SRT and TCP are both retransmitting packets at the same time causing x2 in retransmission traffic.

esquilo444, 2024-11-05:
Thank you, you are the best 😋. I should receive my BPI-R3 tomorrow or the day after tomorrow. I will remain so as soon as possible and I will be able to provide feedback on my observations based on my use... once again thank you.

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

itsgps, 2024-11-05:
> > @Ysurac roughly where are you located - I would be more than willing to supply you with an BPI-R4 unit. I just bought one thinking I could use it for network bonding, initially thinking Speedify would do the trick, but I am hating the concept of the Per Month price associated with it, so OpenMPT seems a far better option. Plus I may eventually have a need for more than one of these...so would be more than happy to ship this one to you, or buy another and drop-ship it - since right now I can't do much with the current unit other than fiddle around with OpenWRT...Tried booting with R64 image and it seems a no-go...
> > In the meantime I will build some test solutions using the X86 images for some 4 Port Qotom Mini-PCs I have which will also eventually be used in production, and link them to my VPS to do some testing.
> > Unless there is simply a means to use OpenWRT itself on the BPI-R4 - and install OpenMPT as a Package just like how Speedify installs ?!
> 
> @itsgps Did you ship him a unit? If not I have BPI R3 that I can ship not sure how much that will be forward compatible with R4 board though. Let me know happy to send him mine as it has been sitting for a while.

Apologies - no I didn't I became extremely busy after that message and simply have had Zero time to attend to it... but subsequent to tha tin the past week now dealing with a family health emergency.

esquilo444, 2024-11-05:
"LMK, if it works.

When it comes to SRT, I think It acts similarly to TCP over TCP, as SRT and TCP are both retransmitting packets at the same time causing x2 in retransmission traffic..'

this phenomenon of doubled flow is present if we send an SRT stream, therefore UDP using any proxy for TCP, and glorytun UDP for UDP. In all my tests the best understood and the most functional is, no VPN and a vray less proxy with UDP support by vray... So no doubling...

I am looking for optimal stability... it is the crux of the matter in video transmission, as I use exclusively 4g lines, it is not easy... But I am almost reaching my goal. .. I continue to optimize the hardware and software (thanks to you for this part)

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

Ysurac, 2024-11-09:
You can try images for BPI-R4 and BPI-R3 in https://snapshots.openmptcprouter.com/6.6/

loriszigliani, 2024-11-11:
I've just tested file for rpi4 "openmptcprouter-v0.62-snapshot-6.6-r0+28016-48028cd102-mediatek-filogic-bananapi_bpi-r4-sdcard" form sd card and it say that is an error on pci-gen3 
[banana-rpi4.txt](https://github.com/user-attachments/files/17706899/banana-rpi4.txt)
I can't do more beacuse i don't know os low level kernel.

BlueWaterCrystal, 2024-11-20:
> You can try images for BPI-R4 and BPI-R3 in https://snapshots.openmptcprouter.com/6.6/

Unfortunately I was unable to boot 6.6 or 5.4 on the BPI-R3.
I managed to compile a working 0.62v for 6.1 but I suspect most of the fixes require 6.6 as IPv6 doesn't work.

> I've just tested file for rpi4 "openmptcprouter-v0.62-snapshot-6.6-r0+28016-48028cd102-mediatek-filogic-bananapi_bpi-r4-sdcard" form sd card and it say that is an error on pci-gen3 [...]

I'm getting the same PCIE error on the BPI-R3,
However I think there is some issue with the bootloader/partitions, that's preventing it from booting. 

```
Alternate GPT is invalid, using primary GPT.
mmcblk0: p1 p2 p3 p4 p5 p6 p7
Waiting for root device /dev/fit0...
random: crng init done
```

Ysurac, 2024-11-20:
You have BPI-R3 or BPI-R3 mini ? For BPI-R3 mini a specific image is needed.

BlueWaterCrystal, 2024-11-20:
> You have BPI-R3 or BPI-R3 mini ? For BPI-R3 mini a specific image is needed.

I have the [BPI-R3](https://wiki.banana-pi.org/Banana_Pi_BPI-R3)

Another thing, by default 1920-omr-network seems to set `eth1` as default lan which is an SFP port.
Ideal if that's your goal connector otherwise has to be manually changed.

itsgps, 2024-12-31:
> > You can try images for BPI-R4 and BPI-R3 in https://snapshots.openmptcprouter.com/6.6/
> 
> Unfortunately I was unable to boot 6.6 or 5.4 on the BPI-R3. I managed to compile a working 0.62v for 6.1 but I guessing most of the fixes require 6.6 as IPv6 doesn't work.
> 
> > I've just tested file for rpi4 "openmptcprouter-v0.62-snapshot-6.6-r0+28016-48028cd102-mediatek-filogic-bananapi_bpi-r4-sdcard" form sd card and it say that is an error on pci-gen3 [banana-rpi4.txt](https://github.com/user-attachments/files/17706899/banana-rpi4.txt) I can't do more beacuse i don't know os low level kernel.
> 
> I'm getting the same PCIE error on the BPI-R3, However I think there is some issue with the bootloader/partitions, that's preventing it from booting.
> 
> ```
> Alternate GPT is invalid, using primary GPT.
> mmcblk0: p1 p2 p3 p4 p5 p6 p7
> Waiting for root device /dev/fit0...
> random: crng init done
> ```

I got to the same point with the R4 - "waiting for root device /dev/fit0" - guessing the R4 doesn't acknowledge that device.  same result if I get it to do the SD install to EMMC/NAND

itsgps, 2024-12-31:
From my last - found the fix in part...at least to allow troubleshooting further.
During boot - Choose **Exit** during the boot process.
you will be in a shell.
do "printenv" - it will show a set of Variables - including "bootargs" - in there is the root=/dev/fit0 rootwait
if you copy that line to notepad, remove the rootwait
then paste in your console:
setenv bootarg "console=etc.etc.etc..." - i.e. the same text you now had in notepad.
then
saveenv
(I also did the following:
setenv bootdelay=60 and saved that too - just to give me time to think during boots.

So - this does then boot, although it keeps looping due to kernel image issues - presumably we can't find the /dev/fit0 where the Kernel is located. 
But at least it gives a start to troubleshoot some more.

Yon2068, 2025-01-11:
image is not working

packlet, 2025-02-07:
> I only support hardware that I own.

I have a feeling you'll be the proud owner of a BPI-R4 very soon...  :-)   de rien.

Ysurac, 2025-02-12:
Thanks, I just received it. I will begin to work on it asap

Ysurac, 2025-02-19:
The image here should work now: https://snapshots.openmptcprouter.com/6.6/bpi-r4/targets/mediatek/filogic/

[5 more replies]

closed 2025-05-26 by github-actions[bot]