Ysurac/openmptcprouter #3549: Problem with quectel detection on RPI5

yannick2801, 2024-09-23:
when i start my OPENMPTCPROUTER with my 3 Quectel cards, i have this : 

```
root@OpenMPTCProuter:~# mmcli -L
    /org/freedesktop/ModemManager1/Modem/3 [Quectel] RM520N-GL
    /org/freedesktop/ModemManager1/Modem/0 [Quectel] RM520N-GL
    /org/freedesktop/ModemManager1/Modem/2 [Quectel] RM520N-GL
```
but after few minutes, i have only one left : 

```
root@OpenMPTCProuter:~# mmcli -L
    /org/freedesktop/ModemManager1/Modem/0 [Quectel] RM520N-GL
```
[image]
[image]

## Specifications

  - OpenMPTCProuter version: Linux OpenMPTCProuter 6.1.71 #0 SMP Tue Jan 16 14:55:02 2024 aarch64 GNU/Linux
  - OpenMPTCProuter VPS version: 0.1029-test 6.1.0-18cloud-amd64
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform: RPI5

BR 
Yannick

Ysurac, 2024-09-23:
Use latest snapshot release, and also check Status->System log

yannick2801, 2024-09-23:
i have used this version : 
[image]

it's not the last version release ? 

here is the system Log : 

Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3] enabled modem
Sep 23 13:15:46 OpenMPTCProuter daemon.notice [27350]: <msg> [modem3] state changed (enabling -> searching)
Sep 23 13:15:46 OpenMPTCProuter daemon.notice netifd: QUECTEL2 (4723): successfully enabled the modem
Sep 23 13:15:46 OpenMPTCProuter daemon.notice netifd: QUECTEL2 (4723): setting allowed mode to 'any'
Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3] processing user request to set current modes to 'any' (preferred 'none')...
Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3] current modes set to 'any' (preferred 'none')
Sep 23 13:15:46 OpenMPTCProuter daemon.notice netifd: QUECTEL2 (4723): successfully set current modes in the modem
Sep 23 13:15:46 OpenMPTCProuter daemon.notice netifd: QUECTEL2 (4723): starting connection with apn ''...
Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3] processing user request to connect modem...
Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3]   apn: 
Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3]   ip type: ipv4v6
Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3]   allowed auth: unknown
Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3]   allow roaming: yes
Sep 23 13:15:46 OpenMPTCProuter daemon.notice [27350]: <msg> [modem3] simple connect started...
Sep 23 13:15:46 OpenMPTCProuter daemon.notice [27350]: <msg> [modem3] simple connect state (4/10): wait to get fully enabled
Sep 23 13:15:46 OpenMPTCProuter daemon.notice [27350]: <msg> [modem3] simple connect state (5/10): wait after enabled
Sep 23 13:15:46 OpenMPTCProuter daemon.notice [27350]: <msg> [modem3] simple connect state (6/10): register
Sep 23 13:15:46 OpenMPTCProuter daemon.info [27350]: <inf> [modem3] launching automatic network registration...
Sep 23 13:15:55 OpenMPTCProuter daemon.err /usr/bin/ss-local[11982]: fast_open_connect: Network unreachable
Sep 23 13:16:00 OpenMPTCProuter user.notice post-tracking-001-post-tracking: No answer from QUECTEL3 (), restart interface
Sep 23 13:16:00 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Set QUECTEL3 down
Sep 23 13:16:00 OpenMPTCProuter user.notice post-tracking-001-post-tracking: No answer from QUECTEL1 (), restart interface
Sep 23 13:16:00 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Set QUECTEL1 down
Sep 23 13:16:00 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Sep 23 13:16:00 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Sep 23 13:16:01 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sep 23 13:16:01 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Sep 23 13:16:05 OpenMPTCProuter daemon.err /usr/bin/ss-local[11982]: fast_open_connect: Network unreachable
Sep 23 13:16:05 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Set QUECTEL3 up
Sep 23 13:16:05 OpenMPTCProuter daemon.notice netifd: Interface 'QUECTEL3' is setting up now
Sep 23 13:16:05 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Set QUECTEL1 up
Sep 23 13:16:05 OpenMPTCProuter daemon.notice netifd: Interface 'QUECTEL1' is setting up now
Sep 23 13:16:05 OpenMPTCProuter daemon.notice netifd: QUECTEL3 (6495): modem available at /org/freedesktop/ModemManager1/Modem/4
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem4] processing user request to enable modem...
Sep 23 13:16:05 OpenMPTCProuter daemon.notice [27350]: <msg> [modem4] state changed (disabled -> enabling)
Sep 23 13:16:05 OpenMPTCProuter daemon.notice netifd: QUECTEL1 (6513): modem available at /org/freedesktop/ModemManager1/Modem/1
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem4] checking if SIM was swapped...
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem4/sim4] processing user request to send PIN...
Sep 23 13:16:05 OpenMPTCProuter daemon.notice netifd: QUECTEL1 (6513): error: couldn't send PIN code to the SIM: 'GDBus.Error:org.freedesktop.ModemManager1.Error.Core.WrongState: Wrong state: Cannot send PIN: device is not SIM-PIN locked'
Sep 23 13:16:05 OpenMPTCProuter daemon.notice netifd: QUECTEL1 (6570): stopping network
Sep 23 13:16:05 OpenMPTCProuter daemon.notice netifd: QUECTEL1 (6570): couldn't load bearer path: disconnecting anyway
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem1] processing user request to disconnect modem: all bearers
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem1] all requested bearers disconnected
Sep 23 13:16:05 OpenMPTCProuter daemon.notice netifd: Interface 'QUECTEL1' is now down
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem4] SIM identifier has not changed: ###
Sep 23 13:16:05 OpenMPTCProuter daemon.notice [27350]: <msg> [modem4] power state updated: on
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem4] checking if SIM was swapped...
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem4] SIM imsi has not changed: ###
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem4] SIM identifier has not changed: ###
Sep 23 13:16:05 OpenMPTCProuter daemon.info [27350]: <inf> [modem4] SIM imsi has not changed: ###
[... cut ...]
Sep 23 13:33:13 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Can't find QUECTEL2 in ModemManager
Sep 23 13:33:48 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Can't find QUECTEL2 in ModemManager
Sep 23 13:34:23 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Can't find QUECTEL2 in ModemManager
Sep 23 13:34:25 OpenMPTCProuter daemon.err uhttpd[6201]: [info] luci: accepted login on / for root from 192.168.100.240
Sep 23 13:35:00 OpenMPTCProuter user.notice post-tracking-001-post-tracking: Can't find QUECTEL2 in ModemManager

BR
Yannick

yannick2801, 2024-09-24:
it's strange because it work's fine with only 2 Quectel Boards (2 sims)

[image]

Ysurac, 2024-09-24:
When I talk about latest snapshot release: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
I made some changes related to ModemManager, so it's better to test on this.

yannick2801, 2024-09-24:
new VPS version installed : (3 Quectel cards but only one works)

[image]

Ysurac, 2024-09-25:
You are using a powered USB hub ?

yannick2801, 2024-09-25:
Eatch Quectel board are on USB on the Rpi5 
I have the power to eatch Quectel board from USB but from PIN 4 & 6 from de Rpi5 too.

Do you think it's a power problem ?

BR
Yannick

Ysurac, 2024-09-25:
It's often the case with USB modems.

yannick2801, 2024-09-26:
what  should i try ? external power for the USB Modems ?

Ysurac, 2024-09-26:
Yes, a powered USB hub.

closed 2024-12-30 by github-actions[bot]