Ysurac/openmptcprouter #3550: ICMP, UDP and VPN only work on router, not LAN

BlueWaterCrystal, 2024-09-24:
Hi, 

I'm not sure what happened, but LAN clients cannot use ICMP or UDP, only TCP and when set to VPN only, there is no connectivity. 

However, the router itself works correctly, when I send a ping from the router I see it go via the VPN to the server.
Additionally, LAN can ping the router but not beyond. 

I've tried reinstalling the server but the issue persists.

## Expected Behavior
1. ICMP and UDP works for LAN Clients and Router.
2. LAN Clients to still have Internet access while only using VPN instead of Proxy.

## Current Behavior
1. ICMP and UDP only works for the Router.
2. VPN only gives the Router Internet but LAN Clients have no Internet access.

## Specifications

  - OpenMPTCProuter version: v0.61-snapshot-master-
  - OpenMPTCProuter kernel: 6.1.71
  - OpenMPTCProuter VPS version: 6.1.0-25-amd64 0.1030
  - OpenMPTCProuter VPS provider: Qubitbytes
  - OpenMPTCProuter platform: Bananapi BPI-R3

Ysurac, 2024-09-24:
Can you take a screenshot of status page ?

BlueWaterCrystal, 2024-09-24:
Hi,

Here's a screenshot of the status page:
[image]

Ysurac, 2024-09-25:
What is the VPN you use ?

BlueWaterCrystal, 2024-09-25:
OpenVPN TCP, I've also tried Glorytun TCP and the UDP version.

Ysurac, 2024-09-25:
Did you change any network configuration on LAN interface (apart the IP) ?
I would need, via SSH on the router, the result of ```uci show network``` and ```uci show firewall```.

Ysurac, 2024-09-26:
All seems ok.
What is the configuration on your client computer ?

BlueWaterCrystal, 2024-09-26:
All clients are affected, and all use DHCP.

When I do a ping from the diagnostics page, I see the packets go via `tun0` and all works fine.
However, when I ping from a client, I see the packet enter on the LAN port but it doesn't appear on 'tun0' and results in request timed out.

tcpdump of tun0: (ping via diagnostics page to www.openmptcprouter.com)

```
11:32:18.972972 IP 10.255.255.2 > 10.255.255.1: ICMP echo request, id 14161, seq 1, length 64
11:32:19.019249 IP 10.255.255.1 > 10.255.255.2: ICMP echo reply, id 14161, seq 1, length 64
11:32:24.255800 IP 10.255.255.2 > 143-244-38-136.bunnyinfra.net: ICMP echo request, id 36989, seq 1, length 64
11:32:24.282290 IP 143-244-38-136.bunnyinfra.net > 10.255.255.2: ICMP echo reply, id 36989, seq 1, length 64
11:32:24.290445 IP 10.255.255.2 > 10.255.255.1: ICMP echo request, id 18393, seq 1, length 64
11:32:24.319758 IP 10.255.255.1 > 10.255.255.2: ICMP echo reply, id 18393, seq 1, length 64
11:32:25.256853 IP 10.255.255.2 > 143-244-38-136.bunnyinfra.net: ICMP echo request, id 36989, seq 2, length 64
11:32:25.285642 IP 143-244-38-136.bunnyinfra.net > 10.255.255.2: ICMP echo reply, id 36989, seq 2, length 64
11:32:26.258512 IP 10.255.255.2 > 143-244-38-136.bunnyinfra.net: ICMP echo request, id 36989, seq 3, length 64
11:32:26.284823 IP 143-244-38-136.bunnyinfra.net > 10.255.255.2: ICMP echo reply, id 36989, seq 3, length 64
11:32:27.260520 IP 10.255.255.2 > 143-244-38-136.bunnyinfra.net: ICMP echo request, id 36989, seq 4, length 64
11:32:27.296849 IP 143-244-38-136.bunnyinfra.net > 10.255.255.2: ICMP echo reply, id 36989, seq 4, length 64
11:32:28.261570 IP 10.255.255.2 > 143-244-38-136.bunnyinfra.net: ICMP echo request, id 36989, seq 5, length 64
11:32:28.288616 IP 143-244-38-136.bunnyinfra.net > 10.255.255.2: ICMP echo reply, id 36989, seq 5, length 64
```

tcpdump of LAN - (client pinging www.openmptcprouter.com)
```
11:34:05.975133 IP 192.168.200.206 > host.fireflywhm.com: ICMP echo request, id 35, seq 51029, length 40
11:34:06.340655 IP 192.168.200.206 > a92-123-241-137.deploy.static.akamaitechnologies.com: ICMP echo request, id 58352, seq 121, length 9
11:34:06.447571 IP 192.168.200.206 > one.one.one.one: ICMP echo request, id 42715, seq 20282, length 9
11:34:06.453222 IP 192.168.200.206 > dns.google: ICMP echo request, id 42710, seq 20282, length 9
**11:34:06.464622 IP 192.168.200.206 > 143-244-38-136.bunnyinfra.net: ICMP echo request, id 35, seq 51033, length 1008** <----
11:34:06.480825 IP 192.168.200.206 > li-in-f94.1e100.net: ICMP echo request, id 35, seq 51035, length 40
```

tcpdump of LAN - (client pinging router)
```
**11:37:31.406019 IP 192.168.200.206 > openmptcprouter.lan: ICMP echo request, id 35, seq 52675, length 1008** <----
**11:37:31.406176 IP openmptcprouter.lan > 192.168.200.206: ICMP echo reply, id 35, seq 52675, length 1008** <----
11:37:31.469657 IP 192.168.200.206 > 216.119.69.148: ICMP echo request, id 35, seq 52677, length 40
11:37:31.469656 IP 192.168.200.206 > a23-52-18-110.deploy.static.akamaitechnologies.com: ICMP echo request, id 35, seq 52678, length 40
11:37:31.469656 IP 192.168.200.206 > 64.85.9.242: ICMP echo request, id 35, seq 52676, length 40
11:37:31.469657 IP 192.168.200.206 > 223.235.187.35.bc.googleusercontent.com: ICMP echo request, id 35, seq 52679, length 40
11:37:31.679910 IP 192.168.200.206 > one.one.one.one: ICMP echo request, id 62958, seq 4, length 8
11:37:31.679910 IP 192.168.200.206 > one.one.one.one: ICMP echo request, id 36433, seq 4, length 8
11:37:31.689916 IP 192.168.200.206 > dns.google: ICMP echo request, id 46483, seq 4, length 8
11:37:31.973752 IP 192.168.200.206 > 91.132.209.35.bc.googleusercontent.com: ICMP echo request, id 35, seq 52681, length 40
11:37:31.973752 IP 192.168.200.206 > 91.195.6.131: ICMP echo request, id 35, seq 52683, length 40
11:37:31.973752 IP 192.168.200.206 > a23-52-21-155.deploy.static.akamaitechnologies.com: ICMP echo request, id 35, seq 52682, length 40
11:37:31.973752 IP 192.168.200.206 > sc105.eboundhost.com: ICMP echo request, id 35, seq 52680, length 40
**11:37:32.417077 IP 192.168.200.206 > openmptcprouter.lan: ICMP echo request, id 35, seq 52684, length 1008** <----
**11:37:32.417235 IP openmptcprouter.lan > 192.168.200.206: ICMP echo reply, id 35, seq 52684, length 1008** <----
```

closed 2024-12-30 by github-actions[bot]