Ysurac/openmptcprouter #3552: RTL8168 NIC support

FermedePommerieux, 2024-09-25:
## Expected Behavior
RTL8168 NIC support

since 0.61 the second ethernet port of my dfrobot routerboard is not recognized, does RTL8168 kernel module included ?

## Current Behavior

## Possible Solution

add 8168 to linux kernel

## Steps to Reproduce the Problem

## Context (Environment)

use 0.61 on dfrobot routerboard with CM4

## Specifications

  - OpenMPTCProuter version:.0,61
  - OpenMPTCProuter platform: CM4 dfrobot router board

Ysurac, 2024-09-26:
This driver was removed because N100 wasn't booting correctly with it (no idea why).
I will add it back for anything that is not x86_64

FermedePommerieux, 2024-09-26:
top merci

closed 2024-12-31 by github-actions[bot]

Slikpro, 2025-04-27:
Good day! Could you please share the compiled r8168.ko module for the latest current x86_64 ext4.efi.img build. Some low-budget industrial computers have an old Realtek RTL8168 network chip on their motherboard. You mentioned that there is instability only with relatively new intel processors. In older builds I found only r8189 but it doesn't fit. and in Ubuntu the network card is detected as and pulls up k8168.ko.