Ysurac/openmptcprouter #3553: Working PPPOE Connection is shown as down

dersch81, 2024-09-25:
I have an issue with a Telekom PPPOE connection. In the Status it is marked as down, but the uplink is working and i can see traffic running over it. How can i get a green check on this?

[image]

[image]

## Specifications

I have to run last rc with Kernel 6.6 because of the N100 support. 

  - OpenMPTCProuter version: v0.61rc2
  - OpenMPTCProuter VPS version: 0.1031-test
  - OpenMPTCProuter VPS provider: HT-Hosting.de
  - OpenMPTCProuter platform: x86_64
  - Country: Germany

## ifstatus PPPOE

And how can i get the IPv6 working on this PPPOE? The ISP will provide me public routable IPv6 as well as a /56 via DHCPv6

```
root@OpenMPTCProuter:~# ifstatus wan6
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 1285,
        "l3_device": "pppoe-wan6",
        "proto": "pppoe",
        "device": "eth2",
        "updated": [
                "addresses",
                "routes"
        ],
        "metric": 14,
        "dns_metric": 0,
        "delegation": false,
        "ipv4-address": [
                {
                        "address": "84.155.146.195",
                        "mask": 32,
                        "ptpaddress": "62.155.246.179"
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [
                {
                        "address": "fdde:43da:e68c::",
                        "mask": 56,
                        "local-address": {
                                "address": "fdde:43da:e68c::1",
                                "mask": 56
                        }
                }
        ],
        "route": [

        ],
        "dns-server": [

        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [
                        {
                                "target": "0.0.0.0",
                                "mask": 0,
                                "nexthop": "62.155.246.179",
                                "source": "0.0.0.0/0"
                        }
                ],
                "dns-server": [
                        "217.237.150.51",
                        "217.237.148.22"
                ],
                "dns-search": [

                ],
                "neighbors": [

                ]
        },
        "data": {

        }
}
```

Ysurac, 2024-09-26:
What do you have in Status->System log ?
What is the result, on the router via SSH, of ```ping -B -I pppoe-wan6 1.1.1.1``` ?

dersch81, 2024-09-26:
I'm reverted the PPPOE back to my pfSense for productive use. I reinstalled the OMR to start with a fresh setup but i'm facing issues and asking for help https://github.com/Ysurac/openmptcprouter/issues/3556

therefore i close this unresolved and need general help with strange issues.

dersch81, 2024-09-29:
Now i would like to open this issue again. I have included the pppoe connection again and it is marked as down again but used and working:

```
ping -B -I pppoe-wan2 1.1.1.1
PING 1.1.1.1 (1.1.1.1) from 79.206.207.113 pppoe-wan2: 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=60 time=4.83 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=60 time=4.71 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=60 time=4.81 ms
64 bytes from 1.1.1.1: icmp_seq=4 ttl=60 time=4.88 ms
64 bytes from 1.1.1.1: icmp_seq=5 ttl=60 time=4.71 ms
64 bytes from 1.1.1.1: icmp_seq=6 ttl=60 time=4.87 ms
64 bytes from 1.1.1.1: icmp_seq=7 ttl=60 time=4.93 ms
^C
--- 1.1.1.1 ping statistics ---
7 packets transmitted, 7 received, 0% packet loss, time 6010ms
rtt min/avg/max/mdev = 4.712/4.820/4.930/0.076 ms

```

[image]

[image]

closed 2025-01-02 by github-actions[bot]