Ysurac/openmptcprouter #3554: 10Gb sfp+ ports not functionng on latest stable release

spritacco, 2024-09-26:
## Expected Behavior
link on 10Gb optical ports

Link lights show while booting 

## Current Behavior

In boot process sfp+ ports link lights work then go dark

I’ve tried a number of sfp+ transceivers and none work. Some report as unsupported but some don’t and none work. On the hardware I’m using, I lose 2 Ethernet ports. This is vanilla Intel hardware and these ports worked on the last openmptcp router with v5 and v6 kernels

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

I'm losing two Ethernet ports

## Specifications

  - OpenMPTCProuter version: 
  - .6 stable
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

Ysurac, 2024-09-26:
I would need to know what chipset is used to check if the driver is available or not.

spritacco, 2024-09-26:
Hi,

I'll grab the boot log... strange thing is this hardware worked fine in the last release.

spritacco, 2024-09-26:
See below... it seems the issue may be around line 872:

[   12.152079] pcieport 0000:00:17.0: can't derive routing for PCI INT A        
[   12.158663] ixgbe 0000:02:00.0: PCI INT A: not connected                     

This hardware was working on your release from last year... 

**********************************************************************
 
        Silicom
 
        80500-0214-G02-SL00x
 
**********************************************************************
Loading Usb Lens...
 Provider:  Insyde Software
 Version:   05.12.02.0040.0009.16
 Product:   BlinkBoot(R) Harrisonville USB Lens
 Licensee:  Silicom Ltd.
 Start:     03/20/2018
 End:       12/31/2099

Validating Signature... [Pass]

Installing Usb Lens...   

Loading Nvme Lens...
 Provider:  Insyde Software
 Version:   05.12.02.0040.0009.16                                               
 Product:   BlinkBoot(R) Harrisonville NVMe Lens                                
 Licensee:  Silicom Ltd.                                                        
 Start:     11/16/2020                                                          
 End:       12/31/2099                                                          
                                                                                
Validating Signature... [Pass]                                                  
                                                                                
Installing Nvme Lens...                                                         
                                                                                
Loading Sata Lens...                                                            
 Provider:  Insyde Software                                                     
 Version:   05.12.02.0040.0009.16                                               
 Product:   BlinkBoot(R) Harrisonville SATA Lens                                
 Licensee:  Silicom Ltd.                                                        
 Start:     03/20/2018                                                          
 End:       12/31/2099                                                          
                                                                                
Validating Signature... [Pass]                                                  
                                                                                
Installing Sata Lens...                                                         
                                                                                
Loading SdMmc Lens...                                                           
 Provider:  Insyde Software                                                     
 Version:   05.12.02.0040.0009.16                                               
 Product:   BlinkBoot(R) Harrisonville SDxx/eMMC Lens                           
 Licensee:  Silicom Ltd.                                                        
 Start:     03/20/2018                                                          
 End:       12/31/2099                                                          
                                                                                
Validating Signature... [Pass]                                                  
                                                                                
Installing SdMmc Lens...                                                        
                                                                                
Mapping table                                                                   
      FS2: Alias(s):F0:                                                         
          Fv(E35C4A77-3D00-4337-A625-B980A9E00F6C)                              
      FS4: Alias(s):F0:                                                         
          Fv(F9657F27-CAE7-4951-A406-D286157B63EF)                              
      FS3: Alias(s):F0:                                                         
          Fv(E7082AE5-3036-407A-8060-FDF027F74632)                              
      FS5: Alias(s):HD31b:;BLK1:                                                
[... 912 lines cut ...]
[   14.569401] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2.
[   14.585174] usbcore: registered new interface driver cdc_mbim                
[   14.591377] usbcore: registered new interface driver ipw                     
[   14.596863] usbserial: USB Serial support registered for IPWireless converter
[   14.611473] usbcore: registered new interface driver option                  
[   14.617236] usbserial: USB Serial support registered for GSM modem (1-port)  
[   14.625255] usbcore: registered new interface driver qcserial                
[   14.631165] usbserial: USB Serial support registered for Qualcomm USB modem  
[   14.640821] usbcore: registered new interface driver rt2800usb               
[   14.649696] usbcore: registered new interface driver rtl8192cu               
[   14.657589] usbcore: registered new interface driver ath9k_htc               
[   14.665047] kmodloader: done loading kernel modules from /etc/modules.d/*    
[   16.336855] pps pps4: new PPS source ptp4                                    
[   16.341100] ixgbe 0000:02:00.0: registered PHC device on eth2                
[   16.534016] 8021q: adding VLAN 0 to HW filter on device eth2                 
[... cut ...]
[   17.061210] pps pps6: new PPS source ptp6                                    
[   17.065441] ixgbe 0000:03:00.0: registered PHC device on eth0                
[   17.073134] 8021q: adding VLAN 0 to HW filter on device eth0                 
[   17.115464] pps pps7: new PPS source ptp7                                    
[   17.119720] ixgbe 0000:03:00.1: registered PHC device on eth1                
[   17.127169] 8021q: adding VLAN 0 to HW filter on device eth1                 
[   17.449384] ixgbe 0000:03:00.1 eth1: detected SFP+: 6

spritacco, 2024-10-08:
Hi,

I loaded 6.1 RC3 with the 6.6 kernel and can confirm the SFP+ ports work again on that version.

closed 2025-01-11 by github-actions[bot]