Ysurac/openmptcprouter #3555: Problem with internet server

infosb, 2024-09-26:
Hello.
I use OPENMPTCPROUTER since many months it work great thanks for the job.
I have a problem with my internet connection .... my server is localised in Sweden.
When i do a test with Fast.com
 Client   Paris, FR   xxx.xxx.xxx.xxx    Serveur(s) Stockholm, SE  |  Helsinki, FI
I use an adsl connexion a 4g and a 5g connexion.
I don't find a solution alone ... 
is there a thing to do in openmptcprouter ?
Could it due to my routers i have a zte 5g and a 4g huawei and i have bought them in Finlande.
Have you any idea ?
Thanks for your help

Ysurac, 2024-09-26:
I don't understand what you mean, here fast.com seems to put you in Paris and use a server in Stockholm for test. It's only a fast.com choice based on latency and status of their servers.

infosb, 2024-09-27:
Sorry but i don't understand why the server is localised in stockholm ... sometime i have some search results in swedish. and i Don't know where is my problem

Ysurac, 2024-09-27:
I don't know what is the problem in fact...

infosb, 2024-09-27:
Thanks but i have more than 100pc with about 700users ... is there no possibility to change it on openmptcprouter ?

Ysurac, 2024-09-27:
It's not related to OpenMPTCProuter, but to the IP of VPS or browser client.
For google, even if it's a french IP, if it was previously used as a proxy/VPN, then you can have this localization problems. With some luck, after setting the good country/language in google this will fix for others users too.

infosb, 2024-09-30:
thanks i will test it

closed 2025-01-03 by github-actions[bot]