Ysurac/openmptcprouter #3556: Fresh OMR does not pass any traffic out - VPS is connected

dersch81, 2024-09-26:
I have a fresh OMR here on a quite powerfull N100, 16 Gb DDR5, NVME and with 4x Intel i226 2,5gig ports. My goal is to have 7 uplinks bonded and i want to pass a lot of traffic. I'm coming from a LB pfSense setup with all it's negative issues and looking for a bonded, seen as one uplink, solution. Any used uplink is tested and used for long time via the pfSense LB way. 

Already played around with openMPTCP and used a Starlink with LTE Backup running on a Proxmox VM to test it. So it is not my first experience with it.

Now started to use the N100 since rc2 on kernel 6.6 is supporting it, i was waiting for this. Since the setup is fresh the only tweak i did was switching to V2Ray VLESS. But it is the same with Shadowsocks.

## Specifications

I have to run last rc with Kernel 6.6 because of the N100 support.

OpenMPTCProuter version: v0.61rc2
OpenMPTCProuter VPS version: 0.1031-test
OpenMPTCProuter VPS provider: HT-Hosting.de
OpenMPTCProuter platform: x86_64
Country: Germany

[image]

But it is not working at all and i can' t pass any traffic thru the OMR. It is not even able to ping anything from the router itself but the VPS connection is established. At the moment i'm using a few, uplinks and my two mains are still at the pfSense for productive usage. All the uplinks i use with the OMR are only available via tagged VLAN, and all NAT'ted because they are coming from my roof via PtP, no way to connect them directly. 

All the gateways are working in that way: 

```
root@OpenMPTCProuter:~# ping -B -I eth1.333 1.1.1.1
PING 1.1.1.1 (1.1.1.1) from 100.117.46.120 eth1.333: 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=58 time=23.8 ms
64 bytes from 1.1.1.1: icmp_seq=2 ttl=58 time=21.0 ms
64 bytes from 1.1.1.1: icmp_seq=3 ttl=58 time=22.3 ms
```

1x Starlink via VLAN 333
1x LTE via VLAN 113
3x VDSL Lines (PtP connected) via VLAN 111, 222, 176

[image]

```
root@OpenMPTCProuter:~# ip r
default via 10.255.252.1 dev tun0
default metric 1
        nexthop via 100.64.0.1 dev eth1.333 weight 100
        nexthop via 192.168.176.1 dev eth1.176 weight 1
        nexthop via 10.10.11.1 dev eth1.111 weight 1
        nexthop via 192.168.33.1 dev eth1.222 weight 1
default via 100.64.0.1 dev eth1.333 metric 7
default via 10.240.96.73 dev eth1.113 metric 8
default via 192.168.33.1 dev eth1.222 metric 9
default via 192.168.176.1 dev eth1.176 metric 10
default via 10.10.11.1 dev eth1.111 metric 11
default via 10.255.252.1 dev tun0 metric 1500
10.10.11.0/24 dev eth1.111 scope link metric 11
10.240.96.64/28 dev eth1.113 scope link metric 8
10.255.252.0/24 dev tun0 proto kernel scope link src 10.255.252.2
10.255.252.0/24 dev tun0 scope link metric 1500
10.255.252.1 via 10.255.252.1 dev tun0
100.64.0.0/10 dev eth1.333 scope link metric 7
127.0.0.0/8 dev lo proto static scope link metric 6
185.xxx.xx.xx metric 1
        nexthop via 100.64.0.1 dev eth1.333 weight 100
        nexthop via 192.168.176.1 dev eth1.176 weight 1
        nexthop via 10.10.11.1 dev eth1.111 weight 1
        nexthop via 192.168.33.1 dev eth1.222 weight 1
185.xxx.xx.xx via 10.240.96.73 dev eth1.113 metric 999
192.168.33.0/24 dev eth1.222 scope link metric 9
192.168.176.0/24 dev eth1.176 scope link metric 10
```

```
root@OpenMPTCProuter:~# omr-test-speed
Select best test server...
[... 108 lines cut ...]
Sep 26 22:21:24 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: sending renew to server 10.240.96.73
Sep 26 22:21:24 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: lease of 10.240.96.72 obtained from 10.240.96.73, lease time 60
Sep 26 22:21:54 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: sending renew to server 10.240.96.73
Sep 26 22:21:54 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: lease of 10.240.96.72 obtained from 10.240.96.73, lease time 60
Sep 26 22:22:24 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: sending renew to server 10.240.96.73
Sep 26 22:22:24 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: lease of 10.240.96.72 obtained from 10.240.96.73, lease time 60
Sep 26 22:22:54 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: sending renew to server 10.240.96.73
Sep 26 22:22:54 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: lease of 10.240.96.72 obtained from 10.240.96.73, lease time 60
Sep 26 22:23:18 OpenMPTCProuter daemon.notice netifd: wan1 (10373): udhcpc: sending renew to server 100.64.0.1
Sep 26 22:23:18 OpenMPTCProuter daemon.notice netifd: wan1 (10373): udhcpc: lease of 100.117.46.120 obtained from 100.64.0.1, lease time 300
Sep 26 22:23:24 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: sending renew to server 10.240.96.73
Sep 26 22:23:24 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: lease of 10.240.96.72 obtained from 10.240.96.73, lease time 60
Sep 26 22:23:54 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: sending renew to server 10.240.96.73
Sep 26 22:23:54 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: lease of 10.240.96.72 obtained from 10.240.96.73, lease time 60
Sep 26 22:24:24 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: sending renew to server 10.240.96.73
Sep 26 22:24:24 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: lease of 10.240.96.72 obtained from 10.240.96.73, lease time 60
Sep 26 22:24:54 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: sending renew to server 10.240.96.73
Sep 26 22:24:54 OpenMPTCProuter daemon.notice netifd: wan2 (10358): udhcpc: lease of 10.240.96.72 obtained from 10.240.96.73, lease time 60
```

Testing from my pfSense with OMR as Gateway. (I used this before in the same way with my Proxmox test VM.)

[image]

What is my failure? I'm really out of idea...

Ysurac, 2024-09-27:
wan1 & wan2 are giving DHCP address for too short time. Fix them or use static IP.
What is the result of ```ip mptcp endpoint``` ?
Did you try to set connections one after one ?

dersch81, 2024-09-27:
Many thanks for your reply. 

WAN1 is Starlink in Bridge Mode
WAN2 is LTE Mikrotik Gateway in Passthrough Mode

So I can't set both to static. But if the short leasetime is no general problem I will accept it. But I did not notice this with pfSense. Anyway not the big problem I guess.

This time I set up every WAN in one shot. So you suggest to add them one by one?

I will check the command later when I'm back on site. 

Another question: I have own IPv6 /48 PI assigned subnet. At the moment I use this with my pfsense via wireguard to my BGP VPS at my sponsor datacenter. 

When I complete my OMR Setup as the main Gateway I would like to have OMR as the IPv6 Gateway, too. So i create the wg interface to get my /48 route there. And then passing all the needed IPv6 prefix subents to all the other routers behind the OMR, i also plan to create another LAN interface. Any hints from your side to make this design in best practise? I also would like to use the wg interface with MPTCP to increase my IPv6 bandwith. Or should I create the wg connection on the OMR VPS? How would you do it?

dersch81, 2024-09-27:
```

root@OpenMPTCProuter:~# ip mptcp endpoint
192.168.176.103 id 1 subflow fullmesh dev eth1.176
100.117.46.120 id 2 subflow fullmesh dev eth1.333
10.10.11.30 id 3 subflow fullmesh dev eth1.111
192.168.33.27 id 4 subflow fullmesh dev eth1.222
10.240.96.72 id 5 backup fullmesh dev eth1.113
```

dersch81, 2024-09-27:
Now i have deleted WAN 3, 4 & 5. And just used WAN1 as Master and WAN2 as Backup. And traffic started to flow for a short time right after saving the setting. But then stopped a few seconds afterwards.

dersch81, 2024-09-27:
Its I226-V

For those Uplinks i'm testing i can't because they are all coming from my roof via a trunk. And during my lab testings i had this configured in the same way. Even on the N100 i had this already running for a short time but then played to much with it, got other issues and decided to start with a fresh installed image.

eth0 is my LAN trunk, eth1 will be the tagged vlan trunk interface as shown here, and later when it is running eth2 local availabe cable and eth3 local available VDSL PPPOE interface.

With WAN1 and WAN2 configured as suggested i have the same issue.

Can you be more precise which log is helpful for you?

```
ip mptcp endpoint
192.168.33.27 id 1 subflow fullmesh dev eth1.222
100.117.46.120 id 2 subflow fullmesh dev eth1.333
```

Now i have connected my local available static IP Cable connection to eth2. But it is the same issue:

```
root@OpenMPTCProuter:~# ip mptcp endpoint
192.168.33.27 id 1 subflow fullmesh dev eth1.222
100.117.46.120 id 2 subflow fullmesh dev eth1.333
130.xxx.xxx.xxx id 3 subflow fullmesh dev eth2
root@OpenMPTCProuter:~# ping google.de
ping: google.de: Try again
root@OpenMPTCProuter:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
^C
--- 8.8.8.8 ping statistics ---
6 packets transmitted, 0 received, 100% packet loss, time 5163ms

```
[image]

Ysurac, 2024-09-27:
I226-V have many hardware issues that can give some strange speed/DHCP/Stability problems.
To test, try to not use VLAN on OpenMPTCProuter.
Check also Status->System log, OMR-Tracker should test the VPN via ping.
Check VPN endpoint via ```ping 10.255.252.1```

dersch81, 2024-09-27:
Ok, i wasn't aware of issues with i226-V. Which 2.5 gig chip do you suggest? 

But i don't think the general issue that no traffic is passing the OMR is due to the i226-v

dersch81, 2024-09-28:
Ok, i think i got it and be close to my first goal. I did not even add every Uplink so far :D And i'm still connected via 1 Gbe. When i switch to 2.5 Gbe it will increase i guess.

[image]
[image]

closed 2024-09-28 by dersch81