Ysurac/openmptcprouter #3558: Setting a WAN to "MPTCP over VPN" brings Glorytun into an reconnect / recreation loop

darkman1983, 2024-09-27:
## Expected Behavior
If i set a WAN to "MPTCP over VPN" it should simply work and Glorytun shouldn't reconnect all the time

## Current Behavior
The MPTCPRouter tells me, that my 5G WAN has multipath blocked, but that isn't the case, what doesn't work over that WAN is: Glorytun. If i disable the other WAN, then no portforwarding is working.
So i set the 5G WAN to "MPTCP over VPN" (Wireguard).

But this brings Glorytun into an reconnect / recreation loop.

`Sep 27 22:45:20 OpenMPTCProuter daemon.info glorytun[6662]: INITIALIZED tun0
Sep 27 22:45:20 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Sep 27 22:45:20 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Sep 27 22:45:20 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Sep 27 22:45:20 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
Sep 27 22:45:20 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='tun0'
Sep 27 22:45:20 OpenMPTCProuter daemon.info ModemManager[6692]: hotplug: add network interface tun0: event processed
Sep 27 22:45:20 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sep 27 22:45:21 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Sep 27 22:45:21 OpenMPTCProuter daemon.notice ttyd[22711]: [2024/09/28 00:45:21:1842] N: rops_handle_POLLIN_netlink: DELADDR
Sep 27 22:45:21 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: connected
Sep 27 22:45:21 OpenMPTCProuter daemon.err glorytun[6662]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep 27 22:45:31 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: key exchange failed
Sep 27 22:45:31 OpenMPTCProuter daemon.info glorytun[6662]: STOPPED tun0
Sep 27 22:45:32 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: connected
Sep 27 22:45:32 OpenMPTCProuter daemon.err glorytun[6662]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep 27 22:45:42 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: key exchange failed
Sep 27 22:45:42 OpenMPTCProuter daemon.info glorytun[6662]: STOPPED tun0
Sep 27 22:45:43 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: connected
Sep 27 22:45:43 OpenMPTCProuter daemon.err glorytun[6662]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep 27 22:45:53 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: key exchange failed
Sep 27 22:45:53 OpenMPTCProuter daemon.info glorytun[6662]: STOPPED tun0
Sep 27 22:45:54 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: connected
Sep 27 22:45:54 OpenMPTCProuter daemon.err glorytun[6662]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep 27 22:46:01 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Sep 27 22:46:04 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: key exchange failed
Sep 27 22:46:04 OpenMPTCProuter daemon.info glorytun[6662]: STOPPED tun0
Sep 27 22:46:05 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: connected
Sep 27 22:46:05 OpenMPTCProuter daemon.err glorytun[6662]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep 27 22:46:15 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: key exchange failed
Sep 27 22:46:15 OpenMPTCProuter daemon.info glorytun[6662]: STOPPED tun0
Sep 27 22:46:16 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: connected
Sep 27 22:46:16 OpenMPTCProuter daemon.err glorytun[6662]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep 27 22:46:26 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: key exchange failed
Sep 27 22:46:26 OpenMPTCProuter daemon.info glorytun[6662]: STOPPED tun0
Sep 27 22:46:27 OpenMPTCProuter daemon.err glorytun[6662]: 178.254.20.73.65001: connected
Sep 27 22:46:27 OpenMPTCProuter daemon.err glorytun[6662]: couldn't set socket option `TCP_USER_TIMEOUT'
Sep 27 22:46:28 OpenMPTCProuter daemon.info glorytun[6662]: STARTED tun0
Sep 27 22:46:28 OpenMPTCProuter user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.255.1 dev tun0 (was 192.168.30.1)`

The socket option error only occurs on "MPTCP over VPN".

## Possible Solution
Dunno, you tell me :-D

## Steps to Reproduce the Problem

  1. Install a clean install of the latest snapshot build
  2. Configure your instance
  3. Set one WAN to "MPTCP over VPN" (Wireguard)
  4. Look at the Logs, maybe it doesn't start just now, wait a few minutes for the loop to begin.

## Context (Environment)
Portforwarding isn't working as expected, because of that looping it breaks until glorytun has connection again.
We're in a VM context here.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61rc2-6.6 (r0+27346-c7ba5574f5)
  - OpenMPTCProuter VPS version: Version 0.1031-test 6.10.2-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: 1blu.de
  - OpenMPTCProuter platform: x86_64
  - Country: Germany

Ysurac, 2024-09-30:
Use another VPN, it's why glorytun isn't used as default anymore on 6.x

darkman1983, 2024-09-30:
All the other VPN's providing horrible pingtimes! And i need good Latency.

darkman1983, 2024-10-07:
I tried to setup OpenVPN, but it goes into that loop too.
As soon as i enable 'MPTCP over VPN' for WAN2, it goes into that loop.

closed 2025-01-11 by github-actions[bot]