Ysurac/openmptcprouter #3561: Problems with installing

jansiebks, 2024-09-29:
Hello,

First of all, I apologize; I'm not a professional—this is just a hobby of mine. A few months ago, I installed OpenMPTCProuter on my Raspberry Pi. It works, but the Pi has become too slow for my needs.

Yesterday, I bought a "ROUAFWIT Micro Firewall N100 12TH Gen Fanless Mini PC" ([https://www.amazon.de/dp/B0CTHXRPNJ)](https://seoschmiede.at/aitools/chatgpt-tool/%3Ca%20href=)" target="_blank">https://www.amazon.de/dp/B0CTHXRPNJ)">https://www.amazon.de/dp/B0CTHXRPNJ) from Amazon to replace the Pi.

First Question: Is this a supported device?

Second Problem: I'm not sure which software is the right one to use. I tried every "x86-64 (64-bit)" version with a .img extension. I flashed my USB flash drive and selected this USB drive in the BIOS. However, after 30 seconds, the Mini PC reboots. What might be my mistake?

Thanks!

Ysurac, 2024-09-30:
On N100 is often UEFI image v0.61rc2 or snapshot (not working with stable current image).

jansiebks, 2024-09-30:
Thanks for your answer. Now I get this picture. 

When I plug in the LAN Cable to my c
[image]
omputer, the computer doesn't get a ip address - and I can't go to 192.168.100.1...
What ist the next step?

Sorry for my (maybe) stupid questions: :)
[image]

Ysurac, 2024-09-30:
check with ```ip a``` if the IP is set somehwere.

dersch81, 2024-09-30:
Maybe your computer network interface is not set to static? Or you are plugged in the wrong interface of the N100 (should be ETH0). Try to set your computers IP to 192.168.100.100/24 and ping the OMR.

jansiebks, 2024-09-30:
Thanks! I works now! I use the wrong LAN Port, with ETH0 works!

dersch81, 2024-09-30:
Glad I've could help you. Then you can close the issue, I guess.

closed 2025-01-05 by github-actions[bot]