Ysurac/openmptcprouter #3563: WIFI disconnecting in AC Mode at high throughput

Sorab8888, 2024-09-30:
## Expected Behavior

Hi any help is most welcome. Thanks
## Current Behavior

When using WIFI AC mode, any channel, the WIFI spot just disappears each time I do a speed test and the throughput gets higher than 60-80 Mbs. Then I have to restart the router to be able to see and connect to the WIFI again
## Specifications

  - OpenMPTCProuter version: V 0.60 6.1
  - OpenMPTCProuter VPS version: V 6.1
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: Nano PI Rockchip

Ysurac, 2024-09-30:
You can try to enable SQM on the wifi.
What is the full platform name ?

Ysurac, 2024-09-30:
You need to set a download/upload speed in this SQM menu. To check if it crash when higher than a speed or if there is another reason.
Do have also some logs in Status->System log or Status->Kernel log ?

closed 2025-01-03 by github-actions[bot]