Ysurac/openmptcprouter #3564: OpenMPTCP Real Time VS Speedtest.net results INCONSISTENT RESUTLTS

jcardenes-via, 2024-10-01:
## Expected Behavior

Currently I have 3 WAN connections:

 -SIM CARD 1 -> 40 MBps Download
 -ETHERNET CONNECTION -> 460 MBps Download

I would expect the following results:

 -Real time bandwidth on OpenMPTCP I see speed total while triggering a speedtest.net of:
500 MBps average

 -On the speedtest.net page I would expect to see from the VPS:
500 MBps average

## Current Behavior

The current behaviour is, but I dont understand what is the case and how could I match the expected, why so much download speed lost:

 -OpenMPTCP real time bandwidth with speedtest.net 500 MBps average
 -Speedtest.net through VPS 360MBps average

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: debian12 kernel 6.1
  - OpenMPTCProuter VPS provider: AWS
  - OpenMPTCProuter platform: Promox VM

dersch81, 2024-10-01:
With all of my testings i noticed that Speedtest.net is not able to utilize the MPTCP. try dslreports.com nperf.com or wifiman.com if you want to test in a much more relaiable way use iperf3. You can use an own VPS or i use that one: https://speedtest.wtnet.de/

jcardenes-via, 2024-10-02:
Thank you @dersch81 wifiman is providing more consistent results, also I noticed that I was using an AWS bandwidth burstable instance, so AWS was throttling the bandwidth.

closed 2024-10-02 by jcardenes-via