Ysurac/openmptcprouter #3565: x64 router frozen on USB init

yec7uF3ucu2uteku, 2024-10-01:
openMPTCProuter should boot

## Current Behavior
Boot sequence frozen on USB init (see picture) :
hub 1-3.4:1.0 : 4 ports detected
[image]

## Specifications

  - OpenMPTCProuter version: [openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined-efi.img.gz](https://releases.openmptcprouter.com/v0.60-6.1/x86_64/targets/x86/64/openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-x86-64-generic-ext4-combined-efi.img.gz)
  - OpenMPTCProuter VPS version: useless because router is not booting
  - OpenMPTCProuter VPS provider: useless because router is not booting
  - OpenMPTCProuter platform: useless because router is not booting

I successfully booted Debian and Ubuntu on this computer without any issue.
I also tried non EFI and 32 bits and both did not boot at all.
Any help welcome !

Ysurac, 2024-10-03:
Try latest snapshot image: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
You tried to press "Enter" key ? at it doesn't seems to be crashed.

yec7uF3ucu2uteku, 2024-10-27:
Thanks for you answer Yannick.

Nothing happens when I press enter, but when I plug a USB stick I see it mounting (see screenshot)
[image]

I tried the latest snapshot and it stops even before (see screenshot)
I experience this error only on this Dell R360 server, no error on other hardware.
[image]

Ysurac, 2024-10-27:
On the first screen, for me there is no crash, it should answer when you press "enter" with the login message.
For the second screen, this seems to fast to be good. But no real errors. You can try an official OpenWRT image to check if you have same issue or not.

closed 2025-01-30 by github-actions[bot]