Ysurac/openmptcprouter #3566: Raspberry Pi 5 2gb boot-loop

ffrrzz, 2024-10-01:
## Expected Behavior
The images available for the Raspberry Pi 5 should support the new 2gb RAM variant.

## Current Behavior
After imaging an sd card with this image the raspberry pi 5 boot loops continuously: https://releases.openmptcprouter.com/v0.60-6.1/rpi5/targets/bcm27xx/bcm2712/openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-bcm27xx-bcm2712-rpi-5-squashfs-factory.img.gz

## Possible Solution
Other OpenWRT-based firmwares seem to have the same issue. An update to newer releases of OpenWrt seems to fix the problem.

Ysurac, 2024-10-01:
Did you try snapshot image ? https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

ffrrzz, 2024-10-01:
> Did you try snapshot image ? https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

This one worked. Thank you!

https://snapshots.openmptcprouter.com/6.6/rpi5/targets/bcm27xx/bcm2712/

closed 2024-10-01 by ffrrzz