Ysurac/openmptcprouter #3569: Help needed with OMR installation on VPS

w453y, 2024-10-03:
For the past few weeks, I've been facing an issue with VPS. I have terminated a lot of VPS and tried again, but still the following issue persists.
 
## Expected Behavior
Running VPS script and rebooting VPS once, then adding server keys on local OMR

## Current Behavior
VPS script executes completely and VPS is ready to use, but however when I add VPS details on OMR, it shows me the below error.
[image]

I tried a lot to fix this but still couldn't do so.

## Specifications

  - OpenMPTCProuter version:  v0.61-snapshot-5.4 r0+16862-170d9e447d 
  - OpenMPTCProuter VPS version: 0.1030 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: I have tried with both AWS and GCP, even added firewall rules too.
  - OpenMPTCProuter platform: RPI4 B

andrewmackrodt, 2024-10-03:
Looks like the date on your router is way out of sync (by more than 14 months!); maybe it is some kind of certificate trust issue because of this clock skew.

w453y, 2024-10-05:
still can't figure out how to fix that, flashed rpi again and installed OMR on another VPS and still face the same issue. One error I have seen while executing the script shows something like the following "shadowsockes.unit file not found".

Ysurac, 2024-10-06:
The error is not a problem when the script doesn't crash. It's when updating from old release.
Check time and set it on VPS with date command. If the time is not correct, do ```apt-get install systemd-timesyncd``` to keep time updated. On the RPI, go in System->System menu, and sync with browser.

closed 2024-10-10 by w453y