Ysurac/openmptcprouter #3572: Hello, regarding the issues encountered during VPS installation

a834003831, 2024-10-06:
**I am currently in China and have created VPS for Hong Kong and Japan on Alibaba Cloud for installation, but I encountered an error during installation**
[image]

a834003831, 2024-10-06:
Solved, just upgrade the system. If you have any other questions, please consult the author. For X86, can I use an n100 industrial computer? Can you recommend a configuration, preferably a specific model

Ysurac, 2024-10-09:
For n100, I have nothing to recommend for now.
For latency, can you do a ```ping -B -I <5gcpe> <vpsip>``` (replace <5gcpe> by real physical interface and <vpsip> by VPS IP) ?
For screen display, for now support is not planned.

a834003831, 2024-10-09:
> 对于n100，我暂时没什么好推荐的。对于延迟，你能做一个`ping -B -I <5gcpe> <vpsip>`(用真实物理接口和VPS IP替换< 5gcpe >)？对于屏幕显示，目前还没有计划支持。

Hello, are you using SHH for ping? My 5gcpe IP is 192.168.2.1, vpsip：8.138.114.237， What is the complete command? Sorry for asking such a foolish question, as I am a beginner

Ysurac, 2024-10-09:
Via SSH on the router, ```ping -c 3 -B -I 192.168.2.1 8.138.114.237```

a834003831, 2024-10-09:
> ping -c 3 -B -I 192.168.2.1 8.138.114.237

An error occurred. I checked the IP addresses of VPS and 5GCPE, and they are both correct
[image]
[image]

a834003831, 2024-10-10:
> Via SSH on the router, `ping -c 3 -B -I 192.168.2.1 8.138.114.237`

Hello, after testing, I encountered a problem. Can't the optimal route be disconnected? I have connected three WAN in total, but in actual use, only two are running data, and one is the main force. Is this normal

Ysurac, 2024-10-11:
Try ```ping -c 3 -B -I usb0 8.138.114.237```

a834003831, 2024-10-13:
Hello, I have provided the latest feedback on how to make this delay display on the interface
[image]
[image]

PING 8.138.114.160 (8.138.114.160) from 192.168.1.4 eth0: 56(84) bytes of data.
64 bytes from 8.138.114.160: icmp_seq=1 ttl=50 time=16.9 ms
64 bytes from 8.138.114.160: icmp_seq=2 ttl=50 time=18.1 ms
64 bytes from 8.138.114.160: icmp_seq=3 ttl=50 time=22.1 ms

--- 8.138.114.160 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2001ms
rtt min/avg/max/mdev = 16.936/19.035/22.057/2.189 ms

a834003831, 2024-10-13:
What I mean is: why doesn't every WAN port display the delay time

a834003831, 2024-10-14:
The problem described above has been resolved, and a new difficulty has been encountered. The error message means that the file system is in read-only state/please refresh the machine
[image]

Ysurac, 2024-10-14:
You can, via SSH on the router, do ```e2fsck /dev/mmcblk0p2```
It's due to power issue (if you remove power supply to stop or if there is not enough power) or a bad SD card

a834003831, 2024-10-14:
> 通过路由器上的SSH，您可以`e2fsck /dev/mmcblk0p2`这是由于电源问题(如果您删除电源停止或如果没有足够的电力)或一个坏的SD卡

e2fsck 1.47.0 (5-Feb-2023)
e2fsck: No such file or directory while trying to open /dev/mmcblk0p2
Possibly non-existent device?
[image]

I restarted it a few times and it's like this, but flashing it again didn't work

Ysurac, 2024-10-14:
On your platform it can be another device in /dev/, can you do a ```ls /dev/*``` ?
Flashing again should work in all cases.

a834003831, 2024-10-14:
> On your platform it can be another device in /dev/, can you do a `ls /dev/*` ? Flashing again should work in all cases.

Do I need to unplug and restart the WAN every time? The result of my testing here is that it cannot start normally with WAN plugged in

a834003831, 2024-10-14:
At present, during testing, the device cannot start normally after being connected to a USB network card. However, it can start normally after unplugging the USB network card. How can I achieve normal startup without unplugging it?

a834003831, 2024-10-15:
> On your platform it can be another device in /dev/, can you do a `ls /dev/*` ? Flashing again should work in all cases.

Hello, I have found out why the system crashes after restarting and prompts:：system is in read-only state/please refresh the machine，Because my device is an R4S and I am plugged in a USB card, there may be issues with the system due to boot files or boot items. Please tell me how to solve this problem. Thank you

a834003831, 2024-10-16:
Hello, I found a solution online: I need to modify the boot/grub/grub.cfg file, but I encountered a problem: I cannot find the file. Can you tell me the solution? The device is R4S, and the USB card is an 8153 chip

Ysurac, 2024-10-16:
R4S is not using Grub as bootloader but U-Boot.

a834003831, 2024-10-17:
> R4S is not using Grub as bootloader but U-Boot.

Hello, thank you for your patient answer. Therefore, I am planning to replace the j1900 or j4125 CPU for use. I would like to confirm a question: What types of network cards (1G and 2.5G) are supported by the X86-64 system? I will choose the appropriate device based on the network card driver that you have already adapted to

Ysurac, 2024-10-17:
I22x-V are know to have some hardware issues. Else there is many supported network cards.

a834003831, 2024-10-18:
> I22x-V are know to have some hardware issues. Else there is many supported network cards.

Hello, sorry to bother you multiple times. I would like to ask if BPI-R4 is currently compatible. I noticed that this device can connect to multiple modules. Another question is: What are the currently supported 5G modules?Thank you again for your hard work in development

closed 2025-01-21 by github-actions[bot]