Ysurac/openmptcprouter #3576: huge latency variation

talitha77, 2024-10-07:
## Expected Behavior
I would hope to see lowest latency of the three aggregated connections prevail
status page shows all three with around 50 to 70ms, quite acceptable.

## Current Behavior

seems to vary wildly between 50 to 700 and more

yes but nothing leaps out at me

## Specifications

  - OpenMPTCProuter version: 0.60-6.1
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-25-amd65
  - OpenMPTCProuter VPS provider: vpsserver.com
  - OpenMPTCProuter platform: rpi5

64 bytes from 8.8.8.8: icmp_seq=53 ttl=56 time=64.754 ms
64 bytes from 8.8.8.8: icmp_seq=54 ttl=56 time=63.480 ms
64 bytes from 8.8.8.8: icmp_seq=55 ttl=56 time=74.694 ms
64 bytes from 8.8.8.8: icmp_seq=56 ttl=56 time=330.823 ms
64 bytes from 8.8.8.8: icmp_seq=57 ttl=56 time=202.425 ms
64 bytes from 8.8.8.8: icmp_seq=58 ttl=56 time=127.200 ms
64 bytes from 8.8.8.8: icmp_seq=59 ttl=56 time=415.472 ms
64 bytes from 8.8.8.8: icmp_seq=60 ttl=56 time=228.652 ms
64 bytes from 8.8.8.8: icmp_seq=61 ttl=56 time=136.260 ms
64 bytes from 8.8.8.8: icmp_seq=62 ttl=56 time=311.486 ms
64 bytes from 8.8.8.8: icmp_seq=63 ttl=56 time=89.508 ms
64 bytes from 8.8.8.8: icmp_seq=64 ttl=56 time=109.438 ms
64 bytes from 8.8.8.8: icmp_seq=65 ttl=56 time=133.091 ms
64 bytes from 8.8.8.8: icmp_seq=66 ttl=56 time=169.927 ms
64 bytes from 8.8.8.8: icmp_seq=67 ttl=56 time=53.753 ms
64 bytes from 8.8.8.8: icmp_seq=68 ttl=56 time=83.665 ms
[image]

Ysurac, 2024-10-07:
What is the VPN used ? By default it's OpenVPN, ping can be quite bad if some packets are lost.
TCP is using a proxy and anything else the VPN. So TCP can have a better latency than ping.

talitha77, 2024-10-07:
yes its openvpn. i haven't changed any settings at all..  aside from the crazy latency the aggregation is working fine

talitha77, 2024-10-07:
[image]

talitha77, 2024-10-24:
unfortunately not..

closed 2025-01-27 by github-actions[bot]