Ysurac/openmptcprouter #3578: v0.63RC3 Kernel 6.11 VPS install errors/warnings

datapharmer, 2024-10-08:
## Expected Behavior

Should install without error...

## Current Behavior

Threw a number of errors and warnings that may impact functionality (impact unknown at this point, but may be tied to the tls handshake issue I reported with 0.61RC2). I understand some of these are likely safe to ignore such as pip3 but included for full context. Others seem they could be problematic such as `Failed to get unit file state for shadowsocks-libev.service: No such file or directory`

## Possible Solution

I'm not sure how any of these should be addressed, but I'm happy to test out anything that seems it should be fixed.

## Steps to Reproduce the Problem

  1. Ran wget -O - https://www.openmptcprouter.com/server-rc/debian-x86_64.sh | KERNEL="6.11" sh
  2. observe output (relevant logs below)

## Context (Environment)

VPS on BuyVM/Frantech running Debian 12

## Specifications

  - OpenMPTCProuter version:  N/A (but 0.61RC2 currently)
  - OpenMPTCProuter VPS version: 0.1031-test 
  - OpenMPTCProuter VPS provider: BuyVM
  - OpenMPTCProuter platform: N/A (but RPI4)
  - Country: USA

WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
Reading package lists... Done

....

-- pip3 install needed python modules
If you see any error here, I really don't care: it's about a module not used for home users
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
--2024-10-08 09:47:15--  https://www.openmptcprouter.com/server-test/omr-admin.service.in

....

Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|2400:52e0:1a02::974:1|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 386 [application/octet-stream]
Saving to: '/lib/systemd/system/shadowsocks-libev-manager@.service'

/lib/systemd/system 100%[===================>]     386  --.-KB/s    in 0s

2024-10-08 09:47:17 (14.0 MB/s) - '/lib/systemd/system/shadowsocks-libev-manager@.service' saved [386/386]

Failed to get unit file state for shadowsocks-libev.service: No such file or directory
--2024-10-08 09:47:17--  https://www.openmptcprouter.com/server-test/omr-update.service.in

....

(Reading database ... 164388 files and directories currently installed.)
Preparing to unpack /tmp/v2ray-5.7.0-amd64.deb ...
Unpacking v2ray (5.7.0) over (5.7.0) ...
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: trying to overwrite '/usr/bin/geoip.dat', which is also in package xray 1.8.24
dpkg: warning: overriding problem because --force enabled:
dpkg: warning: trying to overwrite '/usr/bin/geosite.dat', which is also in package xray 1.8.24
Setting up v2ray (5.7.0) ...

....

[... 93 lines cut ...]
src/main.c: In function 'gt_setup_crypto':
src/main.c:1228:40: warning: passing argument 1 of 'crypto_aead_aes256gcm_beforenm' from incompatible pointer type [-Wincompatible-pointer-types]
 1228 |         crypto_aead_aes256gcm_beforenm(&ctx->read.key, key_r);
      |                                        ^~~~~~~~~~~~~~
      |                                        |
      |                                        uint8_t (*)[512] {aka unsigned char (*)[512]}
In file included from /usr/include/sodium.h:8,
                 from src/main.c:38:
/usr/include/sodium/crypto_aead_aes256gcm.h:118:65: note: expected 'crypto_aead_aes256gcm_state *' {aka 'struct crypto_aead_aes256gcm_state_ *'} but argument is of type 'uint8_t (*)[512]' {aka 'unsigned char (*)[512]'}
  118 | int crypto_aead_aes256gcm_beforenm(crypto_aead_aes256gcm_state *ctx_,
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
src/main.c:1229:40: warning: passing argument 1 of 'crypto_aead_aes256gcm_beforenm' from incompatible pointer type [-Wincompatible-pointer-types]
 1229 |         crypto_aead_aes256gcm_beforenm(&ctx->write.key, key_w);
      |                                        ^~~~~~~~~~~~~~~
      |                                        |
      |                                        uint8_t (*)[512] {aka unsigned char (*)[512]}
/usr/include/sodium/crypto_aead_aes256gcm.h:118:65: note: expected 'crypto_aead_aes256gcm_state *' {aka 'struct crypto_aead_aes256gcm_state_ *'} but argument is of type 'uint8_t (*)[512]' {aka 'unsigned char (*)[512]'}
  118 | int crypto_aead_aes256gcm_beforenm(crypto_aead_aes256gcm_state *ctx_,
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~
  CC       src/glorytun-option.o
  CC       src/glorytun-tun.o
  CC       src/glorytun-db.o
  CC       src/glorytun-state.o
  CCLD     glorytun
--2024-10-08 09:48:41--  https://www.openmptcprouter.com/server-test/glorytun-tcp-run

Ysurac, 2024-10-08:
Nothing to really care about. Real errors block the script execution.
I can hide these warning/errors but this can hide real errors too and will not help debugging.

datapharmer, 2024-10-08:
no need to hide these if they aren't a concern. Thanks for reviewing to confirm.

closed 2024-10-08 by datapharmer