Ysurac/openmptcprouter #3580: Plex Remote Access Not Utilizing Aggregated Connection

TipsTricksMore, 2024-10-08:
Hi,

I’ve been using OpenMPTCPRouter for over a year in my office to bond two 1 Gbit/s coaxial connections into a combined 2 Gbit/s downlink. My primary goal was to achieve a more reliable uplink of 2x 50 Mbit/s, which has successfully provided me with a steady 100 Mbit/s upload, and it’s been working great for my needs.

I supported this project with a tip a while back because it has been incredibly helpful. Thank you for all your hard work!

Recently, I’ve started hosting a Plex server on my NAS. I have some large video files with bitrates of around 80 Mbit/s that I’d love to stream at home, where I have a 1 Gbit/s connection. In theory, this should work perfectly, but I’ve noticed an issue: Plex seems to only utilize one of my uplinks, maxing out at 50 Mbit/s, which leads to buffering.

This behavior is puzzling because when I run speed tests or upload files to other platforms from my PC on the same office network, OpenMPTCPRouter splits the bandwidth across both connections flawlessly, saturating the full 100 Mbit/s.

I suspect the issue might be related to my port forwarding configuration for Plex on port 32400. Here’s my current setup:

[image]
[image]

And here’s the Plex configuration, which appears to be fine:

[image]

Could there be something else I’m overlooking that might be causing this issue?

Thanks in advance for your help!

TipsTricksMore, 2024-10-14:
@Ysurac do you got any idea regarding this?

Ysurac, 2024-10-17:
I never tested plex, so I'm not sure how it work.
You can try to set another VPN in System->OpenMPTCProuter, "Wizard" tab, "Advanced settings" checkbox.

TipsTricksMore, 2024-10-31:
Thank you for your help! I tried adjusting various VPN settings but, unfortunately, none seemed to resolve the issue at first.

After updating both my VPS and local router to the latest firmware, however, everything is now working smoothly! I also took the opportunity to reinstall my Plex Docker during this process.

I’m not entirely sure what caused the problem initially, but thankfully, it’s all sorted now.

Thanks again for your assistance @Ysurac !

closed 2024-10-31 by TipsTricksMore