Ysurac/openmptcprouter #3581: Upgraded to v0.60rc3 and network devices are not present

Nicras, 2024-10-09:
## Expected Behavior
I upgraded my VPS first and then the router with rc3 and the expected behavior should be that everything works as before.

## Current Behavior
The interfaces on my NanoPi R5S show up with "Network device not present".

[image]

## Possible Solution

## Steps to Reproduce the Problem

  1. Upgrade via Webconsole
  2.
  3.

## Context (Environment)
I have three physical ethernet ports, that I want to make use of. So I changed the default assignment of the WAN interfaces

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61rc3-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: 0.1031 test
  - OpenMPTCProuter VPS provider: Strato
  - OpenMPTCProuter platform: NanoPI R5S
  - Country: Germany

Ysurac, 2024-10-09:
What was the initial version ?
What is the result of ```uci show network``` and ```ip a``` ?

Nicras, 2024-10-09:
I installed and configured from the Downloads page

openmptcprouter-v0.60-6.1-r0+24843-acf40c022e-rockchip-armv8-friendlyarm_nanopi-r5s-ext4-sysupgrade.img.gz

```
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.device='lo'
network.loopback.metric='5'
network.globals=globals
network.globals.ula_prefix='fdab:0ca1:9273::/48'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='4'
network.globals.mptcp_subflows='8'
network.globals.mptcp_add_addr_accepted='1'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_disable_initial_config='0'
network.globals.mptcp_force_multipath='1'
network.globals.mptcpd_enable='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.proto='static'
network.lan.ipaddr='172.16.100.1'
network.lan.netmask='255.255.255.0'
network.lan.device='eth0'
network.lan.ifname='eth1'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.txqueuelen='2000'
network.lan.multipath='off'
network.lan.ip4table='lan'
network.lan.label='LAN'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.metric='9999'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1=interface
network.wan1.device='eth1'
network.wan1.proto='dhcp'
network.wan1.peerdns='0'
network.wan1.ipv6='0'
network.wan1.label='Starlink'
network.wan1.multipath='on'
network.wan1.addlatency='0'
network.wan1.defaultroute='0'
network.wan1.delegate='0'
network.wan1.metric='3'
network.wan2=interface
network.wan2.device='eth2'
network.wan2.proto='static'
network.wan2.ip4table='wan'
network.wan2.multipath='master'
network.wan2.defaultroute='0'
network.wan2.delegate='0'
network.wan2.addlatency='0'
[... 97 lines cut ...]
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 36:f6:97:8f:5c:bd brd ff:ff:ff:ff:ff:ff
    inet 172.16.100.1/24 brd 172.16.100.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::34f6:97ff:fe8f:5cbd/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
3: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 6e7e:fb49:932d::
4: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
5: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr d6fb:5758:260c::
9: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
```

Ysurac, 2024-10-09:
I think kmod-r8125 module that should manage the 2.5G ports on R5S is missing, I will add it in snapshot.

Nicras, 2024-10-09:
sounds great, thank you very much

Nicras, 2024-10-09:
please tell me what to do and when

Nicras, 2024-10-10:
With the latest snapshot I was able to get OMR to work. Thank you very much for your support.

But as some others I can't get OMR-Bypass to work properly. All Clients are configured to use OMR Router as direct DNS.
But still going out via VPS

Ysurac, 2024-10-11:
Thanks for the feedback.
For OMR-ByPass, it's another issue and there is already one open for that. I will close this one.

closed 2024-10-11 by Ysurac