Ysurac/openmptcprouter #3582: How do I add an ap configuration

max93776, 2024-10-09:
## Expected Behavior

I want to add an access point configuration that uses both wan interfaces or at least one of them.
## Current Behavior

I already tried different configurations and suggestions from other users but the AP is displayed as network not associated or permanently as disabled. My network card is the Intel ax201, which as far as I know must be AP capable.
```
root@OpenMPTCProuter:~# lspci -k

00:09.0 Network controller: Intel Corporation Wi-Fi 6 AX201 (rev 20)
        Subsystem: Intel Corporation Device 0074
        Kernel driver in use: iwlwifi
```
## Specifications

  - OpenMPTCProuter version:  v0.61rc1
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64

[image]

```
[...]
Oct  9 08:21:52 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Oct  9 08:21:53 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Oct  9 08:21:53 OpenMPTCProuter user.notice omr-tracker: Launching...
Oct  9 08:22:00 OpenMPTCProuter user.notice omr-tracker: Launched
Oct  9 08:22:00 OpenMPTCProuter user.notice post-tracking-002-error: FM350GL (eth2) switched off because FM350GL may have ip issues, interface have no IPv4 gateway
Oct  9 08:22:00 OpenMPTCProuter daemon.info glorytun: starting glorytun vpn instance vpn
Oct  9 08:22:02 OpenMPTCProuter daemon.notice netifd: Wireless device 'radio0' is now down
Oct  9 08:22:02 OpenMPTCProuter daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Oct  9 08:22:02 OpenMPTCProuter daemon.info ModemManager[26342]: hotplug: add network interface wlan0: event processed
Oct  9 08:22:02 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Oct  9 08:22:02 OpenMPTCProuter daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Oct  9 08:22:02 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Oct  9 08:22:02 OpenMPTCProuter daemon.notice hostapd: wlan0: ACS-STARTED 
Oct  9 08:22:03 OpenMPTCProuter daemon.notice netifd: Wireless device 'radio0' is now up
Oct  9 08:22:03 OpenMPTCProuter daemon.notice netifd: Interface 'test' is enabled
Oct  9 08:22:03 OpenMPTCProuter daemon.notice netifd: Interface 'test' is setting up now
Oct  9 08:22:03 OpenMPTCProuter daemon.notice netifd: Interface 'test' is now up
Oct  9 08:22:03 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of test (wlan0)
Oct  9 08:22:06 OpenMPTCProuter daemon.notice [25349]: <msg> [1728462126.713248] [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:09.0': not supported by any plugin
Oct  9 08:22:10 OpenMPTCProuter daemon.err hostapd: ACS: Unable to collect survey data
Oct  9 08:22:10 OpenMPTCProuter daemon.err hostapd: ACS: All study options have failed
Oct  9 08:22:10 OpenMPTCProuter daemon.err hostapd: Interface initialization failed
Oct  9 08:22:10 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state ACS->DISABLED
Oct  9 08:22:10 OpenMPTCProuter daemon.notice hostapd: wlan0: AP-DISABLED 
Oct  9 08:22:10 OpenMPTCProuter daemon.err hostapd: ACS: Possibly channel configuration is invalid, please report this along with your config file.
Oct  9 08:22:10 OpenMPTCProuter daemon.err hostapd: ACS: Failed to start
Oct  9 08:22:10 OpenMPTCProuter daemon.notice hostapd: wlan0: AP-DISABLED 
Oct  9 08:22:10 OpenMPTCProuter daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Oct  9 08:22:10 OpenMPTCProuter daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Oct  9 08:22:10 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Oct  9 08:22:23 OpenMPTCProuter daemon.notice hostapd: Remove interface 'wlan0'
Oct  9 08:22:23 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Oct  9 08:22:23 OpenMPTCProuter daemon.notice hostapd: wlan0: AP-DISABLED 
Oct  9 08:22:23 OpenMPTCProuter daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING 
Oct  9 08:22:23 OpenMPTCProuter daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Oct  9 08:22:23 OpenMPTCProuter daemon.notice netifd: Interface 'test' is now down
Oct  9 08:22:23 OpenMPTCProuter daemon.notice netifd: Interface 'test' is disabled
Oct  9 08:22:23 OpenMPTCProuter daemon.info ModemManager[27987]: hotplug: remove network interface wlan0: event processed
Oct  9 08:22:23 OpenMPTCProuter daemon.notice [25349]: <msg> [1728462143.891598] [base-manager] port wlan0 released by device '/sys/devices/pci0000:00/0000:00:09.0'
Oct  9 08:22:23 OpenMPTCProuter daemon.notice netifd: Wireless device 'radio0' is now down
Oct  9 08:22:23 OpenMPTCProuter daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Oct  9 08:22:23 OpenMPTCProuter daemon.info ModemManager[28153]: hotplug: add network interface wlan0: event processed
Oct  9 08:22:24 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Oct  9 08:22:24 OpenMPTCProuter daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Oct  9 08:22:24 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ACS
Oct  9 08:22:24 OpenMPTCProuter daemon.notice hostapd: wlan0: ACS-STARTED 
[... 58 lines cut ...]
Oct  9 08:24:33 OpenMPTCProuter daemon.err hostapd: Interface initialization failed
Oct  9 08:24:33 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state ACS->DISABLED
Oct  9 08:24:33 OpenMPTCProuter daemon.notice hostapd: wlan0: AP-DISABLED 
Oct  9 08:24:33 OpenMPTCProuter daemon.err hostapd: ACS: Possibly channel configuration is invalid, please report this along with your config file.
Oct  9 08:24:33 OpenMPTCProuter daemon.err hostapd: ACS: Failed to start
Oct  9 08:24:33 OpenMPTCProuter daemon.notice hostapd: wlan0: AP-DISABLED 
Oct  9 08:24:33 OpenMPTCProuter daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Oct  9 08:24:33 OpenMPTCProuter daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Oct  9 08:24:33 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Oct  9 08:25:00 OpenMPTCProuter user.notice omr-schedule-010-services: Fix Multipath status on test (wlan0)
Oct  9 08:25:00 OpenMPTCProuter user.notice MPTCP: Disabling MPTCP on interface wlan0 not found in enabled multipath list
Oct  9 08:25:04 OpenMPTCProuter daemon.notice netifd: Interface 'test' is now down
Oct  9 08:25:04 OpenMPTCProuter daemon.notice netifd: Interface 'test' is disabled
Oct  9 08:25:04 OpenMPTCProuter daemon.notice hostapd: Remove interface 'wlan0'
Oct  9 08:25:04 OpenMPTCProuter daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Oct  9 08:25:04 OpenMPTCProuter daemon.notice hostapd: wlan0: AP-DISABLED 
Oct  9 08:25:04 OpenMPTCProuter daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING 
Oct  9 08:25:04 OpenMPTCProuter daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Oct  9 08:25:04 OpenMPTCProuter daemon.info ModemManager[13688]: hotplug: remove network interface wlan0: event processed
Oct  9 08:25:04 OpenMPTCProuter daemon.notice [25349]: <msg> [1728462304.324570] [base-manager] port wlan0 released by device '/sys/devices/pci0000:00/0000:00:09.0'
Oct  9 08:25:04 OpenMPTCProuter daemon.notice netifd: Wireless device 'radio0' is now down
Oct  9 08:25:04 OpenMPTCProuter daemon.notice netifd: Wireless device 'radio0' is now up
Oct  9 10:25:05 OpenMPTCProuter daemon.info dnsmasq[29569]: read /etc/hosts - 4 addresses
Oct  9 10:25:05 OpenMPTCProuter daemon.info dnsmasq[29569]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
```

Ysurac, 2024-10-09:
What do you have in Network->Wireless menu ?

max93776, 2024-10-22:
I was able to create an ap, but connection does not wan't to establish:
```
root@OpenMPTCProuter:/etc/config# cat wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'pci0000:00/0000:00:09.0'
        option cell_density '0'
        option country 'DE'
        option htmode 'HT20'
        option channel '6'
        option legacy_rates '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'MeinWLAN'
        option encryption 'none'
        option network 'wan'
```
If I try to connect my phone, it says "IP address is retrieved", stays in this mode then stop the connection :
[image]
```
Logread tail:
Oct 22 12:24:17 OpenMPTCProuter daemon.info hostapd: wlan0: STA 04:4a:6xx IEEE 802.11: authenticated
Oct 22 12:24:17 OpenMPTCProuter daemon.info hostapd: wlan0: STA 04:4a:6xx IEEE 802.11: associated (aid 1)
Oct 22 12:24:17 OpenMPTCProuter daemon.notice hostapd: wlan0: AP-STA-CONNECTED 04:4a:6xx
Oct 22 12:24:53 OpenMPTCProuter daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 04:4a:6xx
```
Maybe it's a firewall issue or something...

Ysurac, 2024-10-22:
You set it on "option network wan", check if this interface exist, and have DHCP enabled (it's often easier to set it on lan interface).

max93776, 2024-10-23:
I set the `option network 'lan'` and added an interface test in wan firewall zone with dhcp client protocol. It seems to work now but slowly. Do you know if this is the prefered setup?
[image]

max93776, 2024-11-08:
My final question on the topic: I saw that a patch from hostapd is necessary for the 5GHz functionality of the Intel cards under Linux or in OpenWRT. Since omr apparently doesn't use hostapd but wpad, my question is how I can solve this problem (patch wpad, patch and use hostapd?). [Here ](https://gist.github.com/iffa/290b1b83b17f51355c63a97df7c1cc60)is the discussion in github. Thank you in advance!

closed 2025-02-11 by github-actions[bot]