Ysurac/openmptcprouter #3584: Need help with configuration of OMR-Bypass

vi7vo7, 2024-10-09:
## Expected Behavior
In version v0.59.1, I was able to configure OMR-Bypass so that a specific WAN interface was used when working with a specific IP address. And it worked perfectly.
However, after upgrading to newer versions, I can’t achieve the same functionality. At first, I thought it was due to bugs in OMR-Bypass, but now I think I might have missed something, and that’s why I can’t set it up properly. 

## Current Behavior
OMR-Bypass and the route does not work properly. 
I can see this, for example, from the results of the traceroute command. 
[image]

## Specifications

  - OpenMPTCProuter version: 0.61rc3-6.6
  - OpenMPTCProuter VPS version: 0.1031-test 6.6.36-x64v2-xanmod1
  - OpenMPTCProuter VPS provider: self-hosted
  - OpenMPTCProuter platform: x64
  
Here is my configuration: 
[image]

I am using V2Ray TROJAN because, based on my configuration and tests, it shows the best performance. For testing purposes, I tried using all other proxies, but the result was always the same—OMR-Bypass doesn't work. 
[image]

OMR-Bypass settings:
[image]

Can someone help me figure out what's going on? Am I configuring it incorrectly, or is this a bug in OMR-Bypass?

Ysurac, 2024-10-10:
Can you test latest snapshot ( https://github.com/Ysurac/openmptcprouter/wiki/Snapshots ) ?

vi7vo7, 2024-10-10:
I've tested this image - https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/openmptcprouter-v0.61rc3-6.6-r0+27346-c7ba5574f5-x86-64-generic-ext4-combined-efi.img.gz  (October 9, 2024) and I don't see any difference. My bypassed IP still goes through VPS.

Ysurac, 2024-10-11:
You are testing only with tracert ? As it seems to be working using TCP/UDP but no ICMP (I will fix that).

vi7vo7, 2024-10-11:
That's the thing! I was indeed testing only ICMP. I checked TCP—it works both on the RC3 and on the snapshot.

hle5128, 2024-10-14:
this rc3 fixed omr-bypass that i have been on rc0 for months...wow...
still get a lot of warning but its actually fixed it.

hle5128, 2024-10-14:
i take this back, it stopped working now very weird. i only got it working 1 time...

closed 2025-01-17 by github-actions[bot]