Ysurac/openmptcprouter #3585: Poor overall performance, particularly Google Meet

arraypad, 2024-10-09:
I'm using OpenMPTCProuter with three WAN connections:

- VDSL: about 50mbps/10mbps
- Starlink: about 150mbps/15mbps
- 4G: about 140mbps/20mbps

Both the router and the "VPS" end of the tunnel are capable amd64 machines.

The other end of the tunnel is a 7gbps/1gbps residential fiber connection which is 10km away.

## Expected Behavior
I expect a connection which is more stable and faster than using any of the individual connections.

## Current Behavior
A speedtest usually reports around 250mbps/50mbps which sounds fine, but in practice the connection is not very stable and is often subjectively worse than using one connection alone.

The most obvious example is Google Meet, where using the OpenMPTCProuter connection usually starts to stall participant video streams with more than one other participant. This is even when limiting the receive resolution to 320p, which should be _easily_ accommodated with the available bandwidth. Using one of the connections individually (e.g. Starlink) is usually fine with 5+ other participants.

I don't want to force all Google Meet traffic thru one WAN connection because that's not making use of the bandwidth and redundancy available.

I've tried setting different WAN connections as the multipath master, and also using SQM. I haven't noticed any great difference in any configuration.

However my experience using OpenMPTCProuter with only the Starlink WAN connection set to master and the other two WAN connections disabled is significantly worse than just connecting directly to Starlink.

## Specifications

  - OpenMPTCProuter version: 0.60-6.1
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-25-amd64
  - OpenMPTCProuter VPS provider: free.fr residential
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-10-10:
Google Meet is using UDP by default, you can try another VPN in System->OpenMPTCProuter, "Wizard" tab, "Advanced settings" checkbox.
Speedtest are done over TCP that is using the proxy part.

datapharmer, 2024-10-14:
It is possible that google is treating your VPS traffic differently than direct traffic. I have encountered several providers that heavily throttle traffic from hosting ranges to limit bot activity and you may be the unfortunate victim of the bot wars.

You also may be having issues due to latency. You can try optimizing the network for latency instead of speed to see if that helps.

arraypad, 2024-10-22:
Thanks for the suggestions!

Switching the VPN to XRay VLESS seems to have helped.

@datapharmer where is the option to optimize for latency? I remember seeing this in the past but can't find it now.

kevinh-csalabs, 2024-10-29:
@arraypad under system > openmptcprouter >advanced settings (tab, not checkbox) > "Enable TCP Low Latency"

closed 2025-02-02 by github-actions[bot]