Ysurac/openmptcprouter #3587: ShadowSocks Rust is not running

pauleharris, 2024-10-10:
## Expected Behavior

## Current Behavior
Hi,
RasPi 5, 6.1 image, debian 12 on OVH.
I was able to get the routers up and running (showing green). The RasPi shows "ShadowSocks Rust is not running" with the following in the logs:
Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
OpenMPTCProuter daemon.warn openvpn(omr)[15346]: Use --help for more information.
OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on X.X.X.X, server API answer on X.X.X.X)

I've reinstalled the VPS multiple times, checked that Starlette and FastAPI are installed. Where and how do I generate the CA?

Thank you

## Possible Solution

## Steps to Reproduce the Problem
100% repo for me.  Install on RASP5 with 6.1.  Install and reinstall on AWS and OVH.

## Context (Environment)
Unable to bond or connect to VPS.

## Specifications
  - OpenMPTCProuter VPS version:  Debian 11 and 12 fresh - all with with the latest update/install scripts (I tried 11, then 12, no difference)
  - OpenMPTCProuter VPS provider: AWS and OVH
  - OpenMPTCProuter platform: RPI5
  - Country: 
(For some reason my when I rebooted my PI it's not in Spanish....)
Modelo | Raspberry Pi 5 Model B Rev 1.0
Arquitectura | ARMv8 Processor rev 1
Plataforma de destino | bcm27xx/bcm2712
Versión del firmware | openmptcprouter v0.60-6.1 r0+24843-acf40c022e / LuCI Master git-24.006.68576-8debeeb
Versión del núcleo | 6.1.71
Hora local | 2024-10-10 19:16:50
Tiempo de actividad | 0h 6m 8s
Carga media | 0.18, 0.17, 0.09

Ysurac, 2024-10-11:
Check your server key.

pauleharris, 2024-10-11:
I feel I've tried this several times and just reinstalled the RPi to check.

I'm using the value from here:
Your OpenMPTCProuter Server key: XXXXXXXXXX
in openmptcprouter_config.txt root of the VPS server

Ysurac, 2024-10-12:
What do you have in Status->System log ?
On the VPS, what is the result of ```journalctl -u omr-admin``` ?

pauleharris, 2024-10-12:
Oct 06 17:25:49 vps-4f94d97b systemd[1]: Started omr-admin.service - OMR-Admin.
Oct 06 20:15:27 vps-4f94d97b systemd[1]: Stopping omr-admin.service - OMR-Admin...
Oct 06 20:15:28 vps-4f94d97b systemd[1]: omr-admin.service: Deactivated successfully.
Oct 06 20:15:28 vps-4f94d97b systemd[1]: Stopped omr-admin.service - OMR-Admin.
Oct 06 20:15:28 vps-4f94d97b systemd[1]: omr-admin.service: Consumed 14.870s CPU time.
-- Boot 7335453b28f34842bd393a242dcd265e --
Oct 06 20:15:34 vps-4f94d97b systemd[1]: Started omr-admin.service - OMR-Admin.
Oct 06 20:36:48 vps-4f94d97b systemd[1]: Stopping omr-admin.service - OMR-Admin...
Oct 06 20:36:48 vps-4f94d97b systemd[1]: omr-admin.service: Deactivated successfully.
Oct 06 20:36:48 vps-4f94d97b systemd[1]: Stopped omr-admin.service - OMR-Admin.
Oct 06 20:36:48 vps-4f94d97b systemd[1]: omr-admin.service: Consumed 4.310s CPU time.
Oct 06 20:36:57 vps-4f94d97b systemd[1]: Started omr-admin.service - OMR-Admin.
Oct 06 20:39:29 vps-4f94d97b systemd[1]: Stopping omr-admin.service - OMR-Admin...
Oct 06 20:39:29 vps-4f94d97b systemd[1]: omr-admin.service: Deactivated successfully.
Oct 06 20:39:29 vps-4f94d97b systemd[1]: Stopped omr-admin.service - OMR-Admin.
Oct 06 20:39:29 vps-4f94d97b systemd[1]: omr-admin.service: Consumed 1.304s CPU time.
Oct 06 20:39:29 vps-4f94d97b systemd[1]: Started omr-admin.service - OMR-Admin.
Oct 06 20:41:05 vps-4f94d97b systemd[1]: Stopping omr-admin.service - OMR-Admin...
Oct 06 20:41:05 vps-4f94d97b systemd[1]: omr-admin.service: Deactivated successfully.
Oct 06 20:41:05 vps-4f94d97b systemd[1]: Stopped omr-admin.service - OMR-Admin.
Oct 06 20:41:05 vps-4f94d97b systemd[1]: omr-admin.service: Consumed 1.212s CPU time.
-- Boot 824cc01c48af4ff7802c052f775f4a67 --
Oct 06 20:41:11 vps-4f94d97b systemd[1]: Started omr-admin.service - OMR-Admin.
Oct 06 20:57:56 vps-4f94d97b systemd[1]: Stopping omr-admin.service - OMR-Admin...
Oct 06 20:57:57 vps-4f94d97b systemd[1]: omr-admin.service: Deactivated successfully.
Oct 06 20:57:57 vps-4f94d97b systemd[1]: Stopped omr-admin.service - OMR-Admin.
Oct 06 20:57:57 vps-4f94d97b systemd[1]: omr-admin.service: Consumed 3.629s CPU time.
-- Boot 3a02b03e107f45f583fc17f95ffacb49 --
Oct 06 20:58:03 vps-4f94d97b systemd[1]: Started omr-admin.service - OMR-Admin.

pauleharris, 2024-10-12:
Oct 12 13:47:22 OpenMPTCProuter daemon.err openvpn(omr)[8399]: Options error: You must define CA file (--ca) or CA path (--capath) and/or peer fingerprint verification (--peer-fingerprint)
Oct 12 13:47:22 OpenMPTCProuter daemon.warn openvpn(omr)[8399]: Use --help for more information.
Oct 12 13:47:26 OpenMPTCProuter user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on X.X.X.X, server API answer on X.X.X.X)

pauleharris, 2024-10-16:
Hi, are there more logs or info that would help me troubleshoot this?

Thanks

Ysurac, 2024-10-16:
"omr-admin" ont the VPS seems to restart really often, check if it's really the case or not (I see many boot entry).
on the router side, via ssh, run ```/etc/init.d/openmptcprouter-vps restart``` give me the result and the result of ```uci show openmptcprouter``` and ```uci show shadowsocks-rust```

pauleharris, 2024-10-16:
root@OpenMPTCProuter:~# uci show openmptcprouter
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.sipalg='1'
openmptcprouter.settings.vpn='openvpn'
openmptcprouter.settings.scaling_min_freq='2400000'
openmptcprouter.settings.scaling_max_freq='2400000'
openmptcprouter.settings.scaling_governor='performance'
openmptcprouter.settings.master='balancing'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='3'
openmptcprouter.settings.status_getip_timeout='1'
openmptcprouter.settings.enable_nodelay='1'
openmptcprouter.settings.menu='OpenMPTCProuter'
openmptcprouter.settings.disableintfrename='1'
openmptcprouter.settings.tracebox='0'
openmptcprouter.settings.version='0.60-6.1'
openmptcprouter.settings.allmptcpovervpn='0'
openmptcprouter.settings.serial='0XXXXXXXXXXXX2'
openmptcprouter.settings.mptcpovervpn='wireguard'
openmptcprouter.settings.dns64='0'
openmptcprouter.settings.proxy='shadowsocks-rust'
openmptcprouter.settings.encryption='aes-256-gcm'
openmptcprouter.settings.apilc='1729118148'
openmptcprouter.settings.ha='0'
openmptcprouter.vps=server
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.set_firewall='1'
openmptcprouter.vps.get_config='1'
openmptcprouter.vps.current='1'
openmptcprouter.vps.password='6XXXXXXXXXXXXXXXXXXX8'
openmptcprouter.vps.admin_error='1'
openmptcprouter.vps.token_error='1'
openmptcprouter.vps.ip='X.X.X.236'
openmptcprouter.omr=router
openmptcprouter.omr.ss_sss0='down'
openmptcprouter.omr.detected_public_ipv4='X.X.X.66'
openmptcprouter.loopback=interface
openmptcprouter.loopback.multipath='off'
openmptcprouter.loopback.metric='5'
openmptcprouter.lan=interface
openmptcprouter.lan.multipath='off'
openmptcprouter.lan.metric='9999'
openmptcprouter.lan.multipathvpn='0'
openmptcprouter.lan.testspeed='0'
openmptcprouter.wan1=interface
openmptcprouter.wan1.multipath='master'
openmptcprouter.wan1.metric='7'
openmptcprouter.wan1.multipathvpn='0'
openmptcprouter.wan1.testspeed='0'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.publicip='X.X.X.66'
openmptcprouter.wan1.latency='54'
openmptcprouter.wan1.asn='NET1'
openmptcprouter.wan1.manufacturer='huawei'
openmptcprouter.wan1.latency_previous='53'
openmptcprouter.wan1.mptcp_status='MPTCP enabled'
openmptcprouter.wan1.lc='1729118148'
openmptcprouter.wan2=interface
openmptcprouter.wan2.multipath='on'
openmptcprouter.wan2.metric='8'
openmptcprouter.wan2.multipathvpn='0'
openmptcprouter.wan2.testspeed='0'
openmptcprouter.wan2.state='up'
openmptcprouter.wan2.publicip='X.X.X.77'
openmptcprouter.wan2.latency='66'
openmptcprouter.wan2.asn='INET2'
openmptcprouter.wan2.manufacturer='huawei'
openmptcprouter.wan2.latency_previous='66'
openmptcprouter.wan2.mptcp_status='MPTCP enabled'
openmptcprouter.wan2.lc='1729119749'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='1500'
openmptcprouter.omr6in4=interface
openmptcprouter.omr6in4.multipath='off'
openmptcprouter.omr6in4.metric='1201'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.omr='0.60-6.1'
openmptcprouter.latest_versions.vps='0.1030'
openmptcprouter.latest_versions.lc='1729118146'

root@OpenMPTCProuter:~# uci show shadowsocks-rust
shadowsocks-rust.hi1=ss_redir
shadowsocks-rust.hi1.server='sss0'
shadowsocks-rust.hi1.local_address='::'
shadowsocks-rust.hi1.local_port='1100'
shadowsocks-rust.hi1.mode='tcp_and_udp'
shadowsocks-rust.hi1.timeout='1000'
shadowsocks-rust.hi1.fast_open='0'
shadowsocks-rust.hi1.verbose='0'
shadowsocks-rust.hi1.syslog='1'
shadowsocks-rust.hi1.reuse_port='0'
shadowsocks-rust.hi1.mptcp='1'
shadowsocks-rust.hi1.ipv6_first='1'
shadowsocks-rust.hi1.no_delay='0'
shadowsocks-rust.hi2=ss_redir
shadowsocks-rust.hi2.server='sss1'
shadowsocks-rust.hi2.local_address='::'
shadowsocks-rust.hi2.local_port='1100'
shadowsocks-rust.hi2.mode='tcp_and_udp'
shadowsocks-rust.hi2.timeout='1000'
shadowsocks-rust.hi2.fast_open='0'
shadowsocks-rust.hi2.verbose='0'
shadowsocks-rust.hi2.syslog='1'
shadowsocks-rust.hi2.reuse_port='0'
shadowsocks-rust.hi2.mptcp='1'
shadowsocks-rust.hi2.ipv6_first='1'
shadowsocks-rust.hi2.no_delay='0'
shadowsocks-rust.ss_rules=ss_rules
shadowsocks-rust.ss_rules.disabled='0'
shadowsocks-rust.ss_rules.redir_tcp='hi1'
shadowsocks-rust.ss_rules.src_default='forward'
shadowsocks-rust.ss_rules.dst_default='forward'
shadowsocks-rust.ss_rules.local_default='forward'
shadowsocks-rust.ss_rules.server='sss0'
shadowsocks-rust.sss0=server
shadowsocks-rust.sss0.disabled='0'
shadowsocks-rust.sss0.server='X.X.X.236'
shadowsocks-rust.sss0.server_port='65280'
shadowsocks-rust.sss0.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss0.obfs='0'
shadowsocks-rust.sss1=server
shadowsocks-rust.sss1.disabled='1'
shadowsocks-rust.sss1.server='X.X.X.3'
shadowsocks-rust.sss1.server_port='65280'
shadowsocks-rust.sss1.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss1.obfs='0'
shadowsocks-rust.dns=ss_tunnel
shadowsocks-rust.dns.disabled='1'
shadowsocks-rust.dns.mode='tcp_and_udp'
shadowsocks-rust.dns.server='sss0'
shadowsocks-rust.dns.local_port='5353'
shadowsocks-rust.dns.tunnel_address='8.8.8.8:53'
shadowsocks-rust.tracker_sss0=ss_local
shadowsocks-rust.tracker_sss0.server='sss0'
shadowsocks-rust.tracker_sss0.local_address='127.0.0.1'
shadowsocks-rust.tracker_sss0.local_port='1111'
shadowsocks-rust.tracker_sss0.mode='tcp_and_udp'
shadowsocks-rust.tracker_sss0.timeout='600'
shadowsocks-rust.tracker_sss0.fast_open='0'
shadowsocks-rust.tracker_sss0.syslog='0'
shadowsocks-rust.tracker_sss0.reuse_port='1'
shadowsocks-rust.tracker_sss0.mptcp='1'
shadowsocks-rust.tracker_sss0.verbose='0'
root@OpenMPTCProuter:~#

Ysurac, 2024-10-17:
It's not able to contact correctly the API.
What is the result, via SSH on the router, of ```curl -k https://<vpsip>:65500/``` ?

pauleharris, 2024-10-17:
url -k https://X.X.X.236:65500/
"Welcome to OpenMPTCProuter Server part"root@OpenMPTCProuter:~#

pauleharris, 2024-10-23:
Hi,
Is more information needed here?
Thanks

pauleharris, 2024-10-23:
I just grabbed the latest image for Pi5 (6.6) and that fixed it.  Not sure why.  Thanks for the support.

closed 2024-10-23 by pauleharris