Ysurac/openmptcprouter #3589: Download speed on single connection is better than when aggregated

muko82, 2024-10-11:
## Expected Behavior
Getting real aggregated connections with two 4g SIM.

## Current Behavior
While using omr-test-speed on wgwan2 and wgwan3 (I need both in VPN, one provider is filtering mptcp) i get this result:

root@OpenMPTCProuter:~# omr-test-speed
Select best test server...
host: scaleway.testdebit.info - ping: 155
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping: 82
host: lyon.testdebit.info - ping:
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping:
host: appliwave.testdebit.info - ping: 51
host: speedtest.frankfurt.linode.com - ping: 32
host: speedtest.tokyo2.linode.com - ping: 256
host: speedtest.singapore.linode.com - ping: 287
host: speedtest.newark.linode.com - ping: 134
host: speedtest.atlanta.linode.com - ping: 133
host: speedtest.dallas.linode.com - ping: 151
host: speedtest.fremont.linode.com - ping: 183
host: speed.hetzner.de - ping: 52
host: ipv4.bouygues.testdebit.info - ping:
host: par.download.datapacket.com - ping: 45
host: nyc.download.datapacket.com - ping: 117
host: ams.download.datapacket.com - ping: 41
host: fra.download.datapacket.com - ping: 36
host: lon.download.datapacket.com - ping: 47
host: mad.download.datapacket.com - ping: 55
host: prg.download.datapacket.com - ping: 46
host: sto.download.datapacket.com - ping: 63
host: vie.download.datapacket.com - ping: 39
host: war.download.datapacket.com - ping: 62
host: atl.download.datapacket.com - ping: 133
host: chi.download.datapacket.com - ping: 131
host: lax.download.datapacket.com - ping: 181
host: mia.download.datapacket.com - ping: 153
host: nyc.download.datapacket.com - ping: 118
host: speedtest.milkywan.fr - ping: 45
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  125M    0  125M    0     0  5179k      0 --:--:--  0:00:24 --:--:-- 6299k

On wgwan2:

root@OpenMPTCProuter:~# omr-test-speed wgwan2
Select best test server...
host: scaleway.testdebit.info - ping: 54
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping: 59
host: lyon.testdebit.info - ping:
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping:
host: appliwave.testdebit.info - ping: 55
host: speedtest.frankfurt.linode.com - ping: 37
host: speedtest.tokyo2.linode.com - ping: 260
host: speedtest.singapore.linode.com - ping: 289
host: speedtest.newark.linode.com - ping: 132
host: speedtest.atlanta.linode.com - ping: 139
host: speedtest.dallas.linode.com - ping: 150
host: speedtest.fremont.linode.com - ping: 181
host: speed.hetzner.de - ping: 46
host: ipv4.bouygues.testdebit.info - ping:
host: par.download.datapacket.com - ping: 42
host: nyc.download.datapacket.com - ping: 117
host: ams.download.datapacket.com - ping: 39
host: fra.download.datapacket.com - ping: 39
host: lon.download.datapacket.com - ping: 44
host: mad.download.datapacket.com - ping: 53
host: prg.download.datapacket.com - ping: 44
host: sto.download.datapacket.com - ping: 63
host: vie.download.datapacket.com - ping: 40
host: war.download.datapacket.com - ping: 57
host: atl.download.datapacket.com - ping: 129
host: chi.download.datapacket.com - ping: 125
host: lax.download.datapacket.com - ping: 181
host: mia.download.datapacket.com - ping: 138
host: nyc.download.datapacket.com - ping: 115
host: speedtest.milkywan.fr - ping: 45
Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 59.8M    0 59.8M    0     0  5729k      0 --:--:--  0:00:10 --:--:-- 6222k

On wgwan3:

root@OpenMPTCProuter:~# omr-test-speed wgwan3
Select best test server...
host: scaleway.testdebit.info - ping: 53
host: bordeaux.testdebit.info - ping:
host: aix-marseille.testdebit.info - ping: 59
host: lyon.testdebit.info - ping:
host: lille.testdebit.info - ping:
host: paris.testdebit.info - ping:
host: appliwave.testdebit.info - ping: 44
host: speedtest.frankfurt.linode.com - ping: 39
host: speedtest.tokyo2.linode.com - ping: 253
host: speedtest.singapore.linode.com - ping: 293
host: speedtest.newark.linode.com - ping: 145
host: speedtest.atlanta.linode.com - ping: 139
host: speedtest.dallas.linode.com - ping: 153
host: speedtest.fremont.linode.com - ping: 178
host: speed.hetzner.de - ping: 41
host: ipv4.bouygues.testdebit.info - ping:
host: par.download.datapacket.com - ping: 42
host: nyc.download.datapacket.com - ping: 118
host: ams.download.datapacket.com - ping: 42
host: fra.download.datapacket.com - ping: 32
host: lon.download.datapacket.com - ping: 52
host: mad.download.datapacket.com - ping: 57
host: prg.download.datapacket.com - ping: 51
host: sto.download.datapacket.com - ping: 67
host: vie.download.datapacket.com - ping: 37
host: war.download.datapacket.com - ping: 61
host: atl.download.datapacket.com - ping: 147
host: chi.download.datapacket.com - ping: 133
host: lax.download.datapacket.com - ping: 190
host: mia.download.datapacket.com - ping: 167
host: nyc.download.datapacket.com - ping: 118
host: speedtest.milkywan.fr - ping: 48
Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 9536M    0  299k    0     0  18706      0   6d 04h  0:00:16   6d 04h 21043

## Specifications

  - OpenMPTCProuter version: 0.59.1r0 with kernel 5.4
  - OpenMPTCProuter VPS version: 0.130 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Aruba Networks
  - OpenMPTCProuter platform: RPI4b

CURRENT CONFIGURATION:
Default VPN: GloryTUN TCP
MPTCP over VPN: Wireguard
Both 4g connection have: MPTCP over VPN, Enable SQM

Multipath TCP scheduler: ECF
SYN Retries: 3
Multipath TCP version: 0
Congestion control: bbr

Ysurac, 2024-10-11:
You have an old version, no more supported.
Are you sure one connection is blocking MPTCP (the message in the status page is only a dirty test that is not always working) ? If only one connection is blocking MPTCP, use MPTCP over VPN only on this connection.
Why are you using ECF ?
What is the proxy you are using ?

muko82, 2024-10-11:
I know but I checked the latest one (0.60) with both kernel 5.4 and 6.1 and it seems to me that it's not stable on rpi4

It always loose the connection.

I'm using shadowsocks and yes I'm sure that mptcp is filtered on one connection but if can give me a cli command to have a definitive test I will do

Ysurac, 2024-10-11:
Put MPTCP over VPN only on blocked connection.
To test if it's working on test, only check without MPTCP over VPN if you have aggregation in Network->MPTCP, "bandwidth" tab.

muko82, 2024-10-11:
> Put MPTCP over VPN only on blocked connection. To test if it's working on test, only check without MPTCP over VPN if you have aggregation in Network->MPTCP, "bandwidth" tab.

I already did that

muko82, 2024-10-11:
I've reinstalled both router and VPS with 0.60 and kernel 6.1 but I have the same issue.
With this solution I cannot use wireguard: it seems that the gateway is not reachable: do I need to apply some additional setting?

Ysurac, 2024-10-12:
Try https://github.com/Ysurac/openmptcprouter/issues/3528
I fixed some MPTCP over VPN issues.

muko82, 2024-10-12:
Kernel 6,1 is enough?

muko82, 2024-10-12:
It seems that with the latest image and kernel 6.6 donwload is now working as expected but I have two issues:

1- Upload si now 2Mbit/s
2- After few minutes of usage filesystem goes in read-only. I already tried with fsck but it doen't detects any issue.

I'm using ext4-factory image of 0.61 rc3 with kernel 6.6

muko82, 2024-10-12:
Update: it seems that the filesystem is being settled in read-only immediately after I try to restart the wifi module.

Ysurac, 2024-10-12:
I think it's due to a power problem. Check power supply and that you use an USB powered hub

muko82, 2024-10-13:
You're right mate. It seems for some reason that rpi is getting throttle even with different power suppliers.
This isn't happening with kernel 5.4: do you know why?

muko82, 2024-10-13:
UPDATE: I got it working fine using 0.61 with kernel 5.4 and with exactly the same behaviour as for kernel 6.6

Searching on internet it seems that kernel 6.6 is giving issues on RPI4 and usb ports (i found a thread on github about a user with home assistant with exactly the same problem and who downgraded the kernel to 6.1).
Any chance to have 0.61 with kernel 6.1? :)

darkman1983, 2024-10-19:
Hopefully he will update Kernel for router to the same Kernel as vor VPS-Part. Have you checked, if the problem is also present in Kernel 6.10?

muko82, 2024-10-20:
> Hopefully he will update Kernel for router to the same Kernel as for VPS-Part. Have you checked, if the problem is also present in Kernel 6.10?

No, I didn't yet but I will this week

closed 2025-01-24 by github-actions[bot]