Ysurac/openmptcprouter #3593: v0.61rc3-6.6 omr-bypass not working and multle errors

hle5128, 2024-10-14:
## Expected Behavior
Expect OMR bypass:
Domain
IP
Should bypass and use any of WAN to bypass VPN:
GET: https://rest-prod.immedia-semi.com should have result 
Blink API
Not 406 Error (VPN detected)

## Current Behavior
OMR did not bypass 
Domain
IP

## Possible Solution
?

## Steps to Reproduce the Problem

  1. set immedia-semi.com bypass DOMAIN
  2.curl -v https://rest-prod.immedia-semi.com
  3. show HTTP/2 406 message indicating behind VPS (detected VPN so they blocked)

## Context (Environment)
Show logs:

```
Oct 14 04:07:01 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Oct 14 04:07:02 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 14 04:07:05 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Oct 14 00:07:11 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) d8:1f:12:cf:ed:ed 
Oct 14 00:07:11 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.143 d8:1f:12:cf:ed:ed 
Oct 14 00:07:11 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.143 d8:1f:12:cf:ed:ed 
Oct 14 00:07:11 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.143 d8:1f:12:cf:ed:ed TY_WR
Oct 14 00:07:35 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.202 60:e8:5b:7e:89:82 
Oct 14 00:07:35 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.202 60:e8:5b:7e:89:82 airthings-view
Oct 14 00:07:42 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) cc:8c:bf:6f:0c:89 
Oct 14 00:07:42 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.230 cc:8c:bf:6f:0c:89 
Oct 14 00:07:42 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.230 cc:8c:bf:6f:0c:89 
Oct 14 00:07:42 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.230 cc:8c:bf:6f:0c:89 TY_WR
Oct 14 04:10:04 OpenMPTCProuter authpriv.info dropbear[13190]: Exit (root) from <192.168.100.169:63817>: Exited normally
Oct 14 00:10:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.202 60:e8:5b:7e:89:82 
Oct 14 00:10:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.202 60:e8:5b:7e:89:82 airthings-view
Oct 14 00:10:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.202 60:e8:5b:7e:89:82 
Oct 14 00:10:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.202 60:e8:5b:7e:89:82 airthings-view
Oct 14 00:10:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.202 60:e8:5b:7e:89:82 
Oct 14 00:10:05 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.202 60:e8:5b:7e:89:82 airthings-view
Oct 14 00:10:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.202 60:e8:5b:7e:89:82 
Oct 14 00:10:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.202 60:e8:5b:7e:89:82 airthings-view
Oct 14 04:10:42 OpenMPTCProuter daemon.info omr-tracker-v2ray: Reload V2Ray rules
Oct 14 04:10:43 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Oct 14 04:10:43 OpenMPTCProuter user.notice omr-bypass: Restart dnsmasq...
Oct 14 00:10:43 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Oct 14 00:10:46 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
[... 225 lines cut ...]
local fe80::873d:ea35:4ad5:3dc2 dev tun0 table local proto kernel metric 0 pref medium
local fe80::a45d:5bff:fe99:8ab dev eth2 table local proto kernel metric 0 pref medium
local fe80::b082:43ff:fef9:af80 dev eth0 table local proto kernel metric 0 pref medium
local fe80::ecbe:b4ff:fe8d:28c8 dev ifb4eth3 table local proto kernel metric 0 pref medium
multicast ff00::/8 dev eth1 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth2 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth0 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev tun0 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth3 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev ifb4eth3 table local proto kernel metric 256 pref medium
```

root@OpenMPTCProuter:~# ip route get 8.8.8.8 mark 0x5393
```
8.8.8.8 via 10.255.252.1 dev tun0 src 10.255.252.2 mark 0x5393 uid 0 
    cache 
```

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61rc3-6.6 r0+27346-c7ba5574f5 / LuCI Master 24.212.79282~65b8002
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:  Version 0.1031-test 6.6.36-x64v3-xanmod1 (Latest available version 0.1030)
  - OpenMPTCProuter platform: R5S
  - Country: USA

Ysurac, 2024-10-14:
You need to fix your firewall redirect rule: "OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part"
Not sure if this can make problems with others rules, so I would need the result of ```uci show omr-bypass```

hle5128, 2024-10-14:
> You need to fix your firewall redirect rule: "OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part" Not sure if this can make problems with others rules, so I would need the result of `uci show omr-bypass`

Hi,
We never set any port forward, these are a fresh reinstallation setup so does VPS (we tried to ensure there is not problem)
we don't know why redirect ports >= 65000

```
root@OpenMPTCProuter:~# uci show omr-bypass
omr-bypass.all=interface
omr-bypass.m6replay=proto
omr-bypass.m6replay.url='m6web.fr' '6play.fr' '6cloud.fr'
omr-bypass.mycanal=proto
omr-bypass.mycanal.url='mycanal.fr' 'canal-plus.com' 'canalplus.com' 'canalplus-cdn.net' 'canalplus.pro' 'canal-plus.net'
omr-bypass.minecraft=proto
omr-bypass.minecraft.url='authserver.mojang.com'
omr-bypass.lesnumeriques=proto
omr-bypass.lesnumeriques.url='lesnumeriques.com' 'botscorner.com' 'app.botscorner.com'
omr-bypass.disneyplus=proto
omr-bypass.disneyplus.url='bamgrid.com' 'disney-plus.net'
omr-bypass.amazonvideo=proto
omr-bypass.amazonvideo.url='cloudfront.net' 'llnw.net'
omr-bypass.lo=interface
omr-bypass.lo.id='6'
omr-bypass.eth0=interface
omr-bypass.eth0.id='3'
omr-bypass.eth1=interface
omr-bypass.eth1.id='9999'
omr-bypass.eth2=interface
omr-bypass.eth2.id='5'
omr-bypass.eth3=interface
omr-bypass.eth3.id='6'
omr-bypass.tun0=interface
omr-bypass.tun0.id='1500'
omr-bypass.tailscale0=interface
omr-bypass.tailscale0.id='9'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='ring.com'
omr-bypass.@domains[0].family='ipv4ipv6'
omr-bypass.@asns[0]=asns
omr-bypass.@asns[0].asn='104.17.128.0/20'
omr-bypass.@asns[0].note='ring'
omr-bypass.@asns[1]=asns
omr-bypass.@asns[1].asn='3.20.0.0/14'
omr-bypass.@asns[1].note='smarthings'
omr-bypass.@asns[2]=asns
omr-bypass.@asns[2].note='disney+'
omr-bypass.@asns[2].asn='139.104.96.0/24'
omr-bypass.@asns[3]=asns
omr-bypass.@asns[3].asn='139.104.97.0/24'
omr-bypass.@asns[3].note='disney+'
omr-bypass.@asns[4]=asns
omr-bypass.@asns[4].asn='139.104.98.0/24'
omr-bypass.@asns[4].note='disney+'
omr-bypass.@asns[5]=asns
omr-bypass.@asns[5].asn='13.248.128.0/20'
omr-bypass.@asns[5].note='ring'
omr-bypass.@asns[6]=asns
omr-bypass.@asns[6].asn='13.248.128.0/20'
omr-bypass.@asns[6].note='ring'
omr-bypass.@asns[7]=asns
omr-bypass.@asns[7].asn='3.33.176.0/20'
omr-bypass.@asns[7].note='ring'
omr-bypass.@asns[8]=asns
omr-bypass.@asns[8].asn='3.33.192.0/20'
omr-bypass.@asns[8].note='ring'
omr-bypass.@asns[9]=asns
omr-bypass.@asns[9].asn='3.33.208.0/20'
omr-bypass.@asns[9].note='ring'
omr-bypass.@asns[10]=asns
omr-bypass.@asns[10].asn='3.20.0.0/14'
[... 106 lines cut ...]
omr-bypass.@dpis[10]=dpis
omr-bypass.@dpis[10].proto='hulu'
omr-bypass.@dpis[10].family='ipv4'
omr-bypass.free=proto
omr-bypass.free.url='free.fr' 'freebox.fr' 'oqee.tv' 'oqee.net' 'proxad.net'
omr-bypass.orange=proto
omr-bypass.orange.url='orange.fr' 'sosh.fr' 'liveperson.net' 'liveperson.com' 'lpsn.net' 'lpsnmedia.net' 'francetelecom.fr' 'sky.com' 'skycdp.com' 'skyanywhere.com' 'epgsky.com' 'skycdn.it'
omr-bypass.sky=proto
omr-bypass.@domains[7]=domains
omr-bypass.@domains[7].name='reddit.com'
omr-bypass.@domains[7].note='reddit.com'
omr-bypass.@domains[7].family='ipv4ipv6'
omr-bypass.@domains[8]=domains
omr-bypass.@domains[8].name='rest-prod.immedia-semi.com'
omr-bypass.@domains[8].family='ipv4'
omr-bypass.@asns[19]=asns
omr-bypass.@asns[19].asn='16509'
omr-bypass.@asns[19].note='blink'
omr-bypass.@ips[8]=ips
omr-bypass.@ips[8].ip='65.8.248.0/21'
omr-bypass.@ips[8].note='blink'
omr-bypass.@ips[9]=ips
omr-bypass.@ips[9].ip='34.160.111.145'
omr-bypass.@ips[9].note='test ifconfig.me'
```

hle5128, 2024-10-14:
45.xx.xx.181                                                                          
jimmyle-m16@Jimmy-MBP-M1M ~ % curl ifconfig.me
129.222.0.72                                                                                                                                                                                                                   
jimmyle-m16@Jimmy-MBP-M1M ~ % curl ifconfig.me
45.xx.xx.181                                                                                                                                                                                             
jimmyle-m16@Jimmy-MBP-M1M ~ % curl ifconfig.me
45.xx.xx.181           

one mine curl ifconfig is showing WAN IP and next second is showing VPS IP,
during that time frame here is the log:

```
Oct 14 13:17:59 OpenMPTCProuter user.notice omr-bypass: Restart dnsmasq...
Oct 14 09:17:59 OpenMPTCProuter daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Oct 14 13:18:00 OpenMPTCProuter user.notice omr-schedule-010-services: Can't resolve via dnsmasq, restart it...
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: DNS service limited to local subnets
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5353
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for test
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for onion
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for local
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for bind
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for use-application-dns.net
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using only locally-known addresses for lan
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: using 1 more local addresses
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c.29699 - 2 names
Oct 14 09:18:02 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Oct 14 13:18:03 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Oct 14 13:18:08 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Oct 14 09:20:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.200 60:e8:5b:7e:89:82 
Oct 14 09:20:06 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.200 60:e8:5b:7e:89:82 airthings-view
Oct 14 09:20:13 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.154 80:30:dc:f2:46:58 
Oct 14 09:20:13 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.154 80:30:dc:f2:46:58 Sense-M649000836
Oct 14 09:20:16 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.166 70:88:6b:10:77:b0 
Oct 14 09:20:16 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.166 70:88:6b:10:77:b0 AWAIR-R2-1077B0
Oct 14 09:20:49 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.202 40:ca:63:6c:19:d8 
Oct 14 09:20:49 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.202 40:ca:63:6c:19:d8 Samsung-FamilyHub
Oct 14 09:21:23 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(eth1) d8:1f:12:cf:ed:ed 
Oct 14 09:21:23 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPOFFER(eth1) 192.168.100.142 d8:1f:12:cf:ed:ed 
Oct 14 09:21:23 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.100.142 d8:1f:12:cf:ed:ed 
Oct 14 09:21:23 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(eth1) 192.168.100.142 d8:1f:12:cf:ed:ed TY_WR
Oct 14 13:21:38 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Oct 14 09:21:59 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Oct 14 09:21:59 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Oct 14 13:22:01 OpenMPTCProuter user.notice omr-schedule-010-services: Can't find omr-bypass rules, restart omr-bypass...
Oct 14 13:22:01 OpenMPTCProuter user.notice omr-bypass: Starting OMR-ByPass...
Oct 14 09:22:19 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Oct 14 09:22:19 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
[... cut ...]
Oct 14 09:22:51 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Oct 14 09:22:51 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Oct 14 09:22:51 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
Oct 14 09:22:51 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dst_bypass_all_4 Error: No such file or directory
```

Ysurac, 2024-10-14:
Can you giver me the result of ```/etc/init.d/firewall stop && /etc/init.d/firewall start``` ?

Ysurac, 2024-10-14:
But it seems OMR-ByPass is very slow to start, maybe due to ASN bypass. I will check.

Nicras, 2024-10-14:
I can confirm it works with v0.61rc3-6.6 but it is indeed very slow. After saving a new bypass rule, it takes up to 10 minutes to get active and meanwhile all other bypass rules are not active as well. Besides that, saving a bypass rule led 2 times to a system crash, where the router botted and as well took up to ten minutes to come up.

closed 2025-01-18 by github-actions[bot]