Ysurac/openmptcprouter #3595: Not aggregate on incoming connections

fzonno, 2024-10-14:
## Expected Behavior
Speed not aggregate on incoming connection 

## Current Behavior

I published a service on TCP port 6180 and 1280. The incoming speed is maximum 20MB but I have two connection FTTC of about 100MB each. If I test with speedtest / nperf I receive an aggregate speed of 200MB as expected.
I have a dual nat pfsense solution as described on Wiki
[image]

## Specifications

  - OpenMPTCProuter version: 0.60-6.1
  - OpenMPTCProuter VPS version: 0.1030 6.1.0-25-amd64
  - OpenMPTCProuter VPS provider: ServerPlan
  - OpenMPTCProuter platform: x86_64

Ysurac, 2024-10-15:
What is the VPN you use ? You can try to change it in System->OpenMPTCProuter, "Wizard" tab,"Advanced settings" checkbox.

fzonno, 2024-10-15:
I left all the configuration on default.
[image]

muko82, 2024-10-15:
Try using GLORYTUN TCP in the screenshot above.
Have a look also on the graphs for bandwidth usage (Network -> MPTCP) and check if during nperf/speedtest both channels are effectively used.

It may happens that the provider (especially in Italy) is filtering mptcp. In that case you should activate the mptcp over VPN on that specific connection

fzonno, 2024-10-15:
Hi muko,
I have 2 FTTC of about 100MB each. If I try a speedtest both the connection work as expected and the aggregation works. See the image down.jpg and nperf.jpg
[image]
[image]

The problem happens when external TCP traffic come from the VPS public IP to the server. I've forwarded two TCP port but in that case the traffic uses just a connection at the maximum speeed of 20MB. I tried to switch the master connection but it always uses just one at 20MB speed.

I tried another installation (with different ISP) with incoming TCP traffic on port 443 and I had the same behaviour.

Ysurac, 2024-10-17:
Did you try another VPN ?

fzonno, 2024-10-17:
Yes I'm trying Glorytun TCP. It worked for a while perfectly (see images).
This is all incoming traffic. 200MG aggregate using both the connections.
[image]
[image]
Now it stops working and it's using just the master connection. Bit confused here! :-(

Ysurac, 2024-10-17:
I would need the Status->System log and timestamp when you loose aggregation.

fzonno, 2024-10-17:
I switched the master / enabled and it continues using just the one configured on eth2.
But if I try a speedtest it uses both and aggregate correctly.
[Log.txt](https://github.com/user-attachments/files/17415228/Log.txt)

muko82, 2024-10-17:
May It be the mptcp scheduler?

muko82, 2024-10-17:
Network -> mptcp

fzonno, 2024-10-17:
The scheduler is on "default" no other option available.
Congestion control I set on "bb3" because the documentation says that it's the best with TPC connections.

fzonno, 2024-10-18:
It's working for a while than it uses just a connection and even if it's saturated it doesn't aggregate the other.
If I do any modification (es. change between cubic and bbr2 or just an "save and apply" in the wizard it starts to works fine again.
Any clue?

Ysurac, 2024-10-18:
What is the result if you do a ```/etc/init.d/mptcp restart``` via SSH on the router when you have the issue ?
I would still need log with timestamp (the timestamp is important, and don't attach the file, paste 2 minutes before/after timestamp here)

BigFanOpenmptcprouter, 2024-10-22:
This issue seem to be duplicate with:
https://github.com/Ysurac/openmptcprouter/issues/3454

fzonno, 2024-10-23:
It worked for same days, now I have the problem again and I tried to restart the service mptcp but I can't see any change.

Oct 22 03:39:21 OpenMPTCProuter daemon.notice netifd: wan2 (9942): udhcpc: lease of 192.168.222.3 obtained from 192.168.222.254, lease time 86400
Oct 22 03:39:22 OpenMPTCProuter daemon.notice netifd: wan1 (10200): udhcpc: sending renew to server 192.168.32.1
Oct 22 03:39:22 OpenMPTCProuter daemon.notice netifd: wan1 (10200): udhcpc: lease of 192.168.32.2 obtained from 192.168.32.1, lease time 86400
Oct 22 03:39:22 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifupdate of wan1 (eth1)
Oct 22 03:39:23 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Oct 22 03:40:00 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps
Oct 22 03:40:01 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 03:54:16 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 03:54:16 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 03:54:18 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 03:59:38 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 05:39:47 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 05:39:47 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 05:39:48 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 06:41:22 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 06:41:22 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 06:41:23 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 07:52:11 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 07:52:11 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 07:52:12 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 08:09:31 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 08:24:15 OpenMPTCProuter daemon.info unbound: [24837:0] info: generate keytag query _ta-4f66. NULL IN
Oct 22 09:37:05 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 09:37:05 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 09:37:06 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 10:34:35 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 10:38:15 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 10:38:19 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 10:38:19 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 10:38:20 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 10:49:47 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 10:54:21 OpenMPTCProuter daemon.err uhttpd[7378]: [info] luci: accepted login on /admin/network/mptcp/bandwidth for root from 192.168.172.254
Oct 22 11:19:04 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 11:40:01 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 11:49:32 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 11:49:32 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 11:49:34 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 13:34:11 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 13:34:11 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 13:34:12 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 14:35:22 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration...
Oct 22 14:35:22 OpenMPTCProuter user.notice post-tracking-020-status: Check API configuration... Done
Oct 22 14:35:23 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 22 15:03:03 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 15:26:22 OpenMPTCProuter daemon.err omr-tracker[18388]: sh: 66.6667: bad number
Oct 22 15:39:21 OpenMPTCProuter daemon.notice netifd: wan2 (9942): udhcpc: sending renew to server 192.168.222.254
Oct 22 15:39:21 OpenMPTCProuter daemon.notice netifd: wan2 (9942): udhcpc: lease of 192.168.222.3 obtained from 192.168.222.254, lease time 86400
Oct 22 15:39:22 OpenMPTCProuter daemon.notice netifd: wan1 (10200): udhcpc: sending renew to server 192.168.32.1
[... cut ...]
Oct 23 10:12:01 OpenMPTCProuter user.notice OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part
Oct 23 10:35:54 OpenMPTCProuter daemon.err uhttpd[7378]: [info] luci: accepted login on /admin/network/mptcp/bandwidth for root from 192.168.172.254
Oct 23 10:38:34 OpenMPTCProuter authpriv.info dropbear[2521]: Child connection from 192.168.172.254:11329
Oct 23 10:38:42 OpenMPTCProuter authpriv.notice dropbear[2521]: Password auth succeeded for 'root' from 192.168.172.254:11329

closed 2025-01-26 by github-actions[bot]